evcc-io/evcc #8626: SMA STP XX-50

hannescl, 2023-06-25:
### Describe the bug

Hallo zusammen,

SMA hat mit den STP XX-50 Wechselrichtern neue Modelle seit 2022 an den Markt gebracht. Diese nutzen nativ ennexos (wie der DataManager von SMA) und lassen sich selbst auch als "Ersatz" für einen DataManager Lite (max. 5 Geräte) verwenden.

Seit ein paar Tagen haben wir einen STP 25-50 im Einsatz. Dieser lässt sich wie der DataManager mit dem Template sma-data-manager in evcc einbinden. Es wird jedoch folgender Fehler alle paar Sekunden gemeldet: "battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'".

### Steps to reproduce

1. Verbinde evcc mit einem SMA STP XX-50 Wechselrichter via Template sma-data-manager.
2. Schaue in das Error Log.
3.
...

### Configuration details

```yaml
- type: template
  name: battery3
  template: sma-data-manager
  usage: battery
  host: 192.168.110.XX
  port: 502
```

### Log details

```text
battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.1

hannescl, 2023-06-25:
Leider kann ich via HAOS das Debugging nicht aktivieren. 

Hier nochmal der Log Extract:
[site  ] ERROR 2023/06/25 12:41:02 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:41:12 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:41:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:41:31 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:41:32 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:41:42 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:41:52 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:42:02 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:42:12 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:42:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:42:32 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:42:42 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:42:52 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:43:02 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:43:12 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:43:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:43:32 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:43:42 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:43:52 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:44:02 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:44:12 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:44:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:44:32 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:44:42 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:44:52 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:45:02 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:45:12 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:45:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:45:32 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:45:42 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:45:52 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:46:02 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:46:12 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:46:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:46:32 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:46:42 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:46:52 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:47:02 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:47:12 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:47:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[... cut ...]
[site  ] ERROR 2023/06/25 12:49:22 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:49:32 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:49:42 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site  ] ERROR 2023/06/25 12:49:52 battery 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'

premultiply, 2023-06-25:
Hat der WR überhaupt eine Batterie???
Wenn je wäre einfach nur dieser Eintrag falsch bzw. zuviel.

hannescl, 2023-06-25:
> 

Der WR selbst natürlich nicht, jedoch gibt es in dem System einen Sunny Island 8.0H-13 als Batterie-WR. Dieser ist auch im STP 25-50 "DataManager" eingebunden.

premultiply, 2023-06-25:
Aha!
Der "DataManager" scheint ihn jedenfalls nicht wie erwartet zu "kennen".

Du solltest die Geräte aber auch wie üblich direkt einbinden können.
Entweder über das das Template "sma-inverter" oder falls das nicht geht über das Standardtemplate "sunspec-inverter".
Für Letzteres muss natürlich auch Modbus am SI aktiv sein. Die Standard-SunSpec-Modbus-ID bei SMA ist dann 126.

hannescl, 2023-06-25:
Also spannend ist, dass er dennoch Werte für die Batterie bereitstellt (z.B. Ladezustand). Ich habe parallel noch ein SMA Energy Meter mit evcc verbunden, jedoch gehe ich nicht davon aus, dass das EM die Werte der Batterie liefert.

sma-inverter hat nicht funktioniert und der WR unterstützt aktuell noch kein Sunspec (laut SMA kommt hier in Q3 ein Update).

premultiply, 2023-06-25:
`sma-inverter` und auch SunSpec sollte aber eigentlich beim Sunny Island funktionieren.
Da muss man aber dran denken auch das richtige Passwort für Speedwire anzugeben und ggf. auch darauf zu achten dass Speedwire unverschlüsselt sein muss.

Der Rest klappt doch über das DataManager-Template vom PV-WR, oder?

hannescl, 2023-06-26:
Genau - ich kann damit Daten direkt Batterie-WR abgreifen. Leider fehlen dann jedoch PV Ertragsdaten.

Generell scheint das sma-data-manager Template auch inkl Batteriedaten am WR zu funktionieren. Es kommt jedoch besagter Fehler im Log.

hannescl, 2023-06-26:
Ich sehe gerade jedoch, dass die Daten zur Erzeugung nicht korrekt sind. Es werden 300kw angezeigt, trotzdem es 800kw sind. Tagsüber bei hohee Erzeugung lagen die Ist- und durch evcc dargestellten Werte näher beieinander.

premultiply, 2023-06-26:
Am besten immer erstmal mit `evcc meter` anfangen und da schauen dass die benötigten Werte dort alle einwandfrei gelistet werden.

Nach Möglichkeit die Daten immer direkt von den einzelnen Geräten holen und nicht vom Data-Manager.

hannescl, 2023-06-26:
Ich habe da auch schon alles ausprobiert. Aktuell hole ich mir aus dem Energy-Meter die Netzanschluss Daten. Aus dem Batterie-WR bekomme ich die Batterie-Daten (diese bekomme ich jedoch auch vom WR über die Data-Manager Schnittstelle). Nur die PV Ertrags-Daten fehlen mir noch.