evcc-io/evcc #8628: Im PV modus started das Laden mit voller Leistung obwohl pv=0W hat.

ickeundso, 2023-06-25:
### Describe the bug

Habe meine ZOE am Freitag Abend angesteckt. keine Sonne / Hausbatterie war bei 54%. 
Es wurde dann aber sofort das Laden mit der in maximalen Leistung gestartet. 
Für die ZOE habe ich ein minSOC von 70% eingestellt. Geladen wurde aber bis 77% und nur weil die Hausbatterie leer war wurde das Laden nach ca. 2h gestoppt.

Meine Erwartung wäre hier, im PV Modus minSOC zu ignorieren und maximal bis zum konfigurierten Batterie Soc von 25% und  dann auch nur mit max. 3,7kW zu laden. 

### Steps to reproduce

Einfach EV bei pv=0W einstecken.
Keine Ahnung ob das reproduzierbar ist oder am Zusammenspiel ZOE 7 go-e charger hängt.

das log enthält nicht alles , 

ein paar Zyklen vor 
[lp-1  ] INFO 2023/06/23 22:35:40 car connected
und nach dem ende habe ich drin gelassen
[lp-1  ] DEBUG 2023/06/24 00:10:47 charge power: 0W

### Configuration details

```yaml
vehicles:
- type: template
  template: renault 
  title: ZOE  
  user: ***  
  password: ***  
  vin: ***  
  capacity: 41  
  phases: 3  
  icon: car  
  cache: 15m  
  minSoc: 70  
  targetSoc: 100  
  minCurrent: 8  
  maxCurrent: 16  
  name: rzoe

loadpoints:
- title: Stellplatz
  charger: goecharger
  vehicle: rzoe
  mode: pv 
  phases: 0
  mincurrent: 8
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    poll:
      mode: always
      interval: 60m
    estimate: true
  enable:
    threshold: -1000
    delay: 5m
  disable:
    threshold: 500
    delay: 5m
          

site:
  title: Heine3
  meters:
    grid: grid1
    pv:
    - sun9kWp
    battery:
    - luna15kWh
[... 861 lines cut ...]
[site  ] DEBUG 2023/06/24 00:20:17 ----
[lp-1  ] DEBUG 2023/06/24 00:20:17 charge power: 0W
[site  ] DEBUG 2023/06/24 00:20:17 static grid tariff
[site  ] DEBUG 2023/06/24 00:20:18 pv power: 0W
[site  ] DEBUG 2023/06/24 00:20:19 battery soc: 5%
[site  ] DEBUG 2023/06/24 00:20:19 battery power: -35W
[site  ] DEBUG 2023/06/24 00:20:21 grid power: 178W
[site  ] DEBUG 2023/06/24 00:20:24 grid currents: [0.45 0.47 0.38]A
[site  ] DEBUG 2023/06/24 00:20:24 giving priority to battery charging at soc: 5%
[site  ] DEBUG 2023/06/24 00:20:24 site power: 328W
[lp-1  ] DEBUG 2023/06/24 00:20:24 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/24 00:20:24 charge meter total import: 5109.842kWh
[lp-1  ] DEBUG 2023/06/24 00:20:24 charger status: B
[lp-1  ] DEBUG 2023/06/24 00:20:24 next soc poll remaining time: 50m30s
[lp-1  ] DEBUG 2023/06/24 00:20:25 pv charge current: 0A = 0A + -1.43A (328W @ 1p)
[site  ] DEBUG 2023/06/24 00:20:32 ----
```

### What type of operating system are you running?

Docker container

### Version

0.118.0

andig, 2023-06-25:
> Meine Erwartung wäre hier, im PV Modus minSOC zu ignorieren

Siehe Doku, das ist nicht der Fall.

closed 2023-06-25 by andig

ickeundso, 2023-06-25:
[more logs](https://share.mailbox.org/ajax/share/04f82c0902364244476000723642490eba00426fcd106c0f/1/8/NTM/NTMvMjc)

ickeundso, 2023-06-25:
ok, wofür ist dann noch der modus Solar+PV. Dach mit PV kann man das genau erreichen