evcc-io/evcc #8629: Nur dreiphasiges Laden von Ioniq5 an KebaP30 / kein Wechsel auf PV möglich

andreasluft76, 2023-06-25:
### Describe the bug

evcc will bei meinem Ioniq5 (Modelljahr 2022) kein PV Laden zulassen, da angeblich zu wenig Überschuss vorhanden ist.  Bei einem Citroen C5 hybrid funktioniert es. Der lädt grundsätzlich nur einphasig und es werden bei vergleichbare Sonne zw. 1,x bis zu 3,0 kWh geladen.

### Steps to reproduce

1. Anschluss der Keba P30 an den Ioniq5
2. Auwahl von "PV"
3. Hinweis "Zu wenig Überschuss. Pausiere in ..."
4. Alternativ geht "Min +PV". Dann wird 3-phasig mit 7 kW geladen.

### Configuration details

```yaml
Hyundai Ioniq5
Keba P30 Wallbox (ist auf 11 kW eingestellt)
RCT Power Storage Wechselrichter (6kW) mit 9,6 kW RCT Akku und RCT Power Switch.
2 Strings á 9x400 Watt - in Summe 7,2kW / aktuell in evcc erkannte Sonne: 4,9 KW bei Hausverbauch von um die 0,5 kW
evcc läuft unter hass.io in einer virtuellen Maschine auf einer Synology DS1522
```

### Log details

```text
grid energy: invalid response: Cmd Response (05) Id #INVALID (43257820) Data [68 249 99 13]
vor 4 Sekunden

pv 1 power: invalid response: Cmd Response (05) Id Total energy batt in [Wh] (5570401B) Data [72 103 65 5]
vor 23 Sekunden

charger: recv timeout
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.1

andig, 2023-06-25:
Wenn Deine Keba 3p angeschlossen ist dann wird sie auch 3p laden wenn das Auto das kann. evcc kann an der Hardware nix ändern.

closed 2023-06-25 by andig

andreasluft76, 2023-06-25:
Danke für die schnelle Antwort. Ist da kein Umschalten auf 1-phasig möglich? Da sie ja grundsätlich 1 phasig Laden kann, dachte ich, dass eine entsprechende Ansteuerung möglich wäre. Dann sind ja 7 kW dreiphasig das Minimum.

andig, 2023-06-25:
Die Box kann nur 3p. Ob das Auto 1p konsumiert ist davon völlig unabhängig.