evcc-io/evcc #8643: Wake-up Vehicle über Porsche Api wird nicht versucht (Requests tauchen nicht im Log auf)

MarkusGH, 2023-06-26:
Konfiguration: openWB mit Phasenumschaltung, Porsche PHEV
Bekanntlich schläft das Fahrzeug (wie wahrscheinlich alle Porsche PHEV) ach 5 Ladeunterbrechungen ein und lässt sich dann über CP Unterbrechung nicht wecken. Wecken über API (Status über Porsche App abfragen) scheint zu funktionieren.

#8204 implementiert den Versuch das in evcc zu nutzen.

Ich habe hier allerdings beim Testen der Funktionalität mit dem aktuellen Nightly ein unverständliches Verhalten gesehen.

Erwartetes Verhalten:

```
Jun 26 10:25:37 evcc[3730455]: [site  ] INFO 2023/06/26 10:25:37   vehicles:
Jun 26 10:25:37 evcc[3730455]: [site  ] INFO 2023/06/26 10:25:37     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Jun 26 10:25:37 evcc[3730455]: [lp-1  ] INFO 2023/06/26 10:25:37   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
Jun 26 10:25:37 evcc[3730455]: [lp-2  ] INFO 2023/06/26 10:25:37   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
Jun 26 10:32:43 evcc[3730455]: [lp-1  ] DEBUG 2023/06/26 10:32:43 wake-up timer: start
Jun 26 10:32:43 evcc[3730455]: [porsche] TRACE 2023/06/26 10:32:43 GET https://api.porsche.com/e-mobility/de/de_DE/E3/***?timezone=Europe/Berlin
JJun 26 10:32:44  evcc[3730455]: [openwb] TRACE 2023/06/26 10:32:44 send openWB/set/isss/Cpulp2: '1'
Jun 26 10:32:44 evcc[3730455]: [porsche] TRACE 2023/06/26 10:32:44 GET https://api.porsche.com/vehicle-data/de/de_DE/status/***
Jun 26 10:32:45 evcc[3730455]: [porsche] TRACE 2023/06/26 10:32:45 GET https://api.porsche.com/e-mobility/de/de_DE/E3/***?timezone=Europe/Berlin
```

Heißt 30 Sekunden nach Start des Wake-Up Timers wird WakeUp erst über Charger, dann über Porsche Api versucht.
(die e-mobility Aufrufe sind wohl reguläre Statusupdates)

Ich sehe aber oft auch sowas:

```
Jun 26 10:33:46 evcc[3750944]: [site  ] INFO 2023/06/26 10:33:46     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Jun 26 10:33:46 evcc[3750944]: [lp-1  ] INFO 2023/06/26 10:33:46   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
Jun 26 10:33:46 evcc[3750944]: [lp-2  ] INFO 2023/06/26 10:33:46   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
Jun 26 10:33:46 evcc[3750944]: [porsche] TRACE 2023/06/26 10:33:46 GET https://api.porsche.com/e-mobility/de/de_DE/E3/***?timezone=Europe/Berlin
Jun 26 10:33:47 evcc[3750944]: [lp-1  ] DEBUG 2023/06/26 10:33:47 wake-up timer: start
Jun 26 10:34:22 evcc[3750944]: [openwb] TRACE 2023/06/26 10:34:22 send openWB/set/isss/Cpulp2: '1'
Jun 26 10:34:52 evcc[3750944]: [porsche] TRACE 2023/06/26 10:34:52 GET https://api.porsche.com/e-mobility/de/de_DE/E3/***?timezone=Europe/Berlin
Jun 26 10:35:52 evcc[3750944]: [porsche] TRACE 2023/06/26 10:35:52 GET https://api.porsche.com/e-mobility/de/de_DE/E3/***?timezone=Europe/Berlin
```

Hier wird nach Senden des "CP Unterbrechen" Befehls kein Wake-Up an das Porsche Api gesendet.

Beim WakeUp gibt es kein Caching - die Requests sollten immer raus gehen.

Gibt es da ein Buffering auf Request - Ebene?

andig, 2023-06-29:
Siehe auch https://github.com/evcc-io/evcc/pull/8684

andig, 2023-06-29:
Mit den Logschnipseln ohne weiteren Kontext leider nicht zu beantworten.

closed 2023-06-29 by andig

MarkusGH, 2023-06-29:
Das hängt aber nicht zusammen - Wenn ein WakeUp über CC Unterbrechung gemacht wird sollte immer auch ein WakeUp  über API versucht werden, falls vom Vehicle unterstützt. Das ist der Fall, aber die Requests gehen nicht raus.

andig, 2023-06-29:
Hat der Loadpoint kein vehicle?

MarkusGH, 2023-06-29:
> Hat der Loadpoint vllt. kein vehicle? 

Schau mein Log an - vehicle ist da und unterstützt WakeUp

> Vielmehr gibts ja nicht das schief gehen kann:

Dachte ich auch.

> Im Zweifel musst Du da noch Debugstatements einbauen. 

Ich stelle einen entsprechenden PR.

> Ist das denn ein echtes Problem oder vermisst Du nur was im Log?

Naja - die Porsches wachen la mit CC Unterbrechung (noch) nicht auf, und so kriegt man nie raus ob und wie aufwecken über API geht...

MarkusGH, 2023-06-29:
Debug logging: #8687