evcc-io/evcc #8649: Using http plugin to connect to Homeassistant API doesn't work

vistalba, 2023-06-26:
### Describe the bug

As I do have some trouble if multiple devices connect to my inverter I want to consume the data from Homeassistant.
So basicly EVCC should connect over https to Homeassistant and get the data over the JSON API.

Basicly this works with curl if I use the following command:

```
curl --silent 'https://homeassistant-url.domain.com:8123/api/states' -H "Authorization: Bearer eyJ....Y" |  jq .[1].state
"-11.808"
```

This is the actual value of PV Power produced in kW.
As the connectivity is working fine if I exec into the docker container and use `wget https://homeassistant-url.domain.com:8123/api/states --header 'Authorization: Bearer eyJ....Y'` I assume that there is an issue with `auth: bearer`.

### Steps to reproduce

Try to use HTTP plugin to access Homeassistant API with Bearer Token.

### Configuration details

```yaml
- name: fronius-ha
  type: custom
  power:
    source: http
    uri: 'https://homeassistant-url.domain.com:8123/api/states'
    method: GET
    headers:
      - content-type: application/json
    auth:
      type: bearer
      password: eyJ....Y
    insecure: false
    jq: .[1].state
    scale: 0.001
    timeout: 5s
```

### Log details

```text
[site  ] ERROR 2023/06/26 17:22:55 pv 1 power: unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.1

andig, 2023-06-27:
Please run with `--log error,http:trace --log-headers` to see whats going on. Closing until log available.

closed 2023-06-27 by andig

vistalba, 2023-07-05:
@andig 
I ran it with the `--log error,http:trace --log-headers` parameters. And I found the error. There was some issue with the Bearer token. It had a space in it. Why ever... most probably some copy&past issue with newline. 
Anyway, we can leave it closed as it has been solved. Sorry and thanks a lot to point me to the right debug parameters.

arcidodo, 2024-10-04:
just for information: i used this jq command, to select an entity instead an index. 

```
      jq: '.[] | select(.entity_id=="sensor.p1_meter_power") | .state'

```

m-kloeckner, 2024-11-29:
I ran into the same "401 (unauthorized)" error and it took me a while to get it working. 
Just in case someone else needs a fix: set the the auth header manually in your configuration.

```
- name: fronius-ha
  type: custom
  power:
    source: http
    uri: 'https://homeassistant-url.domain.com:8123/api/states'
    method: GET
    headers:
      - content-type: application/json
      - Authorization: Bearer YOUR_BEARER_TOKEN
    insecure: false
    jq: .[1].state
    scale: 0.001
    timeout: 5s
```