evcc-io/evcc #8674: stop charging - ich verstehe aber nicht, warum.

abc2006, 2023-06-28:
### Describe the bug

Wallbox stoppt die Ladung und startet wieder - heute tagsüber bei absolut ausreichend PV-Leistung beobachtet, aber auch die nacht mehrfach.
Hab ich was falsch konfiguriert? 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

log: debug
levels:
  cache: error

plant: <Token entfernt>

interval: 10s

sponsortoken: <Token entfernt>

telemetry: false
mqtt: 
  broker: 192.168.0.25
  
meters:
- type: custom
  name: grid
  power: 
    source: mqtt
    topic: tele/EINSPEISEZAEHLER/total_power  
  energy: 
    source: mqtt
    topic: tele/EINSPEISEZAEHLER/total_energy_need
- type: custom
  name: pvs
  power:
    source: mqtt
    topic: tele/SDM230_1/total_power
    scale: -1
  energy: 
    source: mqtt
    topic: tele/SDM230_1/total_energy_feed
- type: custom
  name: pvls1
  power:
    source: mqtt
    topic: tele/fsp10k/Solar_input_power_1
- type: custom
  name: pvls2
  power:
    source: mqtt
    topic: tele/fsp10k/Solar_input_power_2
- type: custom
  name: batul
  power: 
    source: mqtt
    topic: tele/jkbms_unten_links/1_Power_total
    scale: -1
  energy: 
    source: mqtt
[... 224 lines cut ...]
[site  ] DEBUG 2023/06/28 13:55:25 battery power: -10175W
[site  ] DEBUG 2023/06/28 13:55:25 grid power: -3445W
[site  ] DEBUG 2023/06/28 13:55:25 site power: -13520W
[lp-2  ] DEBUG 2023/06/28 13:55:25 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/06/28 13:55:25 charge total import: 235.522kWh
[lp-2  ] DEBUG 2023/06/28 13:55:26 charger status: B
[lp-2  ] DEBUG 2023/06/28 13:55:26 final charge energy: 0.056kWh
[lp-2  ] INFO 2023/06/28 13:55:26 stop charging <-
[telegram] DEBUG 2023/06/28 13:55:26 sending to ***
[lp-2  ] DEBUG 2023/06/28 13:55:26 charge total import: 235.522kWh
[lp-2  ] ERROR 2023/06/28 13:55:26 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'tesla': cannot get vehicles: Get "https://owner-api.teslamotors.com/api/1/vehicles": dial tcp: lookup owner-api.teslamotors.com: i/o timeout
[lp-2  ] DEBUG 2023/06/28 13:55:26 pv charge current: 19.6A = 0A + 19.6A (-13520W @ 3p)
[lp-2  ] DEBUG 2023/06/28 13:55:26 pv timer reset
[lp-2  ] DEBUG 2023/06/28 13:55:26 pv timer inactive
[lp-2  ] DEBUG 2023/06/28 13:55:27 charger enable
[lp-2  ] DEBUG 2023/06/28 13:55:27 wake-up timer: start
```

### What type of operating system are you running?

Docker container

### Version

118,1

andig, 2023-06-29:
Zeig bitte mal ein vollständiges Logfile *nur* für den problematischen Ladepunkt. Hier ist nichtmal klar, welcher Deiner Beiden das Problem hat.

closed 2023-06-29 by andig

abc2006, 2023-06-29:
Hi, 
ich verstehe deinen Wunsch völlig - diesen habe ich auch öfter mal.
Leider steht in der Doku:
https://docs.evcc.io/docs/guides/setup/#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen
_Im Falle einer Docker Installation kann man über docker logs die Nachrichten anzeigen lassen, siehe [Docker Dokumentation](https://docs.docker.com/config/containers/logging/)_

Wenn ich mir dann die Docker Dokumentation anzeigen lasse: 
_By default, docker logs shows the command’s STDOUT and STDERR
In some cases, docker logs may not show useful information unless you take additional steps.

If you use a [logging driver](https://docs.docker.com/config/containers/logging/configure/) which sends logs to a file, an external host, a database, or another logging back-end, and have [“dual logging”](https://docs.docker.com/config/containers/logging/dual-logging/) disabled, docker logs may not show useful information.
If your image runs a non-interactive process such as a web server or a database, that application may send its output to log files instead of STDOUT and STDERR._
Also ist das Logfile meinem Verständnis nach vollständig. 

Es geht um den lp-2, der im Log meldet: 
_[lp-2  ] INFO 2023/06/28 13:55:26 stop charging <-_
Wenn du mir sagt, wie ich ein "vollständiges Logfile _nur_ für den Ladepunkt" erstelle, erfülle ich sehr gerne deine Bitte.
Wenn ich jetzt das logging auf lp-2 beschränke, fehlen aber alle angaben zu verfügbaren Leistungen usw. 

Wenn du wolltest, dass ich alle meldungen rauslösche, bei denen *nicht* lp-2 vorne steht, dann kann ich das auch gerne für dich tun. 

So weiss ich jetzt aber leider nicht, was ich tun kann/soll.

Grüße, 
Stephan