evcc-io/evcc #8689: Es erfolgt nach Phasenumschaltung durch Trennschalter und Neustart von Go-E keine Ladestärken Anpassung

Optic00, 2023-06-29:
### Describe the bug

Auto ist nicht eingestöpselt. Aufgrund der Wetterlage soll auf 1P mittels Leistungstrennschalter umgeschaltet werden.

Verhalten:
EVCC fängt an zu Laden, regelt aber den Ladestrom nicht runter, lädt mit 16A statt auf 6A zu regeln. 

Workaround:
EVCC Neu starten. Ladestrom wird korrekt bis auf 6A herunter geregelt und alles funktioniertAuto ist nicht eingestöpselt. Aufgrund der Wetterlage soll auf 1P mittels Leistungstrennschalter umgeschaltet werden.

### Steps to reproduce

1. Go-E Charger v2 Schutzschalter raus
2. Trennschalter von 3P auf 1P
3. Go-E Charger v2 Schutzschalter rein
4. Auto wird JETZT eingestöpselt
5. EVCC startet die PV-Ladung mit 16A, regelt nicht herunter und bricht das Laden nach Timer ab weil zu wenig Überschuss vorhanden ist für 16A. Für 6-8A wäre genug da.

### Configuration details

```yaml
EVCC 0.118.1 - Bug war in der Vergangenheit auch schon vorhanden
Go-E Charger v2
E-Golf 2P und Tesla MY 3P zeigen dasselbe Verhalten
```

### Log details

```text
Mit Homeassistant nicht verfügbar.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.1

closed 2023-06-29 by andig

Adhoc61, 2023-07-06:
Bei mir wird mit der go e v2 gar nicht mehr Solar-optimiert geladen. Alle Einstellungen bzgl. Ladegeschwindigkeit muss ich manuell in der go e vornehmen. Auch das starten und stoppen. Schade, dass hat bis vor kurzem noch sehr gut von EVCC gesteuert funktioniert

Adhoc61, 2023-07-06:
Bei mir wird mit der go e v2 gar nicht mehr Solar-optimiert geladen. Alle Einstellungen bzgl. Ladegeschwindigkeit muss ich manuell in der go e vornehmen. Auch das starten und stoppen. Schade, dass hat bis vor kurzem noch sehr gut von EVCC gesteuert funktioniert. Ich habe einen Senec v3 Speicher und einen Kia ev6 sowie einen id3 als e.fahrzeuge

premultiply, 2023-07-06:
🔮

andig, 2023-07-06:
> Bei mir wird mit der go e v2 gar nicht mehr Solar-optimiert geladen.

@Adhoc61  Ich muss mich hier auch mal äußern. Diese "alles ist schlecht" Postings ohne jede verwertbare Information sind völlig sinnlos und frustrierend. Auf meckern können wir verzichten, konstruktive Fehlersuche sehr gerne. Und die ist gar nicht schwer:

1. Es ging schonmal? Wunderbar- dann herausfinden, was Du geändert hast!
2. Es ging noch nie? Dann neues Issue mit den notwendigen Informationen und wir werden die Ursache finden.