evcc-io/evcc #8702: Edit charge plan available on computer, not on mobile

kalaws, 2023-06-29:
### Describe the bug

I can set up a charge plan on my phone. Once I have, there is no link under "Departure" any more, meaning that I cannot edit the plan on the phone. This link is visible when accessing EVCC on computer.

### Steps to reproduce

1. Set charge plan in mode min+pv
[image]
[image]

### Configuration details

```yaml
network: 
  schema: http
  host: 0.0.0.0
  port: 7070 # uri for ui #0.0.0.0
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: ***

# log settings
log: debug # org. "error" [error/debug/trace]
levels:
  core: trace
  lp-1: trace
  lp-2: trace
  http: trace
  #cache: debug # org. finns ej

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
- name: grid
  type: template
  template: sunspec-inverter
  usage: grid
  host: 192.168.1.156
  port: 502
  id: 240
  modbus: tcpip

- name: pv
  type: template
  template: sunspec-inverter
  usage: pv
  host: 192.168.1.156
  port: 502
  modbus: tcpip
  id: 1

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
chargers:
- name: go-e
  type: template
  template: go-e-v3
  host: 192.168.1.115 # go-e ip address (local)

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/evcc-io/config#vehicles
vehicles:
- name: ioniq5
  type: hyundai
  title: Ioniq 5 # display name for UI
  capacity: 73 # kWh
  user: *** # user
  password: *** # password
  language: sv
  #onIdentify: # set defaults when vehicle is identified
  #  minSoC: 25 # charge to at least 20% independent of charge mode
    #targetSoC: 80 # limit charge to 80%

# site describes the EVU connection, PV and home battery
site:
  title: Hemma # display name for UI
  meters:
    grid: grid # grid meter
    pv: pv # pv meter
    #battery: battery # battery meter

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Laddbox # display name for UI
  charger: go-e # charger
  #meters:
  #  charge: charge # charge meter
  vehicle: ioniq5
  # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  # - ID.3
  # - e-Up
  mode: minpv
  resetOnDisconnect: true # set defaults when vehicle disconnects
  soc:
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    #target: 100 # always charge to 100%
    estimate: true # set true to interpolate between api updates
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 0 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

# tariffs are the fixed or variable tariffs
# cheap can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: SEK
  grid:
    type: tibber
    token: "*****"
  
  # type: fixed
  #  price: 1.5 # SEK/kWh
#    zones:
#      - days: Mo-So
#        hours: 21:30-0,0-5:30
#        price: 0.2

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: tibber
    token: "*****"
    #type: fixed
    #price: 1.2 # SEK/kWh

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: mqtt
  password: mqtt

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # public: # public key
  # private: # private key

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

### Log details

```text
[lp-1  ] DEBUG 2023/06/29 22:05:31 charge currents: [6.1 5.7 6.1]A
[lp-1  ] DEBUG 2023/06/29 22:05:31 detected active phases: 3p
[lp-1  ] DEBUG 2023/06/29 22:05:31 charge total import: 115.444kWh
[lp-1  ] DEBUG 2023/06/29 22:05:32 charger status: C
[lp-1  ] ERROR 2023/06/29 22:05:32 vehicle soc: vehicle not available: cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request)
[lp-1  ] DEBUG 2023/06/29 22:05:32 planned 2h9m30s until 2023-06-30 10:00:00 +0200 CEST at 11040W: total plan duration: 2h9m30s, avg cost: 1.161
[lp-1  ] TRACE 2023/06/29 22:05:32   slot from: 2023-06-30 02:50:30 +0200 CEST to 2023-06-30 03:00:00 +0200 CEST cost 1.185
[lp-1  ] TRACE 2023/06/29 22:05:32   slot from: 2023-06-30 03:00:00 +0200 CEST to 2023-06-30 04:00:00 +0200 CEST cost 1.158
[lp-1  ] TRACE 2023/06/29 22:05:32   slot from: 2023-06-30 04:00:00 +0200 CEST to 2023-06-30 05:00:00 +0200 CEST cost 1.161
[lp-1  ] DEBUG 2023/06/29 22:05:32 pv charge current: 0A = 6A + -7.02A (4846W @ 3p)
[site  ] DEBUG 2023/06/29 22:05:38 ----
[lp-1  ] DEBUG 2023/06/29 22:05:38 charge power: 4220W
[site  ] DEBUG 2023/06/29 22:05:38 dynamic grid tariff
[site  ] DEBUG 2023/06/29 22:05:38 pv power: 8W
[site  ] DEBUG 2023/06/29 22:05:39 grid power: 4835W
[site  ] DEBUG 2023/06/29 22:05:40 grid powers: [1610 1728 1497]W
[site  ] DEBUG 2023/06/29 22:05:41 grid currents: [7.39 7.57 6.53]A
[site  ] DEBUG 2023/06/29 22:05:41 site power: 4835W
[lp-1  ] DEBUG 2023/06/29 22:05:41 charge currents: [6.1 5.7 6.1]A
[lp-1  ] DEBUG 2023/06/29 22:05:41 detected active phases: 3p
[lp-1  ] DEBUG 2023/06/29 22:05:41 charge total import: 115.456kWh
[lp-1  ] DEBUG 2023/06/29 22:05:41 charger status: C
[lp-1  ] ERROR 2023/06/29 22:05:41 vehicle soc: vehicle not available: cannot create vehicle 'hyundai': login failed: unexpected status: 400 (Bad Request)
[lp-1  ] DEBUG 2023/06/29 22:05:41 planned 2h9m26s until 2023-06-30 10:00:00 +0200 CEST at 11040W: total plan duration: 2h9m26s, avg cost: 1.161
[lp-1  ] TRACE 2023/06/29 22:05:41   slot from: 2023-06-30 02:50:34 +0200 CEST to 2023-06-30 03:00:00 +0200 CEST cost 1.185
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.1

naltatis, 2023-06-30:
@kalaws I assume both devices run latest iOS? Is the result the same if you access the evcc Web-UI directly in Safari and not through HomeAssistant?

kalaws, 2023-06-30:
One is on 16.6, one is on 16.5.1. result is the same on safari.

kalaws, 2023-06-30:
Computer access was through Firefox.

kalaws, 2023-07-11:
Bug persists on MacOS Safari (Version 16.5 (18615.2.9.11.4)), so it is not dependent on computer/mobile. In other words, on computer the bug show up on Safari, but not Firefox.

naltatis, 2023-07-12:
Was a simple one. The Swedish translation contained a broken placeholder. I've fixed the translation and added a Weblate quality check for this type of mistake.

closed 2023-07-12 by naltatis

naltatis, 2023-07-12:
I've also noticed that some Swedish texts are quite long an cause layout problems in some screen sizes. I'll make an update to the layout to truncate (`...`) if necessary. If you have better/shorter alternative Swedish translations, feel free to update the translation 😃

https://hosted.weblate.org/projects/evcc/evcc/sv/

kalaws, 2023-07-12:
Sweet, thanks! I’ll take a look at the translations.

PR #8903 (2023-07-12): Translations update from Hosted Weblate
fixes #8702

Translations update from [Hosted Weblate](https://hosted.weblate.org) for [evcc/evcc](https://hosted.weblate.org/projects/evcc/evcc/).

Current translation status:

[image]