evcc-io/evcc #8704: Moi

manuelefant, 2023-06-29:
### Describe the bug

Moin,

ich versuche evcc im HASS zum laufen zu kriegen.
Nachdem ich mich durch den Dschungel gehangelt habe, scheint es auch laufen zu wollen, fast... gibt eben die Fehlermeldung "wait for initial value" ... leider gehts da nicht weiter. 

Habe diese Anleitung herbeigezogen:
https://herrmayr.de/pv-ueberschussladen-mit-goodwe-wechselrichter-home-assisant-und-evcc/

(bei mir heißt es nur "pvnetto")... 

Jemand ne Idee?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: homeassistant.local
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 

interval: 10s # control cycle interval

sponsortoken: abc

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- name: grid
  type: custom
  power: # Leistung (W)
    source: mqtt
    topic: pvnetto
    scale: -1
    timeout: 10s # don't accept values older than timeout

chargers:
- name: WallboxK26
  type: template
  template: go-e-v3 
  host: 192.168.178.42

vehicles:
- type: template
  template: peugeot 
  title: 208e  
  user: username
  password: password
  vin: VIN
  capacity: 50  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  minSoc: 10  
  targetSoc: 85  
  minCurrent: 6  
  maxCurrent: 16  
  priority: 1  
  name: ev3

loadpoints:
- title: Wallbox K26
  charger: WallboxK26
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  meters:
    grid: grid

# mqtt message broker
mqtt:
  broker: homeassistant.local:1883
  user: username
  password: password
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/06/30 00:27:24 evcc 0.118.1
[main  ] INFO 2023/06/30 00:27:24 using config file: /config/evcc.yaml
[main  ] INFO 2023/06/30 00:27:24 starting ui and api at :7070
[db    ] INFO 2023/06/30 00:27:24 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2023/06/30 00:27:24 connecting evcc-1878919987 at tcp://homeassistant.local:1883
[mqtt  ] DEBUG 2023/06/30 00:27:24 tcp://homeassistant.local:1883 connected
[lp-1  ] WARN 2023/06/30 00:27:25 locking phase config to 3p for switchable charger
[site  ] DEBUG 2023/06/30 00:27:25 static grid tariff
[site  ] INFO 2023/06/30 00:27:25 site config:
[site  ] INFO 2023/06/30 00:27:25   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2023/06/30 00:27:25     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/06/30 00:27:25   vehicles:
[site  ] INFO 2023/06/30 00:27:25     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/06/30 00:27:25 loadpoint 1:
[lp-1  ] INFO 2023/06/30 00:27:25   mode:        pv
[lp-1  ] INFO 2023/06/30 00:27:25   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/06/30 00:27:25   meters:      charge ✓
[lp-1  ] INFO 2023/06/30 00:27:25     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2023/06/30 00:27:25 static grid tariff
[lp-1  ] DEBUG 2023/06/30 00:27:25 phase timer inactive
[lp-1  ] DEBUG 2023/06/30 00:27:25 pv timer inactive
[lp-1  ] DEBUG 2023/06/30 00:27:25 guard timer inactive
[lp-1  ] DEBUG 2023/06/30 00:27:25 max charge current: 6A
[site  ] DEBUG 2023/06/30 00:27:25 ----
[lp-1  ] DEBUG 2023/06/30 00:27:25 charge power: 0W
[site  ] DEBUG 2023/06/30 00:27:25 static grid tariff
[mqtt  ] DEBUG 2023/06/30 00:27:25 pvnetto wait for initial value
```

### What type of operating system are you running?

HomeAssistant Add-on

bastian1607, 2023-06-30:
Ich habe auch EVCC im Hass am laufen. Das „Problem“ ist folgendes:
Dein Topic aktualisiert sich bei neuen Werten. Die Info sagt, dass auf einen Wert gewartet wird. Nachts ist die PV halt aus und es gibt keinen Wert. 
Im Zweifel den Log auf Error stellen. Das ist nur eine Info.

manuelefant, 2023-06-30:
Danke! Heißt also, ich kann evcc nicht starten, solange kein Strom vom Dach kommt?

manuelefant, 2023-06-30:
Habe auf Error gestellt... jetzt kommt nur 

'''
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/06/30 06:13:30 evcc 0.118.1
[main  ] INFO 2023/06/30 06:13:30 using config file: /config/evcc.yaml
[site  ] ERROR 2023/06/30 06:14:02 grid meter: pvnetto outdated: 10s'''

bastian1607, 2023-06-30:
Wie oft wird denn das Topic aktualisiert? Nach 10 Sekunden hast du eingestellt, dass es ein timeout gibt. Bei mir wird das Topic von HA aus alle 5 sek aktualisiert. EVCC braucht halt einen Wert, den es als aktuellen Wert nehmen kann. Und einen initialen Wert.

manuelefant, 2023-06-30:
na afaik so oft, wie neue werte kommen, also quasi permanent. 
oder kann man das irgendwo parametrieren?

bastian1607, 2023-06-30:
Nimm den timeout in der EVCC Config raus. Nachts werden selten neue Werte gepublished, ist ja in der Nacht.
Ansonsten kannst du in HA einstellen, dass einfach immer alle X Sekunden ein Wert gepublished wird. Sollte nur öfter sein als dein timeout in der EVCC Config.

manuelefant, 2023-06-30:
Hab das rausgenommen mit dem Timeout unter grid... aber bleibt bei 10s outdated

bastian1607, 2023-06-30:
Hast du EVCC schon neu gestartet? Wie stiehlt dein Workflow in HA aus?

manuelefant, 2023-06-30:
Ich kann evcc gar nicht starten, weil eben der error kommt.
Wo kann ich einstellen 

> Ansonsten kannst du in HA einstellen, dass einfach immer alle X Sekunden ein Wert gepublished wird. Sollte nur öfter sein als dein timeout in der EVCC Config.

bastian1607, 2023-06-30:
Hast du ggf einen Schreibfehler? Da steht „pvnetto“ in der Config, sollte aber „pv/netto“ sein.

Ansonsten über Automatisierung. Kannst auch ein MQTT Explorer installieren um zu schauen was wann gepublished wird.

manuelefant, 2023-06-30:
Nee. 
das passt schon. Muss nur gleich sein mit dem in node red. Und das ist es. Ging auch mit / nicht. Aber ich kann’s ja nochmal versuchen.

bastian1607, 2023-06-30:
Ich vermute die Kombi von einen höheren Timeout und "/". Ansonsten installier mal den "MQTT Explorer" in HA und schau was da wann gepublished wird auf dem Kanal.

manuelefant, 2023-06-30:
[image]

Aber ich probiere nochmal mit /

manuelefant, 2023-06-30:
Unverändert. Nur dass in der Fehlermeldung jetzt der / drin steht ^^

bastian1607, 2023-06-30:
Also du musst auch in HA auf "pv/netto" publishen. Wie gesagt, installier dir mal den MQTT Explorer und schau was ankommt. Momentan published du in ein Topic rein ohne Payload. pv[Topic] / [Trenner] netto [Payload]. In dem Payload steht der Wert eigentlich, das Topic ist nur der Kanal. Aber ich bin mir nicht sicher ob das trotzdem funktionieren kann, habe es so noch nicht gesehen. 
Deshalb Werte im Exlorer anschauen.

manuelefant, 2023-06-30:
Ja, versuche grad den Explorer zum laufen zu kriegen, bin aber auch hier anscheinend zu minderbemittelt.

bastian1607, 2023-06-30:
Das wird schon. An deiner Stelle war ich Anfang der Woche. ;-) Der Host war das Problem beim Explorer bei mir.

manuelefant, 2023-06-30:
Ich krieg den zum laufen, sehe auch, dass da im millisekundentakt werte einlaufen. die werte selbst seh ich auch nicht und es kommt ein error und ich muss neu starten

bastian1607, 2023-06-30:
Ich kenne mich mit NoteRed nicht aus. Ich habe es hiernach gemacht:
https://community.home-assistant.io/t/evcc-integration-via-mqtt-sensor-entity-config/407959/2

bzw. einer Automatisierung. Das funktioniert.

```
alias: EVCC MQTT Übergabe PV Leistung
description: sensor.pv_leistung -> aktuelle PV Leistung
trigger:
  - platform: state
    entity_id:
      - sensor.pv_leistung
  - platform: time_pattern
    seconds: /5
condition: []
action:
  - service: mqtt.publish
    data:
      topic: pv/power
      payload_template: " {{ states('sensor.pv_leistung') }} "
mode: single

```

manuelefant, 2023-06-30:
So der Fehler mit dem MQTT Explorer lag schon mal an Safari :D 
In Chrome gehts... ich sehe aber nicht, dass das PV/netto einläuft ... Mist.

manuelefant, 2023-06-30:
So. Habe aus deinem Schnipsel eine Automation gemacht, der Einfachheit halber auf "pv/netto" geändert und jetzt sehe ich, dass im MQTT Explorer die Daten einlaufen

bastian1607, 2023-06-30:
Sehr gut! Ich schreibe das mal eben zusammen. Den Kommentar kannst du als "Antwort" markieren, damit es einfacher ist für andere.

Also:
In Home Assistant wurde über MQTT der Wert nicht geplublished. Wahrscheinlich lag mit NoteRed ein Konfig-Problem vor. In HA eine Automatisierung angelegt, die 
1. Den Wert über MQTT published, wenn der Wert sich ändert
2. Alle 5 Sekunden den Wert published

Automatisierung:
```
alias: EVCC MQTT Übergabe PV Leistung
description: sensor.pv_leistung -> aktuelle PV Leistung
trigger:
  - platform: state
    entity_id:
      - sensor.pv_leistung
  - platform: time_pattern
    seconds: /5
condition: []
action:
  - service: mqtt.publish
    data:
      topic: pv/power
      payload_template: " {{ states('sensor.pv_leistung') }} "
mode: single
```

Über den MQTT Explorer konnte man nun den Payload sehen und EVCC konnte den abbonieren. Wert kommt, EVCC läuft.

manuelefant, 2023-06-30:
Jetzt hatte er voll reingepumpt, aber ich hatte auch den falschen Sensor ausgewählt. Nicht den, der den Überschuss zeigt, sondern den, der die Gesamtleistung zeigt. Das passt dann natürlich nicht. 

Wo kann ich als Antwort markieren?