evcc-io/evcc #8710: BMW offline?

TobiasHuber1980, 2023-06-30:
### Describe the bug

pi@raspberrypi:~ $ sudo evcc -l debug vehicle
[main  ] INFO 2023/06/30 08:07:28 evcc 0.118.1 (612cc8a4)
[main  ] INFO 2023/06/30 08:07:28 using config file: /etc/evcc.yaml
[db    ] INFO 2023/06/30 08:07:32 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2023/06/30 08:07:42 creating vehicle my_car_2 failed: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot get vehicles: unexpected status: 404 (Not Found)

my_car_2
--------
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot get vehicles: unexpected status: 404 (Not Found)
Capacity: 42.0kWh
Features: [Offline]

### Steps to reproduce

reboot..... 

### Configuration details

```yaml
- name: my_car_2
    type: template
    template: bmw
    title: BMW i3
    user: user@gmx.de
    password: pass
    vin: WBYxxxxxxxxxxxxx
    capacity: 42
```

### Log details

```text
pi@raspberrypi:~ $ evcc --log debug
[main  ] INFO 2023/06/30 08:53:43 evcc 0.118.1 (612cc8a4)
[main  ] INFO 2023/06/30 08:53:43 using config file: /etc/evcc.yaml
[main  ] INFO 2023/06/30 08:53:43 starting ui and api at :7070
[db    ] INFO 2023/06/30 08:53:47 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2023/06/30 08:53:51 creating vehicle my_car_2 failed: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot get vehicles: unexpected status: 404 (Not Found)
[site  ] DEBUG 2023/06/30 08:53:56 static grid tariff
[site  ] WARN 2023/06/30 08:53:56 battery configured but residualPower is missing (add residualPower: 100 to site)
[site  ] WARN 2023/06/30 08:53:57 vehicle 'BMW i3 (offline)' does not support automatic detection
[site  ] INFO 2023/06/30 08:53:57 site config:
[site  ] INFO 2023/06/30 08:53:57   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/06/30 08:53:57     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/06/30 08:53:57     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/06/30 08:53:57     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/06/30 08:53:57   vehicles:
[site  ] INFO 2023/06/30 08:53:57     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[site  ] INFO 2023/06/30 08:53:57     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/06/30 08:53:57 loadpoint 1:
[lp-1  ] INFO 2023/06/30 08:53:57   mode:        pv
[lp-1  ] INFO 2023/06/30 08:53:57   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/06/30 08:53:57   meters:      charge ✗
[site  ] DEBUG 2023/06/30 08:53:57 static grid tariff
[lp-1  ] DEBUG 2023/06/30 08:53:57 phase timer inactive
[lp-1  ] DEBUG 2023/06/30 08:53:57 pv timer inactive
[lp-1  ] DEBUG 2023/06/30 08:53:57 guard timer inactive
[site  ] DEBUG 2023/06/30 08:53:57 ----
[lp-1  ] DEBUG 2023/06/30 08:53:57 charge power: 0W
[site  ] DEBUG 2023/06/30 08:53:57 static grid tariff
[site  ] DEBUG 2023/06/30 08:53:57 pv power: 2072W
[site  ] DEBUG 2023/06/30 08:53:57 battery soc: 50%
[site  ] DEBUG 2023/06/30 08:53:57 battery power: -14W
[site  ] DEBUG 2023/06/30 08:53:57 grid power: -1856W
[site  ] DEBUG 2023/06/30 08:53:57 site power: -1870W
[lp-1  ] DEBUG 2023/06/30 08:53:57 charger status: B
[lp-1  ] INFO 2023/06/30 08:53:57 car connected
[lp-1  ] DEBUG 2023/06/30 08:53:57 vehicle api refresh
[lp-1  ] DEBUG 2023/06/30 08:53:57 pv timer elapse
[lp-1  ] DEBUG 2023/06/30 08:53:57 pv timer inactive
[lp-1  ] DEBUG 2023/06/30 08:53:57 charger: guard elapse
[lp-1  ] DEBUG 2023/06/30 08:53:57 guard timer inactive
[main  ] DEBUG 2023/06/30 08:54:00 vehicle status: A (VW ID.5)
[lp-1  ] DEBUG 2023/06/30 08:54:00 pv charge current: 2.71A = 0A + 2.71A (-1870W @ 3p)
[lp-1  ] DEBUG 2023/06/30 08:54:00 pv enable timer reset
[lp-1  ] DEBUG 2023/06/30 08:54:00 pv timer inactive
[site  ] DEBUG 2023/06/30 08:54:07 ----
```

### What type of operating system are you running?

Linux

### Version

0.118.1 (612cc8a4)

TobiasHuber1980, 2023-06-30:
Der VW ist jetzt auch offline....

```
vehicles:
  - name: my_car_1
    type: template
    template: id
    title: VW ID.5
    user: user@emailadresse.de
    password: passwortvonuser
    vin: WVGZZZxxxxxxxxxxxx
    capacity: 82
```

andig, 2023-06-30:
Funktioniert https://github.com/TA2k/ioBroker.bmw bei Euch? Falls nein bitte Issue mit Logfile erstellen.

andig, 2023-06-30:
Refs https://github.com/TA2k/ioBroker.bmw/issues/32

hsjirka, 2024-12-12:
hello,
i have the same problem. BMW i3 until October 2024 without problems and now can't create and connect

[main ] INFO 2024/12/12 20:30:14 evcc 0.131.10
[main ] INFO 2024/12/12 20:30:14 using config file: /config/evcc.yaml
[db ] INFO 2024/12/12 20:30:14 using sqlite database: /data/evcc.db
[main ] INFO 2024/12/12 20:30:15 listening at :7070
[main ] ERROR 2024/12/12 20:30:15 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': missing credentials
[site ] INFO 2024/12/12 20:30:16 site config:
[site ] INFO 2024/12/12 20:30:16 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/12/12 20:30:16 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2024/12/12 20:30:16 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/12/12 20:30:16 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/12/12 20:30:16 vehicles:
[site ] INFO 2024/12/12 20:30:16 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/12/12 20:30:16 loadpoint 1:
[lp-1 ] INFO 2024/12/12 20:30:16 mode: pv
[lp-1 ] INFO 2024/12/12 20:30:16 charger: power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/12/12 20:30:16 meters: charge ✓
[lp-1 ] INFO 2024/12/12 20:30:16 charge: power ✓ energy ✗ currents ✗
[site ] DEBUG 2024/12/12 20:30:16 set buffer soc: 50
[site ] DEBUG 2024/12/12 20:30:16 set buffer start soc: 75
[site ] DEBUG 2024/12/12 20:30:16 set priority soc: 35
[lp-1 ] DEBUG 2024/12/12 20:30:16 set smart cost limit: -0.0
[lp-1 ] DEBUG 2024/12/12 20:30:16 phase timer inactive
[lp-1 ] DEBUG 2024/12/12 20:30:16 pv timer inactive
[my_charger] TRACE 2024/12/12 20:30:16 modbus: send 00 01 00 00 00 06 01 03 00 64 00 01
[my_charger] TRACE 2024/12/12 20:30:16 modbus: recv 00 01 00 00 00 05 01 03 02 00 00
[site ] DEBUG 2024/12/12 20:30:16 ----
[lp-1 ] DEBUG 2024/12/12 20:30:16 charge power: -22W
[site ] DEBUG 2024/12/12 20:30:16 grid power: -22W
[site ] DEBUG 2024/12/12 20:30:16 pv power: 0W
[site ] DEBUG 2024/12/12 20:30:16 battery soc: 0%
[site ] DEBUG 2024/12/12 20:30:16 battery power: 549W
[site ] DEBUG 2024/12/12 20:30:16 site power: 627W
[my_charger] TRACE 2024/12/12 20:30:16 modbus: send 00 02 00 00 00 06 01 03 00 8b 00 01
[my_charger] TRACE 2024/12/12 20:30:16 modbus: recv 00 02 00 00 00 05 01 03 02 00 42
[lp-1 ] DEBUG 2024/12/12 20:30:16 charger status: B
[lp-1 ] INFO 2024/12/12 20:30:16 car connected
[lp-1 ] DEBUG 2024/12/12 20:30:16 vehicle api refresh
[lp-1 ] DEBUG 2024/12/12 20:30:16 pv timer elapse
[lp-1 ] DEBUG 2024/12/12 20:30:16 pv timer inactive
[my_charger] TRACE 2024/12/12 20:30:16 modbus: send 00 03 00 00 00 06 01 03 00 64 00 01
[my_charger] TRACE 2024/12/12 20:30:16 modbus: recv 00 03 00 00 00 05 01 03 02 00 00
[lp-1 ] DEBUG 2024/12/12 20:30:16 pv charge current: 0A = 0A + -0.909A (627W @ 3p)
[lp-1 ] DEBUG 2024/12/12 20:30:16 pv enable timer reset
[lp-1 ] DEBUG 2024/12/12 20:30:16 pv timer inactive
[site ] DEBUG 2024/12/12 20:30:46 ----
[lp-1 ] DEBUG 2024/12/12 20:30:46 charge power: -11W
[site ] DEBUG 2024/12/12 20:30:46 grid power: -11W
[site ] DEBUG 2024/12/12 20:30:46 pv power: 0W
[site ] DEBUG 2024/12/12 20:30:46 battery soc: 51%
[site ] DEBUG 2024/12/12 20:30:46 battery power: 549W
[site ] DEBUG 2024/12/12 20:30:46 site power: 538W
[my_charger] TRACE 2024/12/12 20:30:46 modbus: send 00 04 00 00 00 06 01 03 00 8b 00 01
[my_charger] TRACE 2024/12/12 20:30:46 modbus: recv 00 04 00 00 00 05 01 03 02 00 42
[lp-1 ] DEBUG 2024/12/12 20:30:46 charger status: B
[my_charger] TRACE 2024/12/12 20:30:46 modbus: send 00 05 00 00 00 06 01 03 00 64 00 01
[my_charger] TRACE 2024/12/12 20:30:46 modbus: recv 00 05 00 00 00 05 01 03 02 00 00
[lp-1 ] DEBUG 2024/12/12 20:30:46 pv charge current: 0A = 0A + -0.78A (538W @ 3p)
[site ] DEBUG 2024/12/12 20:31:16 ----
[lp-1 ] DEBUG 2024/12/12 20:31:16 charge power: 16W
[site ] DEBUG 2024/12/12 20:31:16 grid power: 16W
[site ] DEBUG 2024/12/12 20:31:16 pv power: 0W
[site ] DEBUG 2024/12/12 20:31:16 battery soc: 51%
[site ] DEBUG 2024/12/12 20:31:16 battery power: 549W
[site ] DEBUG 2024/12/12 20:31:16 site power: 565W
[my_charger] TRACE 2024/12/12 20:31:16 modbus: send 00 06 00 00 00 06 01 03 00 8b 00 01
[my_charger] TRACE 2024/12/12 20:31:16 modbus: recv 00 06 00 00 00 05 01 03 02 00 42
[lp-1 ] DEBUG 2024/12/12 20:31:16 charger status: B
[lp-1 ] DEBUG 2024/12/12 20:31:16 vehicle api refresh

andig, 2024-12-12:
There is no point in posting „me too“ after issue has been solved and released. Thanks!

hsjirka, 2024-12-12:
sorry, I'm not that experienced in this. If I understood the thread correctly, this error appeared in 2023 and was probably fixed. However, I've only had the same problem now - last month and I really don't know how to fix it. Can you please advise me on what to do? thank you