evcc-io/evcc #8720: Myporsche not connecting due to captcha addition

glebiller, 2023-06-30:
### Describe the bug

The new login mechanism in 0.118.2 seems to have fixed the previous error, however, as of today, a new error is triggered. 
It looks like a Captcha has been added to the Myporsche login page.

### Steps to reproduce

Configure a vehicle using ``porsche`` template.

### Configuration details

```yaml
vehicles:
  - name: porsche
    type: template
    template: porsche
    title: Porsche
    user: <user_email>
    password: <user_password>
```

### Log details

```text
ERROR 2023/06/30 16:24:56 creating vehicle porsche failed: cannot create vehicle 'template': cannot create vehicle 'porsche': login failed: Invalid captcha value
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.2

andig, 2023-06-30:
> It looks like a Captcha has been added to the Myporsche login page.

So isses. Und es gibt überhaupt nichts, was wir dagegen tun könnten :(

closed 2023-06-30 by andig

glebiller, 2023-06-30:
I see that the Telsa module is handling Captcha, wouldn't it be possible to do the same behavior?

andig, 2023-06-30:
Tesla is not handling the captcha either. The "real" solution would be for Porsche to provide OAuth- maybe they will one day if enough customers increase pressure...