evcc-io/evcc #8730: Car connected not detected

cburghardt, 2023-07-01:
### Describe the bug

I have the following car config:
`vehicles:
  - name: disco
    title: Discovery Sport
    type: custom
    capacity: 15
    phases: 1 
    soc:
      source: mqtt  
      topic: homeassistant/sensor/disco_sport_battery/state`

When I plug in the vehicle to my wallbox, evcc doesn't react at all, the loading point just continues to print:
Jul 01 09:25:54 evcc evcc[142]: [lp-1  ] DEBUG 2023/07/01 09:25:54 charge power: 0W

When I reboot evcc, the car is recognized and loading starts:
`Jul 01 09:28:42 evcc evcc[855]: [main  ] INFO 2023/07/01 09:28:42 evcc 0.118.1
Jul 01 09:28:42 evcc evcc[855]: [main  ] INFO 2023/07/01 09:28:42 using config file: /etc/evcc.yaml
Jul 01 09:28:42 evcc evcc[855]: [main  ] INFO 2023/07/01 09:28:42 starting ui and api at :7070
Jul 01 09:28:52 evcc evcc[855]: [db    ] INFO 2023/07/01 09:28:52 using sqlite database: /var/lib/evcc/evcc.db
Jul 01 09:28:52 evcc evcc[855]: [mqtt  ] INFO 2023/07/01 09:28:52 connecting evcc at tcp://192.168.178.109:1883
Jul 01 09:28:52 evcc evcc[855]: [site  ] INFO 2023/07/01 09:28:52 site config:
Jul 01 09:28:52 evcc evcc[855]: [site  ] INFO 2023/07/01 09:28:52   meters:      grid ✓ pv ✓ battery ✓
Jul 01 09:28:52 evcc evcc[855]: [site  ] INFO 2023/07/01 09:28:52     grid:      power ✓ energy ✗ currents ✓
Jul 01 09:28:52 evcc evcc[855]: [site  ] INFO 2023/07/01 09:28:52     pv 1:      power ✓ energy ✗ currents ✗
Jul 01 09:28:52 evcc evcc[855]: [site  ] INFO 2023/07/01 09:28:52     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Jul 01 09:28:52 evcc evcc[855]: [site  ] INFO 2023/07/01 09:28:52   vehicles:
Jul 01 09:28:52 evcc evcc[855]: [site  ] INFO 2023/07/01 09:28:52     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:28:52 loadpoint 1:
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:28:52   mode:        pv
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:28:52   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:28:52   meters:      charge ✓
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:28:52     charge:    power ✓ energy ✓ currents ✓
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:52 phase timer inactive
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:52 pv timer inactive
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:52 guard timer inactive
Jul 01 09:28:52 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:28:52 vehicle updated: unknown -> Discovery Sport
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 charge power: 0W
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 charge currents: [0 0 0]A
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 charge total import: 110.970kWh
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 charger status: B
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:28:53 car connected
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 pv timer elapse
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 pv timer inactive
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 charger: guard elapse
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 guard timer inactive
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 charge total import: 110.970kWh
Jul 01 09:28:53 evcc evcc[855]: [main  ] INFO 2023/07/01 09:28:53 new version available: 0.118.2
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 vehicle soc: 63%
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 pv charge current: 16.1A = 0A + 16.1A (-3713W @ 1p)
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 site power -3713W <= 0W enable threshold
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 pv enable in 0s
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 pv enable timer elapsed
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 max charge current: 6A
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 charger enable
Jul 01 09:28:53 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:28:53 wake-up timer: start
Jul 01 09:29:03 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:29:03 charge power: 1294W
Jul 01 09:29:03 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:29:03 charge currents: [5.85 0 0]A
Jul 01 09:29:03 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:29:03 charge total import: 110.970kWh
Jul 01 09:29:03 evcc evcc[855]: [lp-1  ] DEBUG 2023/07/01 09:29:03 charger status: C
Jul 01 09:29:03 evcc evcc[855]: [lp-1  ] INFO 2023/07/01 09:29:03 start charging ->`

### Steps to reproduce

See above

### Configuration details

```yaml
chargers:
  - name: wallbox
    type: template
    template: go-e-v3
    host: go-echarger-217527

site:
  title: Home
  bufferSoc: 80
  prioritySoc: 50
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 100
  meters:
    pv: pv
    grid: grid
    battery: battery  

loadpoints:
  - title: Garage
    charger: wallbox
    mode: pv
    mincurrent: 6
    maxcurrent: 32
    resetOnDisconnect: true
    vehicle: disco  

meters:
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/power_solar_generation/state
      timeout: 40s  
  - name: grid      
    type: custom
    power:  
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_m1_ac_power/state
      scale: -1
      timeout: 40s  
    currents:
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_a/state
        timeout: 40s  
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_b/state
        timeout: 40s  
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_c/state
        timeout: 40s  
  - name: battery
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_b1_dc_power/state
      scale: -1
      timeout: 4h 
    soc:
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_b1_state_of_energy/state
      timeout: 4h
```

### Log details

```text
See above
```

### What type of operating system are you running?

Linux

### Version

0.118.1

cburghardt, 2023-07-01:
One more observation: during the load the soc is not updated:
`mosquitto_sub -h docker -t homeassistant/sensor/disco_sport_battery/state -v
homeassistant/sensor/disco_sport_battery/state 92`
[image]

andig, 2023-07-01:
Happy to look at this but need readable log.

closed 2023-07-01 by andig

cburghardt, 2023-07-01:
[log.txt](https://github.com/evcc-io/evcc/files/11926533/log.txt)

cburghardt, 2023-07-01:
@andig let me know if the above is sufficient. Why is the issue closed?

andig, 2023-07-01:
Der Fahrzeug SoC hat absolut nichts mit dem Ladebeginn zu tun. Es ist völlig unklar, was hier die Frage ist oder wo das im Logfile zu sehen ist. 

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

cburghardt, 2023-07-01:
Wenn der soc nicht wichtig ist, welche Information fehlt dann um das Laden zu starten? Laut Doku wird nur der soc bei einem vehicle benötigt. Falls noch etwas anderes benötigt wird, wäre das wichtig zu wissen. Die Fahrzeug Daten kommen per mqtt.