evcc-io/evcc #8736: KIA Bluelink Car not

DottoreAltobelli, 2023-07-01:
### Describe the bug

Testing the configuration of Kia Bluelink ...
   Error: cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request)

Car is detcted and linked and visible via KIA Connect APP.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
evcc --log debug
[main  ] INFO 2023/07/01 14:31:28 evcc 0.118.2
[main  ] INFO 2023/07/01 14:31:28 using config file: /home/pi/evcc.yaml
[main  ] INFO 2023/07/01 14:31:28 starting ui and api at :7070
[main  ] FATAL 2023/07/01 14:31:28 While parsing config: yaml: line 76: found character that cannot start any token
[main  ] FATAL 2023/07/01 14:31:28 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.118.02

andig, 2023-07-01:
Duplikat: https://github.com/evcc-io/evcc/issues/8696

closed 2023-07-01 by andig

DottoreAltobelli, 2023-07-03:
Car still not reachable via evcc. Login not possible.ChearsVon meinem iPhone gesendet