evcc-io/evcc #8740: Tasmota Steckdose On Off ohne Funktion

rrdd, 2023-07-01:
### Describe the bug

Ich nutze eine tasmota Nous Steckdose. Ich habe das template aus https://docs.evcc.io/docs/devices/chargers/#schaltbare-steckdosen verwendet. 
Ich habe mode off eingestellt. Wenn ich evcc starte, wird die Steckdose einfach aktiviert. Im log steht dann:
charger out of sync: expected disabled, got enabled
charger status: B

Das ändern des Status in der Weboberfläche wird komplett ignoriert. 

### Steps to reproduce

1. Tasmota in die Steckdose und konfigurieren
2. evcc neu starten mod off
3.  Steckdose per toggle in der weboberfläche abschalten
4. evcc schaltet die steckdose an

### Configuration details

```yaml
meters:

  - name: tasmota1
    type: template
    usage: grid
    template: tasmota
    host: 172.16.6.49

chargers:
  - name: tasmotapower1
    type: template
    template: tasmota
    host: 172.16.6.49
    channel: 1 # Nummer des Schaltkanals (1-8), bei Geräten mit mehr als einem Schalter
    standbypower: 15
    #integrateddevice: false
```

### Log details

```text
charger out of sync: expected disabled, got enabled
charger status: B
```

### What type of operating system are you running?

Linux

### Version

0.118.2

VolkerK62, 2023-07-01:
Die Steckdose ist gleichzeitig auch als Grid-Meter konfiguriert?

rrdd, 2023-07-01:
Danke für den Hinweise, habe die Teile usage: grid entfernt.

andig, 2023-07-01:
/cc @thierolm

thierolm, 2023-07-01:
@rrdd also entweder nutzt du die Dose als `charger` oder als `meter` ... ==> Lösche die Meter Definition
Kannst du bitte mal ein Trace log mitschreiben und teilen, wenn die von dir genannte Fehlermeldung kommt?

rrdd, 2023-07-01:
ich habe das meter entfernt. problem noch so vorhanden. hier das trace:

Jul  1 21:30:16 s8 evcc[18270]: [tasmota] TRACE 2023/07/01 21:30:16 GET http://172.16.6.49/cm?cmnd=Status+8&password=&user=
Jul  1 21:30:17 s8 evcc[18270]: [tasmota] TRACE 2023/07/01 21:30:17 {"StatusSNS":{"Time":"2023-07-01T20:30:17","ENERGY":{"TotalStartTime":"2023-07-01T14:23:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
Jul  1 21:30:17 s8 evcc[18270]: [lp-1  ] DEBUG 2023/07/01 21:30:17 charger status: B
Jul  1 21:30:17 s8 evcc[18270]: [tasmota] TRACE 2023/07/01 21:30:17 GET http://172.16.6.49/cm?cmnd=Status+0&password=&user=
Jul  1 21:30:17 s8 evcc[18270]: [tasmota] TRACE 2023/07/01 21:30:17 {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota1"],"Topic":"tasmota_17E4ES","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T04:35:15","StartupUTC":"2023-07-01T14:55:02","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2023-07-01T14:23:46","SaveCount":75,"SaveAddress":"F9000"},"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-09-01T10:06:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"331/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["12a",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1460C4","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"tasmota-17E4ES-1258","IPAddress":"172.16.6.49","Gateway":"172.16.4.1","Subnetmask":"255.255.255.0","DNSServer1":"172.16.4.8","DNSServer2":"0.0.0.0","Mac":"ZZZZ","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_17E4ES","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-07-01T19:30:17","Local":"2023-07-01T20:30:17","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:56"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2023-07-01T20:30:17","ENERGY":{"TotalStartTime":"2023-07-01T14:23:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}},"StatusSTS":{"Time":"2023-07-01T20:30:17","Uptime":"0T04:35:15","UptimeSec":16515,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"123a","BSSId":"ZZZZ","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:06"}}}
Jul  1 21:30:17 s8 evcc[18270]: [lp-1  ] WARN 2023/07/01 21:30:17 charger out of sync: expected enabled, got disabled
Jul  1 21:30:17 s8 evcc[18270]: [tasmota] TRACE 2023/07/01 21:30:17 GET http://172.16.6.49/cm?cmnd=Power1+on&password=&user=
Jul  1 21:30:17 s8 evcc[18270]: [tasmota] TRACE 2023/07/01 21:30:17 {"POWER":"ON"}

thierolm, 2023-07-02:
Der Status wird aus der STSResponse korrekt ausgelesen. Das Log ist etwas kurz, um weitere Schlüsse zu ziehen. Kannst du die Dose über die UI mal einschalten, 30 Sekunden warten, dann ausschalten, 30 Sekunden warten und dann einschalten ... und das Log dann teilen?

Eine weiter Frage: Ist auch sichergestellt, dass die App oder wie auch immer die Nous Plus sonst gesteuert wird nicht dazuwischen funkt und `evcc` exklusiv steuert?

rrdd, 2023-07-02:
Ich habe in der Dose jetzt noch den PowerOnState auf 0 also OFF gesetzt. Das Verhalten hat sich aber nicht geändert. Ich konfiguriere und starte evcc. Stecke dann die Dose in die Steckdose. Sie ist erstmal ausgeschalten (POWERONSTATE 0). Dann greift evcc ein und aktiviert trotz mode: off die Dose. Wenn ich die Dose in evcc auskommentiere, zeigt sie das Verhalten nicht. Es wird also nicht durch andere Software beeinflusst.

Die Funktion An oder Schnell im Frontend schaltet die Dose nicht.

Der Log Eintrag und das anschalten der Dose erfolgt unmittelbar nach dem Check des Status der Dose. Die Message wird nur einmalig hier im Code genutzt: https://github.com/evcc-io/evcc/blob/f96a4266f61591d109101da1a1814d59228b601f/core/loadpoint.go#L639 . 

Was mir nur auffällt, ist dass ihr für StatusSTSResponse  via StatusSTS Power prüft, im JSON StatusSTS  aber POWER (UPPERCASE) steht. Im json Status am Anfang des Trace steht die variable Power.

Trace
Jul  2 18:35:49 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 18:35:49 GET http://172.16.6.49/cm?cmnd=Status+0&password=&user=
Jul  2 18:35:52 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 18:35:52 {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota1"],"Topic":"tasmota_17E4EA","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:00:04","StartupUTC":"2023-07-02T16:35:45","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2023-07-01T14:23:46","SaveCount":98,"SaveAddress":"FA000"},"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-09-01T10:06:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"331/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZZZ",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1460C4","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"tasmota-17E4ED-1258","IPAddress":"172.16.6.49","Gateway":"172.16.6.1","Subnetmask":"255.255.255.0","DNSServer1":"172.16.6.8","DNSServer2":"0.0.0.0","Mac":"2341234","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusTIM":{"UTC":"2023-07-02T16:35:49","Local":"2023-07-02T17:35:49","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2023-07-02T17:35:49","ENERGY":{"TotalStartTime":"2023-07-01T14:23:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}},"StatusSTS":{"Time":"2023-07-02T17:35:49","Uptime":"0T00:00:04","UptimeSec":4,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":40,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ZZZ","BSSId":"7C:FF:4D:15:EF:05","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}

Hier noch der Status, wenn die Dose durch evcc aktiviert wurde:
Jul  2 18:47:27 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 18:47:27 {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota1"],"Topic":"tasmota_17E4ED","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:09:04","StartupUTC":"2023-07-02T16:38:21","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2023-07-01T14:23:46","SaveCount":107,"SaveAddress":"F9000"},"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-09-01T10:06:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"331/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZZZZ",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1460C4","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"tasmota-17E4EA-1258","IPAddress":"172.16.6.149","Gateway":"172. [... cut ...] 000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage":301,"Current":0.000}},"StatusSTS":{"Time":"2023-07-02T17:47:25","Uptime":"0T00:09:04","UptimeSec":544,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"ZZZZ","BSSId":"7C:FF:4D:15:EF:05","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}

Im Vergleich ist Status-> Power 0 bzw 1.
StatusSTS POWER einmal ON einmal OFF

andig, 2023-07-02:
> Dann greift evcc ein und aktiviert trotz mode: off die Dose

Mal völlig losgelöst von der Tasmotaimplementierung. Gibts dafür einen Beleg? Logfile?

rrdd, 2023-07-02:
Im log oben ist der Aufruf zu finden, der die Dose aktiviert:
Jul 1 21:30:17 s8 evcc[18270]: [lp-1 ] WARN 2023/07/01 21:30:17 charger out of sync: expected enabled, got disabled
Jul 1 21:30:17 s8 evcc[18270]: [tasmota] TRACE 2023/07/01 21:30:17 GET http://172.16.6.49/cm?cmnd=Power1+on&password=&user=

"Power 1 on"

andig, 2023-07-02:
Da wurde enabled schon erwartet. Warum? Nicht ersichtlich… Logschnipselchen sind völlig nutzlos um ein potenziell komplexes Fehlerbild zu verstehen.

thierolm, 2023-07-02:
@rrdd Bin bei @andig . @rrdd Du lieferst nur Snapshots deines Verhaltens (du bist nicht er Einzige evv Tasmota User ...). Wo bleibt dein evcc Log, wie  oben beschrieben?

rrdd, 2023-07-02:
Hier ein längerer Log. Ist das ausreichend so?

Jul  2 21:13:39 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:39 GET http://172.16.6.49/cm?cmnd=Status+8&password=&user=
Jul  2 21:13:44 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:44 GET http://172.16.6.49/cm?cmnd=Status+8&password=&user=
Jul  2 21:13:44 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:44 {"StatusSNS":{"Time":"2023-07-02T20:13:44","ENERGY":{"TotalStartTime":"2023-07-01T14:23:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
Jul  2 21:13:44 s8 evcc[29238]: [lp-1  ] DEBUG 2023/07/02 21:13:44 charge power: 0W
Jul  2 21:13:44 s8 evcc[29238]: [site  ] DEBUG 2023/07/02 21:13:44 static grid tariff
Jul  2 21:13:45 s8 evcc[29238]: [script] DEBUG 2023/07/02 21:13:45 /home/pi/solarmanpv/get.sh: 0
Jul  2 21:13:45 s8 evcc[29238]: [site  ] DEBUG 2023/07/02 21:13:45 pv power: 0W
Jul  2 21:13:45 s8 evcc[29238]: [http  ] TRACE 2023/07/02 21:13:45 GET http://172.16.6.100/cm?cmnd=Status%208
Jul  2 21:13:45 s8 evcc[29238]: [http  ] TRACE 2023/07/02 21:13:45 {"StatusSNS":{"Time":"2023-07-02T20:13:45","DWS7420":{"zählerstand_total":120399.9,"wirkenergie_total":253961.8,"aktuelle_wirkleistung":314.49}}}
Jul  2 21:13:45 s8 evcc[29238]: [site  ] DEBUG 2023/07/02 21:13:45 grid power: 314W
Jul  2 21:13:45 s8 evcc[29238]: [site  ] DEBUG 2023/07/02 21:13:45 site power: 314W
Jul  2 21:13:45 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:45 GET http://172.16.6.49/cm?cmnd=Status+8&password=&user=
Jul  2 21:13:45 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:45 {"StatusSNS":{"Time":"2023-07-02T20:13:45","ENERGY":{"TotalStartTime":"2023-07-01T14:23:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
Jul  2 21:13:45 s8 evcc[29238]: [lp-1  ] DEBUG 2023/07/02 21:13:45 charge total import: 0.004kWh
Jul  2 21:13:45 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:45 GET http://172.16.6.49/cm?cmnd=Status+8&password=&user=
Jul  2 21:13:45 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:45 {"StatusSNS":{"Time":"2023-07-02T20:13:45","ENERGY":{"TotalStartTime":"2023-07-01T14:23:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
Jul  2 21:13:45 s8 evcc[29238]: [lp-1  ] DEBUG 2023/07/02 21:13:45 charger status: B
Jul  2 21:13:45 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:45 GET http://172.16.6.49/cm?cmnd=Status+0&password=&user=
Jul  2 21:13:48 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:48 {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota1"],"Topic":"tasmota_17E4ED","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:00:05","StartupUTC":"2023-07-02T19:13:40","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2023-07-01T14:23:46","SaveCount":122,"SaveAddress":"FA000"},"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-09-01T10:06:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"331/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZZZ",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1460C4","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"tasmota-17E4EA-1258","IPAddress":"172.16.6.49","Gateway":"172.16.6.1","Subnetmask":"255.255.255.0","DNSServer1":"172.16.6.8","DNSServer2":"0.0.0.0","Mac":"123","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusTIM":{"UTC":"2023-07-02T19:13:45","Local":"2023-07-02T20:13:45","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2023-07-02T20:13:45","ENERGY":{"TotalStartTime":"2023-07-01T14:23:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}},"StatusSTS":{"Time":"2023-07-02T20:13:45","Uptime":"0T00:00:05","UptimeSec":5,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ZZZ","BSSId":"7C:FF:4D:15:EF:05","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
Jul  2 21:13:48 s8 evcc[29238]: [lp-1  ] WARN 2023/07/02 21:13:48 charger out of sync: expected enabled, got disabled
[... cut ...]
Jul  2 21:13:50 s8 evcc[29238]: [http  ] TRACE 2023/07/02 21:13:50 {"StatusSNS":{"Time":"2023-07-02T20:13:49","DWS7420":{"zählerstand_total":120400.4,"wirkenergie_total":253961.8,"aktuelle_wirkleistung":318.01}}}
Jul  2 21:13:50 s8 evcc[29238]: [site  ] DEBUG 2023/07/02 21:13:50 grid power: 318W
Jul  2 21:13:50 s8 evcc[29238]: [site  ] DEBUG 2023/07/02 21:13:50 site power: 318W
Jul  2 21:13:50 s8 evcc[29238]: [tasmota] TRACE 2023/07/02 21:13:50 GET http://172.16.6.49/cm?cmnd=Status+8&password=&user=

rrdd, 2023-07-02:
ich habe ganz oben noch die loadpoints yaml ergänzt. die fehlte leider.

andig, 2023-07-03:
Bitte Log formatieren, danke!

andig, 2023-07-03:
Ich mache mal zu bis wir ein formatiertes Log haben.

thierolm, 2023-07-03:
Ich schaue es mir heute Abend mal an ...

thierolm, 2023-07-03:
Hi @rrdd , ich hab' mir das Log mal angeschaut. Irgend etwas, was nichts mit evcc zu hat schaltet deinen Tasmota Dose aus ...

Im Log kann man nachvollziehen, dass die Dose im eingeschalteten Zustand (`"POWER":"ON"`) um 09:56:30 ins Rennen geht, nachdem du evcc gestartet hattest.
Du kannst mal nach `POWER` (gross geschrieben) im Log suchen, dann findest du alle relevanten Status-Ausgaben ...
Um 09:56:31, gibt es nochmals eine Status-Ausgabe, die das nochmals bestätigt.
Dann wir es um 09:56:40 komisch. Nachdem evcc eine paar Status 8 Abfragen geschickt hat (d.h. evcc hat KEINEN Aus-Schaltbefehl an die Dose geschickt, liefert die abschliessende Status 0 Abfrage  `"POWER":"ON"`

thierolm, 2023-07-03:
Um zu zeigen , dass WAS passiert? Langsam verwirrst du mich ...

Wenn du hier extern ein und ausschaltest ist die Meldung vollkommen normal, siehe meine Beschreibung oben.

thierolm, 2023-07-03:
Die Dose ging im eingeschalteten Zustand in Rennen, siehe erste Status Ausgabe um 09:56:30!

thierolm, 2023-07-03:
> Die Dose ist Mode Off eingestellt
Was meinst du damit? In evcc, in Tasmota ...???

thierolm, 2023-07-03:
Setze auch mal `mincurrent`auf 6 und `maxcurrent` auf 7