evcc-io/evcc #8744: Preisgrenze "keine" wird als <0 cent interpretiert

blenni, 2023-07-01:
### Describe the bug

Tibber Stromtarif, morgen sind die Preise sehr günstig und gehen tlw. bis -40cent.
Bei Smartes Netzladen ist als Preisgrenze "keine" eingestellt, allerdings werden trotzdem die Zeitfenster, in denen der Strompreis negativ ist, ausgewählt.

[image]

Nachteil in dem Fall: Auto ist bereits so voll, dass es in ca. 1h vollgeladen werden kann.
Jetzt würde aber das Zeitfenster gewählt werden, in dem der Strompreis bei 0 c liegt, und nicht, wenn er bei -40c liegt.

### Steps to reproduce

1. Negative Strompreise
2. Preisgrenze auf "keine" stellen

### Configuration details

```yaml
:
  currency: EUR
  grid:
    type: tibber
    token: "***"
    homeid: "***"
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.2

andig, 2023-07-02:
Wäre das denn schlecht bei negativen Preisen zu laden? /cc @naltatis

naltatis, 2023-07-02:
> Nachteil in dem Fall: Auto ist bereits so voll, dass es in ca. 1h vollgeladen werden kann.

Heißt du würdest dir wünschen, dass man auch noch kleinere Limits als "< 1 ct" auswählen kann, richtig?

@andig Uns fällt hier die Logik "`smartCostLimit == 0` bedeutet `kein Limit`" auf die Füsse. Wir sollten 0 (und negative Werte) als valide Optionen mit aufnehmen und das entfernen des Limits über nen `DELETE` Aufruf lösen. Wenn du keine Einwände hast kümmerst ich mich um den Umbau.

andig, 2023-07-02:
In der yaml (falls es das noch gibt) wär das dann ein Pointer. Ich wäre aber generell dafür, Yaml-EInträge für UI-konfigurierbare Werte zu entfernen und eine Deprecation auszugeben. Hier würde ich gerne noch verstehen, was gegen Lade bei negativen preisen überhaupt spricht?

steve0564, 2023-07-02:
Es spricht grundsätzlich natürlich nichts gegen das Laden bei negativen Preisen. Wenn es aber wie heute gleich mal mehrere Stunden sind, möchte man natürlich sein Auto hauptsächlich dann laden, wenn die Negativpreise am höchsten sind, da dir die Beträge ja damit gut geschrieben werden. Man will dann sein Ladeverhalten möglichst auf genau diese Stunden legen.

blenni, 2023-07-02:
Sorry war unterwegs. Genau, wie @steve0564 schreibt, mein Auto wäre dann zu einem Preis von -0,037 Cent vollgeladen worden statt zu einem Preis von -40 Cent.
Klar wäre das nicht schlimm, aber 2 oder 3€ hätte ich dadurch dann "verloren".

andig, 2023-07-02:
Jau, verstanden, danke 🙏

foto-andreas, 2023-07-04:
Ich weiß, ich bin unverschämt, aber könnte man die Balken auch für negative Preise anzeigen? Getrenntes Ticket aufmachen?

naltatis, 2023-07-05:
@foto-andreas Im Zusammenhang mit diesem Issue würde ich auch die Balkendarstellung anpassen. Aktuell sind die min=0 basiert, was ich bis zum letzten Wochenende auch sinnvoll fand. Vermutlich würd ich das so umstellen, dass das Minimum von 0 weiterhin als default bleibt. Sollten aber negative Werte in den Preisdaten sein würde ich die Schwelle entsprechend absenken.

Ist das das was du meinst?

negesti, 2023-07-07:
Ich würde es auch schön finden, wenn man auch die Netznutzungsgebühr einstellen kann. Bei einem statischen Tarif kann man sich das ja ausrechnen und entsprechen konfigurieren. Ist man aber Tibber oder aWATTar Kunde, wird den Strompreis automatisch ermittelt und die 3-6 Cent Netznutzungsgebühr werden nicht berücksichtigt. 

Zumindest in Ö wird das bei jedem Netzbetreiber fällig und ist bei Einspeisung nicht zu berücksichtigen

@naltatis  die negativen Werte < -50 Cent von letzte Wochen waren ihmo ein Bug. Der Wert war nur eine Stunde so niedrig und danach wieder im normalen Bereich. Laut meinem Grafana Dashboard (leider erst seit 22.4) gabe es davor nur 3 mal Preise < -5 Cent.

naltatis, 2023-07-07:
@negesti glaube das können wir bereits haben es aber bislang noch nicht in der Doku stehen. Schau mal hier:

https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml#L175

@VolkerK62 magst du das für die Doku aufnehmen?

negesti, 2023-07-07:
@naltatis perfekt danke
@VolkerK62 charges funktioniert bei mir lokal, elering und energinet find ich im verlinkten yaml aber nicht in der Doku. Octopusenergy steht in der doku aber nicht im yaml hab mich aber auch nicht "weiter umgsehen"
[Charges + taxes](https://github.com/evcc-io/docs/pull/408)

VolkerK62, 2023-07-08:
@naltatis  das ist komplett an mir vorbei gegangen .... es sollte da einen Weg geben, dass zumindest ein Issue für die Doku aufgemacht wird

steve0564, 2023-07-08:
Die Nebenkosten inkl. Netznutzung sind in BRD leider sehr viel teurer und nicht einheitlich.

andig, 2023-07-09:
> @naltatis das ist komplett an mir vorbei gegangen .... es sollte da einen Weg geben, dass zumindest ein Issue für die Doku aufgemacht wird

@VolkerK62 entschuldige, das geht auf mein Konto. Der PR dazu war https://github.com/evcc-io/evcc/pull/8088.

foto-andreas, 2023-07-12:
> die negativen Werte < -50 Cent von letzte Wochen waren ihmo ein Bug.

Dann war das ein Bug an der Börse. Andere Apps lieferten die gleichen Werte. Und ich hab bei Tibber mit Laden, waschen, Spülmaschine 4€ verdient. Vorschlag mit default = 0 und optional negativem Bereich schadet ja nicht.

tobe4it, 2023-07-12:
> Die Nebenkosten inkl. Netznutzung sind in BRD leider sehr viel teurer und nicht einheitlich.

EVCC zieht die Bruttopreise von Tibber. Deswegen werden ja access token und homeid in die Konfiguration eingetragen. Für die reinen Börsenpreise (netto) wäre kein Zugriff auf die Anschlussdaten erforderlich.

andig, 2023-07-12:
Wenn dann muss der Default "kein Laden" sein. Die Beschwerde war ja, dass 0€ schlechter war als -50ct.

naltatis, 2023-07-12:
> Wenn dann muss der Default "kein Laden" sein. Die Beschwerde war ja, dass 0€ schlechter war als -50ct.

Ja, das was ich oben geschrieben habe. Das Limit muss aktiverer/deaktivierbar sein.

Das was @foto-andreas mit default=0 meint ist die Untergrenze der grafischen Balken. Da gehts um die optische Darstellung, nicht um die Mechanik.

PR #9233 (2023-08-05): UI: Support negative prices and smart cost limit
fixes #8744

- ensure a minimum bar height even for negative prices
- show negative price filter options depending on the lowest price

Note: `0` currently has a special meaning (no limit). That's why `0` can't be selected as a limit.

[image]

[image]
closed 2023-08-05 by naltatis