evcc-io/evcc #8748: Vehicle (SEAT) offline - 401 (Unauthorized)

KalkiDE, 2023-07-02:
### Describe the bug

Since two days, my car is shown as offline in EVCC. The situation is different than the well known server issues with VW.
- The car itself is online (the globe is white)
- The App is fully working and shows the live status of the car.

Looking to the logfile there seems to be an authorization issue - I'm afraid that VW changed the login procedure:
[main  ] ERROR 2023/07/02 08:18:13 creating vehicle vehicle_leon failed: cannot create vehicle 'seat': unexpected status: 401 (Unauthorized)

Looks like it occurs more often:
#8741 SOC Cupra
#8712 alle Fahrzeuge (offline)
The last one also includes some traces.

### Steps to reproduce

1. Reboot
2. Changed vehicle name

### Configuration details

```yaml
vehicles:
  - name: vehicle_leon
    title: SEAT Leon Sportstourer
    type: seat
    user: xxx@xxx.xxx
    password: xxx
    capacity: 10.4
    phases: 1
    onidentify:
      mode: pv
      minSoc: 30
      targetSoc: 80
```

### Log details

```text
[main  ] ERROR 2023/07/02 08:18:13 creating vehicle vehicle_leon failed: cannot create vehicle 'seat': unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

Docker container

### Version

0.118.2

gr5604, 2023-07-02:
I have the same problem with a Seat Mii electric

Operating System: Linux raspberrypi 5.10.103-v7l

EVCC version: 0.118.2

steps to reproduce: execute command 'evcc vehicle' 

result:
{"description":"IdentityKit Response: 401 : {\"error\":\"invalid_client\",\"error_description\":\"Client authentication failed\"}","errorCode":"IDENTITY_KIT_RESPONSE_ERROR"}
[main  ] ERROR 2023/07/02 09:22:30 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'seat': unexpected status: 401 (Unauthorized)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'seat': unexpected status: 401 (Unauthorized)

Additinal test with template 'cupra':

according to andig's idea in [https://github.com/evcc-io/evcc/discussions/8712]
i tried to test my car with template 'cupra'. With this template there was a successfull authentication, but the vehicle wasn't found ( for me no suprise because it's not a 'cupra' )

result:
[cupra ] TRACE 2023/07/02 09:23:15 GET https://identity-userinfo.vwgroup.io/oidc/userinfo
[cupra ] TRACE 2023/07/02 09:23:16 {"sub":"3266f51a-c857-40ba-be87-c2238316f4f7","name":"MY_NAME","given_name":"MY_GIVEN_NAME","family_name":"MY_FAMILY_NAME","email":"***","email_verified":true,"updated_at":1688247679,"picture":"https://customer-pictures.vwgroup.io/v1/3266f51a-c857-40ba-be87-c2238316f4f7/profile-picture"}
[cupra ] TRACE 2023/07/02 09:23:16 GET https://ola.prod.code.seat.cloud.vwgroup.com/v1/users/3266f51a-c857-40ba-be87-c2238316f4f7/garage/vehicles
[cupra ] TRACE 2023/07/02 09:23:17 {"vehicles":[]}
[main  ] ERROR 2023/07/02 09:23:17 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'cupra': cannot find vehicle: MY_VIN

I added tracefile with template 'seat' and template 'cupra'

Tracefiles of tests for both template and configuration file 'evcc.yaml' are added

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/11929064/evcc.yaml.txt)
[seat.txt](https://github.com/evcc-io/evcc/files/11929066/seat.txt)
[cupra.txt](https://github.com/evcc-io/evcc/files/11929067/cupra.txt)

andig, 2023-07-02:
Geht `cupra` statt `seat`?

gr5604, 2023-07-02:
Leider nein, siehe tracefile cupra.txtBei cupra klappt zwar der Login, aber er findet kein Fahrzeug.Gruß Gerhard --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

SandnerSoft, 2023-07-02:
Ich habe das gleiche Problem!
Aktuell leider kein Login möglich

andig, 2023-07-02:
Könnt ihr bitte mal iobroker siehe oben probieren? Geht das?

andig, 2023-07-02:
> Looks like it occurs more often:

It doesn‘t unless it is the same error message. A timeout isn‘t…

SandnerSoft, 2023-07-02:
> Könnt ihr bitte mal iobroker siehe oben probieren? Geht das?

ja das geht bei mir über vw-connect -> Profil "Seat Cupra 2"

Bin zurück zur 0.118.0, das war die letzte wo es ging, aber scheinbar scheint es ein allgemeines Problem beim Login zu sein

andig, 2023-07-02:
Also iobroker cupra geht nicht, evcc cupra geht nicht, iobroker cupra2 geht, richtig?

blenni, 2023-07-02:
So wie ich es verstanden habe ja.

https://github.com/Farfar/seatconnect
Hier hat's scheinbar auch schon jemand gefixt, aber der PR ist noch nicht online.

gr5604, 2023-07-02:
Ja genau so verhält es sich. Nur cupra2 funktioniert.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

SandnerSoft, 2023-07-03:
> Also iobroker cupra geht nicht, evcc cupra geht nicht, iobroker cupra2 geht, richtig?

Ja genau so ist es aktuell.

Vielleicht sollte in dem Zuge die Dokumentation von EVCC angepasst werden, damit das mit Seat und Cupra nicht so verwirrend ist.
Aktuell muss man mehr oder weniger selber testen, wo sein Fahrzeug erkannt wird.

Richtig wäre:
was jetzt Seat ist in "Seat/Cupra Gen 3" umzubenennen und was aktuell Cupra ist in "Cupra Gen 4".

Welche Fahrzeuge darunter fallen, sieht man hier recht gut:
[https://www.cupraofficial.de/services/connect/overlay-gen3](url)
[https://www.cupraofficial.de/services/connect/overlay-gen4](url)

andig, 2023-07-03:
Danke, das kannte ich nicht. Wo würde da deiner Meinung nach cupra2 rein passen? Bei ioroker sind es ja 3 Kategorien?

gr5604, 2023-07-03:
Ich würde 'seat' / 'cupra 2' bei gen3 sehen und den Cupra Born bei gen4. Vielleicht kann jemand einen Cupra Born mit Iobroker testen. Wenn 'cupra' funktioniert und 'cupra 2' nicht, dann würde das die Vermutung bestätigen.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

andig, 2023-07-03:
Könnte mir jemand einen Seat/Cupra Gen3 an info@evcc.io schicken?

Matze03, 2023-07-03:
Nehme an, du meinst Zugangsdaten?

andig, 2023-07-03:
Genau!

andig, 2023-07-03:
@SandnerSoft das ist jetzt ein bisschen merkwürdig. Das ioborker `seatcupra2` entspricht eigentlich dem evcc `cupra`. Die alte `seat` Integration hat iobroker zugunsten `seatcupra2` umgebogen. Das sieht ähnlich aus wie wir von `audi` auf `etron` umsteigen mussten.
Jetzt gehts erst weiter wenn ich ein Testobjekt habe...

Matze03, 2023-07-03:
Hätte da welche, würde die aber ungern aus der Hand geben, da nicht meine. Ist ein Seat Mii electric....
Kannst Du dich bei mir aufschalten, bringt das was?

andig, 2023-07-03:
Nope, das kostet zuviel Zeit, sorry.

closed 2023-07-03 by andig

Matze03, 2023-07-03:
Reicht Dir ein Mitbenutzer?

andig, 2023-07-03:
Fix in 15min im nightly. Mitbenutzer geht normalerweise, hab sich aber (für dieses mal) erledigt :)

gr5604, 2023-07-03:
Gerade getestet. Funktioniert. Besten Dank für die schnelle Korrektur.Lasse das Passwort noch bis zu nächsten Release , falls doch noch jemandem was auffällt. Falls ich wieder einmal helfen kann. Gerne melden.GrußGerhard --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

andig, 2023-07-03:
Danke @gr5604. @Matze03 könntest Du bitte auch mal den Mii testen?

Matze03, 2023-07-03:
yes, perfekt und Danke!!!!

Kaeferfahrer, 2023-07-03:
moin

habe seit ca2 Tagen das gleiche Problem.
bin mir nur nicht sicher was ich jetzt ändern muß damit es wieder funktioniert. 
habe die 0.118.2 version 
Danke

KalkiDE, 2023-07-03:
Bestätigung ebenfalls von meiner Seite - unser Leon ist wieder online.
Vielen Dank für den schnellen Fix! Ich habe jetzt aktuell die 0.118.5 laufen (ein paar Versionssprünge heute - Respekt!)

andig, 2023-07-18:
Zuletzt vmtl. per https://github.com/evcc-io/evcc/pull/9009 gelöst.