evcc-io/evcc #8759: Easee API access geht net

olli0815, 2023-07-02:
### Describe the bug

Hallo,

die configuration sollte richtig sein ich kann mich auch via https://easee.cloud mit meinen Daten anmelden, bei Start von evcc bekomme ich allerdings:

cannot create charger 'hans'
cannot create charger 'template'
cannot create charger 'easee'
Post "https://api.easee.com/api/accounts/login"
dial tcp
lookup api.easee.com: i/o timeout

### Steps to reproduce

1.
2.
3.
...
configure easy boy and try to start

### Configuration details

```yaml
chargers:
 - name: hans
   type: template
   template: easee
   user: mail
   password: pw
   charger: EH_charger
```

### Log details

```text
......
[main  ] FATAL 2023/07/02 17:32:43 cannot create charger 'hans': cannot create charger 'template': cannot create charger 'easee': Post "https://api.easee.com/api/accounts/login": dial tcp: lookup api.easee.com: i/o timeout
[main  ] FATAL 2023/07/02 17:32:43 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.118.2

andig, 2023-07-02:
Serverproblem

closed 2023-07-02 by andig

olli0815, 2023-07-03:
okay, sporadisch oder "ausnahmsweise"?

andig, 2023-07-03:
Die Frage musst du dem Easee Support stellen. Timeouts sind immer ein Netzwerk- oder Serverthema.

olli0815, 2023-07-03:
Serivs sagt: If you want to ping the api, you can do it via: curl https://api.easee.com/health, Service is up and running

**´ping geht nicht:**
root@evcc:~# ping api.easee.com
PING d-wvxoekiiy7.execute-api.eu-west-1.amazonaws.com (52.215.57.49) 56(84) bytes of data.
--- d-wvxoekiiy7.execute-api.eu-west-1.amazonaws.com ping statistics ---
40 packets transmitted, 0 received, 100% packet loss, time 39943ms
**google ist erreichbar.**
root@evcc:~# ping google.com   
PING google.com (142.250.184.206) 56(84) bytes of data.
64 bytes from fra24s11-in-f14.1e100.net (142.250.184.206): icmp_seq=1 ttl=118 time=16.1 ms
64 bytes from fra24s11-in-f14.1e100.net (142.250.184.206): icmp_seq=2 ttl=118 time=16.1 ms
64 bytes from fra24s11-in-f14.1e100.net (142.250.184.206): icmp_seq=3 ttl=118 time=16.2 ms
^C64 bytes from 142.250.184.206: icmp_seq=4 ttl=118 time=16.0 ms

--- google.com ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 15113ms
rtt min/avg/max/mdev = 15.976/16.095/16.178/0.075 ms