evcc-io/evcc #8760: Ladelimit wird nicht beachtet

nalxnet, 2023-07-02:
### Describe the bug

Das Zielladen bis zum gesetzen Ladelimit (Energiemenge) funktioniert nicht mehr für Fahrzeuge, deren SoC nicht ermittelt werden kann. Nach erreichen der geladenen Soll-Energiemenge wird die Ladung nicht beendet sondern fortgeführt.

Konnte ich bisher beim Auto an der Wallbox wie auch beim Fahrrad an der Fritz-Steckdose beobachten.

### Steps to reproduce

1. Ladevorgang starten
2. Limit des Ladevorgangs im UI setzen
3. Ladung endet nicht beim eingestellten Limit sondern wird fortgeführt, bis das Fahrzeug voll geladen ist.

### Configuration details

```yaml
[...]
vehicles:
- type: template
  template: offline
  title: Peugeot iOn
  capacity: 14.5
  phases: 1
  mode: pv
  icon: car
  name: ion
- type: template
  template: offline
  title: R&M Multicharger
  capacity: 0.625
  phases: 1
  minCurrent: 0.85
  maxCurrent: 1
  mode: pv
  icon: bike
  name: xmulticharger
[...]
```

### Log details

```text
Wird bei Bedarf nachgeliefert.
```

### What type of operating system are you running?

Linux

### Version

1.118.2

nalxnet, 2023-07-02:
@andig Welcher Commit beinhaltet denn den Fix für dieses Verhalten? Ich habe in der History nichts passendes gesehen, bevor ich den Issue aufgemacht hab.

andig, 2023-07-02:
Welches Verhalten? Mangels Logfile keins erkennbar.

nalxnet, 2023-07-02:
Naja, die Ladung wird nicht gestoppt. Das ist auch ohne Log erkennbar und IMHO kein Grund, ein Issue nach einer Minute kommentarlos zu schließen...

Habe ein Debug-Log erzeugt und im Report ergänzt. Es ist zur besseren Lesbarkeit auf den lp-2 gefiltert, das ganze Log kann ich gerne nachliefern, da ist aber auch nicht mehr erkennbar. Ladevorgang nur vom Fahrrad mit der Fritz-Steckdose, da das Auto aufgrund von diesem Bug heute Mittag voll geladen hat. Aber auch da sieht man nur, dass die Ladung nicht durch evcc beendet wird:

Zähler Start: 457.837 kWh
Ladelimit gesetzt auf 50 Wh
Zähler Stop Soll: 457.887

Bei dem Zählerstand lädt er fröhlich weiter, bis ich die Ladung bei 457.896 kWh manuell übers UI beendet habe.

andig, 2023-07-02:
> kein Grund, ein Issue nach einer Minute kommentarlos zu schließen...

die Regeln stehen bereits im Issuetemplate.

andig, 2023-07-03:
Kann den Fehler in der Demo nachvollziehen.

closed 2023-07-03 by andig

nalxnet, 2023-07-03:
Vielen Dank für den schnellen Fix! 👍

andig, 2023-07-03:
Danke für deine Geduld- wenn ich den Fehler (dank Log) nachvollziehen kann gehts dann auch manchmal ganz schnell.