evcc-io/evcc #8762: KEBA: "unexpected status" bei Fahrzeugverbindung, Ladung startet erst beim 2. Versuch

nalxnet, 2023-07-02:
### Describe the bug

An meiner KEBA P30x (aktuelle Firmware) zeigt evcc beim ersten Ladeversuch nach dem Verbinden fast immer einen "unexpected status" und bricht den Ladevorgang ab. Nach Ende des Regelintervalls wird die Ladung erneut gestartet und funktioniert dann problemlos. Man hat damit aber immer ca. 15 Sekunden Verzögerung, bis das Fahrzeug lädt und einen angezeigten Fehler im UI.

### Steps to reproduce

Wallbox: KEBA P30x (aktuelle Firmware)

1. Fahrzeug mit Wallbox verbinden
2. Fahrzeug in evcc auswählen und ggf. Ladevorgang starten
3. Ladevorgang wird gestartet und beendet sich sofort wieder.
4. Nach dem Ende des Regelintervalls startet der Ladevorgang erneut, Fahrzeug wird nun geladen.

### Configuration details

```yaml
chargers:
- type: template
  template: keba 
  host: PRIVATEIP
  serial: REMOVED
  name: wallbox2

loadpoints:
- title: Ladestation
  charger: wallbox2
  mode: pv
  phases: 3
  maxCurrent: 32
  resetOnDisconnect: true
  priority: 5
  disable:
    threshold: 300
    delay: 5m
```

### Log details

```text
Jul 02 10:52:12 evcc evcc[81955]: [lp-1  ] ERROR 2023/07/02 10:52:12 charger: unexpected status: {ID:2 Serial:REMOVED State:4 Error1:6 Error2:32769 Plug:7 AuthON:0 AuthReq:0 EnableSys:1 EnableUser:1 MaxCurr:0 MaxCurrPercent:1000 CurrHW:20000 Curruser:6000 CurrFS:0 TmoFS:0 CurrTimer:0 TmoCT:0 SetEnergy:0 Output:0 Input:0 Sec:12406848}
Jul 02 10:52:32 evcc evcc[81955]: [lp-1  ] ERROR 2023/07/02 10:52:32 charger: unexpected status: {ID:2 Serial:REMOVED State:4 Error1:6 Error2:32769 Plug:7 AuthON:0 AuthReq:0 EnableSys:1 EnableUser:1 MaxCurr:0 MaxCurrPercent:1000 CurrHW:20000 Curruser:6000 CurrFS:0 TmoFS:0 CurrTimer:0 TmoCT:0 SetEnergy:0 Output:0 Input:0 Sec:12406868}
```

### What type of operating system are you running?

Linux

### Version

0.118.2

andig, 2023-07-02:
Interessant. Sagt die Doku etwas über State 4 aus?

premultiply, 2023-07-02:
[image]

andig, 2023-07-02:
Dann gibts einen Fehler. Den können wir klarer zeigen, lösen können wir ihn nicht. Die Frage was da los ist geht an den Keba support.

nalxnet, 2023-07-02:
Noch eine Ergänzung dazu: Das Verhalten tritt nicht auf, wenn die Keba-Wallbox ohne Steuerung verwendet wird. Ich frag mal bei KEBA nach, was hinter den Ziffern steht...

andig, 2023-07-03:
Closed in 2687f0b3ba8bc5d4f5a8beb38b1d597c78ed2283

closed 2023-07-03 by andig