evcc-io/evcc #8767: Ladestations-Template lässt sich nicht mehr

WUhle, 2023-07-03:
### Describe the bug

[FATAl] cannot create charger 'template':
Die folgende Sequenz ist seit 0.118.3 die meine Ladestation (NRGKick) nicht mehr ansprechbar macht.
Jul 03 09:58:07 IoBSVHome systemd[1]: Started evcc.service - evcc.
Jul 03 09:58:13 IoBSVHome evcc[858]: [main  ] INFO 2023/07/03 09:58:13 evcc 0.118.3
Jul 03 09:58:13 IoBSVHome evcc[858]: [main  ] INFO 2023/07/03 09:58:13 using config file: /etc/evcc.yaml
Jul 03 09:58:13 IoBSVHome evcc[858]: [main  ] INFO 2023/07/03 09:58:13 starting ui and api at :7070
Jul 03 09:58:15 IoBSVHome evcc[858]: [db    ] INFO 2023/07/03 09:58:15 using sqlite database: /var/lib/evcc/evcc.db
Jul 03 09:58:15 IoBSVHome evcc[858]: [mqtt  ] INFO 2023/07/03 09:58:15 connecting evccGW15 at tcp://localhost:1883
Jul 03 09:58:15 IoBSVHome evcc[858]: [mqtt  ] DEBUG 2023/07/03 09:58:15 tcp://localhost:1883 connected
Jul 03 09:58:16 IoBSVHome evcc[858]: [main  ] FATAL 2023/07/03 09:58:16 cannot create charger 'wallbox6': cannot create>
Jul 03 09:58:16 IoBSVHome evcc[858]: [main  ] FATAL 2023/07/03 09:58:16 will attempt restart in: 5m0s

wenn ich selber im Stop-Zustand den charger anspreche ist folgendes zu sehen:
pi@IoBSVHome:~ $ evcc charger
[main  ] INFO 2023/07/03 09:48:04 evcc 0.118.3
[main  ] INFO 2023/07/03 09:48:04 using config file: /etc/evcc.yaml
[db    ] INFO 2023/07/03 09:48:05 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/07/03 09:48:05 connecting evccGW15 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/07/03 09:48:05 tcp://localhost:1883 connected
[main  ] FATAL 2023/07/03 09:48:05 cannot create charger 'wallbox6': cannot create charger 'template': cannot create charger 'nrgkick-bluetooth': Properties.GetAll org.bluez.Adapter1: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist
pi@IoBSVHome:~ $
Bis zur Version 0.188.2 habe ich laden können, aber das Fahrzeug (IONIQ5) wurde nicht mehr beim Einstecken des Type2 erkannt.
Was kann ich noch testen?

### Steps to reproduce

1. einfach starten (statisch vorhanden)
2.
3.
...

### Configuration details

```yaml
pi@IoBSVHome:~ $ cat /etc/evcc.yaml
# open evcc at http://evcc.local:7070
sponsortoken: eyJhbGci-----------------oA  # your token from sponsor.

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: dda0b---------------------------407

interval: 10s # control cycle interval

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: GW15/evcc # root topic for publishing, set empty to disable publishing
  clientid: evccGW15
  user: usernm
  password: password

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168xxxxx
  port: 502
  name: grid1
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.xxxxx
  port: 502
  name: pv2
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.xxxxx1
  port: 502
  name: battery3
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.xxxxx
  port: 502
  name: battery4

chargers:
- type: template
  template: nrgkick-bluetooth
  mac: '00:1E:C0:76:77:81'
  pin: 1234
  name: wallbox6

vehicles:
- type: template
  template: hyundai
  title: IONIQ 5 3R
  user: 'wu@email'
  password: 'Password'
  vin: KM-------------7
  capacity: 72
  phases: 3
  icon: car
  cache: 15m
  mode: pv
  minSoc: 10
  targetSoc: 99
  minCurrent: 6
  maxCurrent: 17
  language: en
  name: ev5

loadpoints:
- title: EVC-Garage
  charger: wallbox6
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 18
  resetOnDisconnect: false

site:
  title: GW15-27252
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
    - battery4
pi@IoBSVHome:~ $
```

### Log details

```text
pi@IoBSVHome:~ $ evcc --log debug
[main  ] INFO 2023/07/03 10:05:51 evcc 0.118.3
[main  ] INFO 2023/07/03 10:05:51 using config file: /etc/evcc.yaml
[main  ] INFO 2023/07/03 10:05:51 starting ui and api at :7070
[db    ] INFO 2023/07/03 10:05:51 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/07/03 10:05:51 connecting evccGW15 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/07/03 10:05:51 tcp://localhost:1883 connected
[main  ] FATAL 2023/07/03 10:05:52 cannot create charger 'wallbox6': cannot create charger 'template': cannot create charger 'nrgkick-bluetooth': Properties.GetAll org.bluez.Adapter1: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

[main  ] FATAL 2023/07/03 10:05:52 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.118.3

closed 2023-07-03 by andig

olli0815, 2023-07-03:
ich habe den gleichen Fehler, wie gesagt.
Kann google pingen, Easee sagt API is up and running, curl funktioniert.
Hier meine Struktur innerhalb der Definition:

chargers:
 - name: hans
   type: template
   template: easee
   user: xxx.xxxxx@xxx.xx
   password: xxxxxxxxxx
   charger: EHxxxxxx

WUhle, 2023-07-06:
Ich denke die Ursache ist identifiziert und liegt irgendwie am bluetooth stack unter bookworm. Ich sehe die "device" im bluetoothctl und kann sie pairen, connecten usw. Aber evcc bekommt keinen automatischen handle auf dieses device, über das NRGKick zu erreichen ist. Die status ausgaben sagen:
Jul 06 18:14:03 IoBSVHome evcc[9663]: [lp-1  ] ERROR 2023/07/06 18:14:03 charge meter: find device: MapToStruct: Field not found: Bonded
Jul 06 18:14:05 IoBSVHome evcc[9663]: [lp-1  ] ERROR 2023/07/06 18:14:05 charge total import: find device: MapToStruct: Field not found: Bonded
Jul 06 18:14:07 IoBSVHome evcc[9663]: [lp-1  ] ERROR 2023/07/06 18:14:07 charger: find device: MapToStruct: Field not found: Bonded

Es scheint irgendwie mit geänderten defaults in Bookworm zu tun zu haben. 
Eine Lösung steht weiterhin aus. Falls einer einen Tipp hat - sehr gerne!