evcc-io/evcc #8775: Configure with EEBUS device collides with config plant ID generation

DerAndereAndi, 2023-07-03:
### Describe the bug

- Using an EEBUS device and connecting to that device, `machine.ProtectedID()` is used, see https://github.com/evcc-io/evcc/blob/master/charger/eebus/eebus.go#L68
- Using `configure` the `plant` value is only created at the end in `RenderConfiguration()`, see https://github.com/evcc-io/evcc/blob/master/cmd/configure/configure.go#L128
- Therefor on the next start of evcc using the generated configuration, the result of `machine.ProtectedID()` will be different to what was used when pairing the EEBUS device. And the remote device will decline the connection. To fix this, the user has to pair the charger in its web interface with evcc again.

### Steps to reproduce

1. Start `evcc configure`
2. Create a configuration with an EEBUS charger
3. Finish configuration
4. Start evcc with the generated configuration file
5. Connection to the charger now fails, even though it was successful in step 2

### Configuration details

```yaml
Not relevant to the issue.
```

### Log details

```text
On evcc startup the log will show something like this:

[eebus ] DEBUG 2023/07/03 10:17:45 trying to connect to „SKI Wallbox“ at „IP“
[eebus ] DEBUG 2023/07/03 10:17:45 initiating connection to „SKI Wallbox“ at „IP:4712“
[eebus ] DEBUG 2023/07/03 10:17:46 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/07/03 10:17:46 „SKI Wallbox“ websocket read error:  websocket: close 4450: SHIP id mismatch
```
```

### What type of operating system are you running?

Linux

### Version

0.118.3

andig, 2023-07-03:
Sieht aus als müsste einfach die Reihenfolge geändert werden? PR wäre super.

DerAndereAndi, 2023-07-03:
I would assume this issue might also exists for OCPP and SEMP.

andig, 2023-07-03:
Stutzig macht mich ja das hier: 

> don't use actual machine-id as file might be copied around to a different machine.

Das ist natürlich korrekt, aber wie wollen wir das lösen? Wenn es eine echte `machine-id` gibt sollten wir gar nix ausgeben. Falls nein könnten wir das tun, müssten es dann aber früher machen?

> I would assume this issue might also exists for OCPP and SEMP.

Bei Ocpp sollte die stationid eigentlich unwichtig sein.

DerAndereAndi, 2023-07-03:
Den Kommentar kann man auch inzwischen streichen, da ja immer die plant generiert und damit auch genutzt wird.

Bei OCPP und SEMP wird `machine.ProtectedID()` aufgerufen, und damit wird auch diese dann beim nächsten Start sich verändern. Durch den PR sollte das nicht mehr passieren.

closed 2023-07-03 by andig