evcc-io/evcc #8787: Loadpoint mode "off" is not set into default

bastian1607, 2023-07-03:
### Describe the bug

Nach dem Neustart oder nach dem Abstecken des E-Autos stellt sich der Loadpoint nicht wieder in default mode. Default sollte "off" sein. 
Mit 1.188.3 kein Problem gehabt.

(Bildschirmfoto wurde gemacht nachden das Log erstellt war)

[image]

### Steps to reproduce

1. Set default mode in yaml at loadpoint to "off".
2. Restart EVCC OR disconnect car
3. Mode is not set to off
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: payyours
plant: green

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: error
levels:
  site: error
  lp-1: error
  lp-2: error
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
[... 212 lines cut ...]
[lp-1  ] WARN 2023/07/03 19:26:58 poll mode '{always 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] WARN 2023/07/03 19:26:58 bufferStartSoc must be larger than bufferSoc
[site  ] INFO 2023/07/03 19:26:58 site config:
[site  ] INFO 2023/07/03 19:26:58   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/07/03 19:26:58     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/07/03 19:26:58     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/07/03 19:26:58     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/07/03 19:26:58   vehicles:
[site  ] INFO 2023/07/03 19:26:58     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/07/03 19:26:58 loadpoint 1:
[lp-1  ] INFO 2023/07/03 19:26:58   mode:        off
[lp-1  ] INFO 2023/07/03 19:26:58   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/07/03 19:26:58   meters:      charge ✓
[lp-1  ] INFO 2023/07/03 19:26:58     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2023/07/03 19:26:58 vehicle updated: unknown -> Ioniq
[lp-1  ] INFO 2023/07/03 19:26:58 car disconnected
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

1.188.5

bastian1607, 2023-07-03:
Gut zu sehen, dass der Default "off" überschrieben wird.

```
[site  ] DEBUG 2023/07/03 19:37:55 ----
[lp-1  ] DEBUG 2023/07/03 19:37:55 charge power: 0W
[site  ] DEBUG 2023/07/03 19:37:55 static grid tariff
[site  ] DEBUG 2023/07/03 19:37:55 pv power: 3961W
[site  ] DEBUG 2023/07/03 19:37:55 battery soc: 100%
[site  ] DEBUG 2023/07/03 19:37:55 battery power: 6W
[site  ] DEBUG 2023/07/03 19:37:55 grid power: -3820W
[site  ] DEBUG 2023/07/03 19:37:55 site power: -3714W
[lp-1  ] DEBUG 2023/07/03 19:37:55 vehicle odometer: 1750km
[lp-1  ] DEBUG 2023/07/03 19:37:55 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/07/03 19:37:55 charger status: A
[lp-1  ] INFO 2023/07/03 19:37:55 car disconnected
```
**[lp-1  ] DEBUG 2023/07/03 19:37:55 set charge mode: off**
```
[lp-1  ] DEBUG 2023/07/03 19:37:55 set min current: 6
[lp-1  ] DEBUG 2023/07/03 19:37:55 set max current: 16
[lp-1  ] DEBUG 2023/07/03 19:37:55 set min soc: 0
[lp-1  ] DEBUG 2023/07/03 19:37:55 set target soc: 100
```
**[lp-1  ] DEBUG 2023/07/03 19:37:55 set charge mode: pv**
```
[lp-1  ] DEBUG 2023/07/03 19:37:55 set min soc: 10
[lp-1  ] DEBUG 2023/07/03 19:37:55 set target soc: 100
[lp-1  ] DEBUG 2023/07/03 19:37:55 vehicle status: A
[lp-1  ] DEBUG 2023/07/03 19:37:55 vehicle soc: 74%
[lp-1  ] DEBUG 2023/07/03 19:37:55 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/07/03 19:37:55 vehicle range: 319km
[site  ] DEBUG 2023/07/03 19:37:55 ----
```

bastian1607, 2023-07-03:
[image]
[image]

Vielleicht interessant: Selbst wenn ich im Lademodus auf „Aus“ stelle, stellt sich EVCC nach dem disconnecten vom Auto wieder auf PV. 

Sind neue Einstellungen in der yaml notwendig?

premultiply, 2023-07-03:
Das Fahrzeug `Ioniq` ist in der Konfig als `mode: pv` konfiguriert.
Und genau das wird auch gemacht.

closed 2023-07-03 by premultiply

bastian1607, 2023-07-03:
Das tut EVCC aber nicht. Bitte Beschreibung, Screenshots, yaml und Logfile beachten.
Der Default mode ist „off“. Nach dem Abstecken des Autos (Modus aus) wird der Lademodus auf PV gestellt (Modus PV) -> Screenshot im Kommentar 
Wenn kein Auto angesteckt ist und EVCC neu gestartet wird, wird der Lademodus auf PV gestellt -> Logfile und Screenshots in der Beschreibung

bastian1607, 2023-07-03:
Ich denke es hat was mit der Fahrzeugerkennung zu tun, so wie das Debug aussieht. Konnte es wieder reproduzieren bei ausgestecktem (!) Auto.

Diese Zeilen stehen direkt nacheinander. Also ja, @premultiply , default bei car ist `pv`, aber `car` ist nicht eingesteckt, daher müsste er ja den default des `loadpoints` nehmen. Das Verhalten war sonst auch immer so. 

`[lp-1  ] INFO 2023/07/03 21:22:15 vehicle updated: unknown -> Ioniq
[lp-1  ] DEBUG 2023/07/03 21:22:15 set charge mode: pv `

Auszug aus der yaml:
`loadpoints:
  - title: Carport # display name for UI
    charger: wallbox # charger
    vehicle: my_car 
    mode: "off"  `