evcc-io/evcc #8789: > Mein Vorschlag: wenn EVCC bei verbundenem Fahrzeug ein Charger out of sync entdeckt, sollte die Ladung gestoppt und nach einer Pause (z.B. 1 Minute) neu gestartet werden. Dies könnte das Problem evtl. unschädlich machen.

ammann-ruchti, 2023-07-03:
> Mein Vorschlag: wenn EVCC bei verbundenem Fahrzeug ein Charger out of sync entdeckt, sollte die Ladung gestoppt und nach einer Pause (z.B. 1 Minute) neu gestartet werden. Dies könnte das Problem evtl. unschädlich machen.

Das ist schon so- nutzt aber nix wenn die Box nicht reagiert. Passiert das immer oder nur einmal? Falls nur einmal würde ich es ignorieren, falls immer bitte nochmal ein Logfile mit der Wallbox auf `trace`. 

Ich mache so lange zu bis wir ein passendes Logfile haben um das besser bewerten zu können- danke!

_Originally posted by @andig in https://github.com/evcc-io/evcc/issues/7635#issuecomment-1520365990_

ammann-ruchti, 2023-07-03:
Am 24. Juni hatte ich nochmals den Fall des "Charger out of sync" mit meiner Juice Charger Me Wallbox während eines Ladevorgangs. Das Logfile mit lp-1 auf Trace findet ihr beiliegend. Der Fall trat um 12:20:31 auf. Im Log des Wechselrichters sieht man den Einbruch des Ladevorgangs um ca. 12:30 deutlich.

Soll man das einfach ignorieren oder gibt es da noch etwas zu tun?

Andi, kannst Du damit etwas anfangen?
Vielen Dank, Hans-Jakob
[evcc - Charger out of sync 20230624.log](https://github.com/evcc-io/evcc/files/11940213/evcc.-.Charger.out.of.sync.20230624.log)

andig, 2023-07-05:
Schiebst Du das Log bitte ins alte Issue? Brauchen wir nicht doppelt.

closed 2023-07-05 by andig

ammann-ruchti, 2023-07-05:
Soeben erledigt.

andig, 2023-09-10:
Ich denke das out of sync Verhalten sollte in der aktuellen Version jetzt passen?