evcc-io/evcc #8803: Fahrzeug nicht erkannt, laden beginnt nicht

cburghardt, 2023-07-04:
### Describe the bug

Wenn ich mein Auto an die Wallbox anstecke, reagiert evcc nicht. Logfile habe ich angehängt.
Sobald ich evcc neu starte, wird das Fahrzeug erkannt und der Ladevorgang beginnt. Ich kann das jedes Mal reproduzieren.

### Steps to reproduce

1. Auto anstecken
2. Laut Dashboard ist das Auto nicht verbunden, im logfile steht das gleiche
3. Neustart
4. Laden beginnt

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 2dd502a030abe216cc6cdde4984bc60a198e279b6e5a52e250212eea441cb430

interval: 10s # control cycle interval

meters:
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/power_solar_generation/state
      timeout: 40s  
    currents:
      - source: mqtt    
        topic: homeassistant/sensor/solar_total_current_a/state
        timeout: 40s  
      - source: mqtt    
        topic: homeassistant/sensor/solar_total_current_b/state
        timeout: 40s  
      - source: mqtt    
        topic: homeassistant/sensor/solar_total_current_c/state
        timeout: 40s  
  - name: grid      
    type: custom
    power:  
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_m1_ac_power/state
      scale: -1
      timeout: 40s  
    currents:
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_a/state
        timeout: 40s  
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_b/state
        timeout: 40s  
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_c/state
        timeout: 40s  
  - name: battery
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_b1_dc_power/state
      scale: -1
      timeout: 4h 
[... 141 lines cut ...]
Jul 04 10:20:57 evcc evcc[3704]: [site  ] DEBUG 2023/07/04 10:20:57 battery soc: 77%
Jul 04 10:20:57 evcc evcc[3704]: [site  ] DEBUG 2023/07/04 10:20:57 battery power: -4230W
Jul 04 10:20:57 evcc evcc[3704]: [site  ] DEBUG 2023/07/04 10:20:57 grid power: 203W
Jul 04 10:20:57 evcc evcc[3704]: [site  ] DEBUG 2023/07/04 10:20:57 grid currents: [6.2 2.9 0]A
Jul 04 10:20:57 evcc evcc[3704]: [site  ] DEBUG 2023/07/04 10:20:57 site power: 303W
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 charge currents: [5.51 0 0]A
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 detected active phases: 1p
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 charge total import: 123.124kWh
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 charger status: C
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 vehicle soc: 100%
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 vehicle range: 53km
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 pv charge current: 4.68A = 6A + -1.32A (303W @ 1p)
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 site power 303W >= 0W disable threshold
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 pv disable in 0s
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 pv disable timer elapsed
Jul 04 10:20:57 evcc evcc[3704]: [lp-1  ] DEBUG 2023/07/04 10:20:57 guard enable in 4m40s
```

### What type of operating system are you running?

Linux

### Version

0.118.2

Adminius, 2023-07-04:
@cburghardt lösche mal bitte deine eMail und Passwort raus. 
und das Password darfst du ändern, denn Github vergisst ihn nicht

cburghardt, 2023-07-04:
Oh mann, danke!

andig, 2023-07-05:
Das Auto lädt:

    charger status: C
    
also alles wie erwartet.

closed 2023-07-05 by andig