evcc-io/evcc #8805: Zoomlevel wird im Kopfbereich ungünstig umgesetzt (rein optisch)

p7lejo, 2023-07-04:
### Describe the bug

Um mehr als 4 Ladepunkte auf einer Seite darzustellen, habe ich das Zoom-Level vom Browser genutzt:
Bei 67 % im Microsoft Edge passen bequem 4 Ladepunkte auf eine Seite. Der Kopfbereich lässt sich durch Klick auf den gelb-grünen Balken ganz oben sehr hübsch ein- oder ausblenden. In letzterem Zustand sieht es so aus:
[image]

Um nun 6 Ladepunkte zu sehen, habe ich das Zoomlevel weiter reduziert. Ab 50 % bleibt auch im ausgeblendeten Zustand ein großer leeren Bereich zwischen den Ladepunkten und dem gelb-grünen Balken ganz oben:
[image]

Vielleicht lässt sich das relativ einfach fixen? Noch schöner wäre freilich ein UX-Schalter: kompakte Darstellung oder so und dann schrumpfen alle Abstände, so dass einfach mehr lesbar auf einen kleinen Bildschirm passt (1980x1080 px). Vielen Dank und Grüße!

### Steps to reproduce

1. Definiere (mind.) 4 Ladepunkte
2. Zoomlevel 67 % im Browser Edge
3. Zoomlevel 50 %
...

### Configuration details

```yaml
needed?
```

### Log details

```text
needed?
```

### What type of operating system are you running?

Linux

### Version

0.118.5

VolkerK62, 2023-07-05:
Ah, jetzt ist mir das Verhalten klar. Es liegt an der vertikalen Zentrierung. Ich denke, die Erwartung ist/war, dass die Loadpoints immer unterhalb des Energiefluss beginnen.

p7lejo, 2023-07-05:
Danke, ihr habt das beide korrekt erkannt. Die Erklärung mit der Zentrierung ist schon sinnig, wenn es z.B. auf einem Hochkant-Bildschirm auch schick aussehen soll. Ticket kann geschlossen werden.

closed 2023-07-05 by p7lejo

naltatis, 2023-07-06:
Ja, Prio 1 sollte immer optimale Platzausnutzung haben. Danach dann ästhetische Aspekte wie Symmetrie. Wenn du einen Anwendungsfall hast wo wir das noch verbessern können (anderes Raster, ...) gerne her damit. Hast du dein Display auch im Hochformat?