evcc-io/evcc #8811: tariffs dynamisch über MQTT

lapixo, 2023-07-04:
Es wäre gut, wenn bei den tariffs eine Option kommt, 
in der man alternativ ein MQTT Pfad angeben kann, wo der aktuelle Strompreis hergeholt wird. . 

Das Problem ist nämlich, dass ich die Tibber API in dem Fall bei EVCC angeben MUSS und parallel in Home Assistant habe.
Jedoch ist die API nicht die schnellste und sorgt in dem Falls für eine unnötige Auslastung. 

Ziel ist es die Tibber API nur in Homeassistant einzusetzen, in der alle nötigen Werte ausgelesen werden und diese per MQTT an EVCC weiterzureichen.

mdkeil, 2023-07-05:
Welche Auslastung?-- Tibber Preise werden glaube um 12:00/13:00 für den nächsten Tag rausgegeben, sprich die API wird genau einmal/Tag für die Preise abgerufen.. sollte so auch implementiert sein.

andig, 2023-07-05:
Es braucht alle Werte, nicht nur den aktuellen. Die mqtt Struktur dafür ist unklar.

closed 2023-07-05 by andig

lapixo, 2023-07-05:
Die Tibber API hat ein Limit.
https://developer.tibber.com/docs/overview

Aber gut, wenn die Preise nur einmal pro Tag geholt werden, frage ich mich, was passiert wenn die API genau zu dem Zeitpunkt ausgefallen ist, was auch bei Tibber möglich ist. 
Denn ich habe das Problem, dass ich gestern und heute keine Preise mehr in EVCC angezeigt bekomme. 

Daher nochmal meine Bitte eine Möglichkeit zu bieten, den **aktuellen** Preis per MQTT auszulesen. 
Die Stundenpreise für die nächsten 24 Stunden können ja gerne weiterhin per Tibber API geholt werden.

andig, 2023-07-05:
Werden wir nicht bauen da die Ladeplanung alle Preise braucht. Für dein konkretes Problem bräuchte es ein Logfile.

lapixo, 2023-07-05:
Danke für deine Hilfestellung @andig ! Ich hatte ein Update eingespielt und alles neugestartet. Der Preis ist wieder da. Woran das jetzt lag, ist schwer nachzuvollziehen. 

Auf die Frage, was passiert, wenn die Tibber API nicht erreichbar ist, wurde leider nicht geantwortet. Ich habe im Quellcode gesehen, dass die Preise stündlich in einer Schleife geholt werden.

Ich werde es beobachten.

Danke für die Hilfe und Erklärung.