evcc-io/evcc #8817: [UI/UX] In dark mode totals on charging sessions page not readible

hyperbart, 2023-07-05:
### Describe the bug

[image]

[image]

### Steps to reproduce

1. Set EVCC to dark mode
2. Go to charging sessions, choose a month
3. Totals are not readable but when you select they become visible

### Configuration details

```yaml
N/A
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Linux

### Version

v118.5 nightly

StefanSchoof, 2023-07-05:
Sound like #8752

hyperbart, 2023-07-05:
Duplicate indeed: https://github.com/evcc-io/evcc/issues/8752

closed 2023-07-05 by hyperbart