evcc-io/evcc #8818: Error telemetry: upload failed

Snited, 2023-07-05:
### Describe the bug

telemetry: upload failed: unexpected status: 403 (Forbidden) every few seconds. reboot does not change anything.
it worked for a long time. i changed nothing in the config.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxx

interval: 20s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

mqtt:
  broker: 192.168.177.175:1883
  topic: # root topic for publishing, set empty to disable publishing
  clientid: evcc
  user: xxx
  password: xxx

meters:
- name: grid1
  type: custom
  power:
    source: mqtt
    topic: homeassistant/sensor/power_meter_active_power/state
    scale: -1 # Bezug/Einspeisung umtauschen
- name: pv_dach
  type: custom
  power:
    source: mqtt
    topic: homeassistant/sensor/inverter_input_power/state
- name: pv_gartenhaus
  type: custom
  power:
    source: mqtt
    topic: homeassistant/sensor/pv_gartenhaus_p_ac/state
- name: battery
  type: custom
  soc:
    source: mqtt
    topic: homeassistant/sensor/battery_state_of_capacity/state
  power:
    source: mqtt
    topic: homeassistant/sensor/battery_charge_discharge_power/state
    scale: -1 # Laden/Entladen umtauschen
  capacity: 10

chargers:
- type: template
  template: easee 
  user: xxx  
  password: xxx 
  charger: xxx 
  name: wallbox3

vehicles:
- type: template
  template: id
  title: ID4
  user: xxx
  password: xxx
  vin: xxx
  capacity: 77
  phases: 3
  icon: car
  cache: 15m
  mode: pv
  minSoc: 10
  targetSoc: 85
  minCurrent: 6
  maxCurrent: 25
  name: ev2

loadpoints:
- title: Carport
  charger: wallbox3
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 25
  resetOnDisconnect: false

site:
  title: zu Hause
  meters:
    grid: grid1
    pv:
      - pv_dach
      - pv_gartenhaus
    battery: battery
  prioritySoc: 75
  residualPower: 200
  bufferSoc: 80
  smartCostLimit: 0.10

tariffs:
  grid:
    type: tibber
    token: "xxx"
  feedin:
    type: fixed
    price: 0.0683
```

### Log details

```text
site  ] ERROR 2023/07/05 12:02:09 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:02:11 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:02:17 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:02:37 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:02:45 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:02:57 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:03:17 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:03:57 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:04:17 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:04:37 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:04:57 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:05:17 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:05:37 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:05:57 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:06:17 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/05 12:06:37 telemetry: upload failed: unexpected status: 403 (Forbidden)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.5

StefanSchoof, 2023-07-05:
Klingt nach #8117. Es wird meines Wissen immer noch einer ausreichend detaillierteren Log gesucht.

VolkerK62, 2023-07-05:
Was mir aufgefallen ist: 
Irgendwie ist dann wohl ein Teil des Sponsoring verloren gegangen. 
Im UI wird das Sponsoring zwar noch angezeigt, aber in MQTT unter `evcc/site/sponsor` steht dann "not available" oder " unavailable"

andig, 2023-07-05:
Passiert das eigentlich immer von Anfang an (nach einem evcc Neustart) oder erst "zwischendrin"?

Snited, 2023-07-05:
> Passiert das eigentlich immer von Anfang an (nach einem evcc Neustart) oder erst "zwischendrin"?

Das passiert dann direkt nach dem Neustart wieder. Hab ein neues Sponsor Token generiert und in die Condor gepackt, jetzt ist der Fehler weg. Aber das alte war definitiv noch nicht über 1000 Tage alt.

andig, 2023-07-05:
@Snited hast Du das alte Token noch und könntest mir das mal an info@evcc.io schicken?

VolkerK62, 2023-07-05:
Schwer zu sagen, wann der Auslöser passiert. Die Fehlermeldung kommt ja nur, wenn geladen wird.
Bei mir ist es nach einem Neustart dann wieder okay.

andig, 2023-07-06:
Note to self: anscheinend gültiges Token per 403 abgelehnt.

Snited, 2023-07-07:
Heute habe ich auch mit dem neuen Token das gleiche Problem. Zusätzlich mit der easee Wallbox 

[easee ] ERROR 2023/07/07 05:04:04 token refresh: unexpected status: 401 (Unauthorized)
[easee ] ERROR 2023/07/07 06:04:03 token refresh: unexpected status: 401 (Unauthorized)
[easee ] ERROR 2023/07/07 07:04:02 token refresh: unexpected status: 401 (Unauthorized)
[easee ] ERROR 2023/07/07 08:04:02 token refresh: unexpected status: 401 (Unauthorized)
[lp-1  ] INFO 2023/07/07 08:13:51 car connected
[lp-1  ] INFO 2023/07/07 08:14:11 start charging ->
[lp-1  ] WARN 2023/07/07 08:14:13 charger out of sync: expected disabled, got enabled
[lp-1  ] ERROR 2023/07/07 08:14:14 charger: Post "https://api.easee.com/api/chargers/EHCG646Y/commands/pause_charging": unexpected status: 401 (Unauthorized)
**[site  ] ERROR 2023/07/07 08:14:14 telemetry: upload failed: unexpected status: 403 (Forbidden)**
[lp-1  ] WARN 2023/07/07 08:14:31 charger out of sync: expected disabled, got enabled
[lp-1  ] ERROR 2023/07/07 08:14:31 charger: Post "https://api.easee.com/api/chargers/EHCG646Y/commands/pause_charging": unexpected status: 401 (Unauthorized)

andig, 2023-07-07:
Easee siehe https://github.com/evcc-io/evcc/discussions/8844.

orthoenergy, 2023-07-08:
Bei mir derselbe Fehler mit der aktuellen Version 0.118.7 beim Laden. 

telemetry: upload failed: unexpected status: 403 (Forbidden)

2x Easee Home

andig, 2023-07-09:
@Snited ich kann es leider- auch mit Deinem Token- nicht reproduzieren :(. Könnt ihr bitte mal folgende Konfiguration einstellen:

```
log: debug
levels:
  site: trace
```

Und mir dann- wenn es wieder passiert- das Log an info@evcc.io schicken?

VolkerK62, 2023-07-09:
>@VolkerK62 passiert das nur nach einem Neustart oder kann das auch beliebig zwischendrin auftreten?

Kann ich leider nicht sagen. Da ich sonst keinen Token brauche, fällt es mir erst auf, wenn geladen wird.
Deshalb habe ich jetzt immer noch den MQTT Explorer offen.
Dabei habe ich festgestellt, dass gelegentlich nach einem Neustart unter /evcc/site/sponsor "Sponsor not available" angezeigt wird. Gleichzeitig bekomme ich aber im UI (unten rechts) angezeigt, dass ich Sponsor bin. Ein weiterer Neustart behebt das Problem dann.

andig, 2023-07-09:
Das klingt nach möglichem root cause. Ich schaue mal ob es dazu Logging gibt. Evtl. Könntest du mal „main“ auf trace stellen und schauen ob da was zu sehen ist.

VolkerK62, 2023-07-09:
so wie es aussieht, wird bzgl Sponsoring nichts geloggt. Ich habe 2x einen Neustart gemacht. Beim zweiten war dann im MQTT das Sponsoring weg. Aber im Log unter `main: trace` sind nur Einträge bzgl. latest release.

Schlafog0r, 2023-07-14:
Ich habe den Fehler "telemetry: upload failed: unexpected status: 403 (Forbidden) " ebenfalls alle paar Sekunden, wenn geladen wird. An sich funktioniert sonst alles einwandfrei. Wie beschrieben tritt der Fehler nur beim Laden auf. Ich bin mir nicht zu 100% sicher, aber ich glaube ich hatte den Fehler zum ersten mal bereits mit Version 118.4
Ich führe jetzt daws Update auf 118.7 aus und melde mich nochmal (incl. log), wenn das Problem weiterhin auftritt.

andig, 2023-07-16:
Interessanter wäre ob die 118.3 den Fehler nicht hat- dann könnte ich nach Unterschieden fahnden.

andig, 2023-07-20:
Wir haben jetzt server-side Logging, wenns wieder passiert bitte melden damit ich sofort rein schauen kann- DANKE!

orthoenergy, 2023-07-22:
Fehler ist wieder da„telemetry: upload failed: unexpected status: 403 (Forbidden)“
Während des Ladens an Easee Home. 
EVCC v0.118.9

andig, 2023-07-22:
Ich sehe die Logs von 12:08 bis 12:42. Kein Fehler. Merkwürdig.

Schlafog0r, 2023-07-22:
Ich habe seit 118.8 das Problem nicht mehr. Lade aber auch an einer ABL Wallbox.

VolkerK62, 2023-07-22:
Gerade einen Neustart gemacht.
Seitdem tritt der Fehler wieder auf.
MQTT `evcc/site/sponsor` meldet "sponsorship unavailable"

Erneut neu gestartet, Sponsor ist wieder da, Fehler tritt nicht mehr auf.

orthoenergy, 2023-07-23:
Mit 0.118.10 und restart kein Fehler mehr.

andig, 2023-07-23:
> Mit 0.118.10 und restart kein Fehler mehr.

Das Problem tritt ja sporadisch auf. "Jetzt hab ich grad kein Problem" hilft also als Update auch nicht.

erikgieseler, 2023-07-29:
Mit der 0.118.11 besteht das Problem scheinbar auch noch.
Log von gerade eben:
```
[site  ] DEBUG 2023/07/29 18:10:31 ----
[lp-1  ] DEBUG 2023/07/29 18:10:31 charge power: 1W
[lp-2  ] DEBUG 2023/07/29 18:10:31 charge power: 3W
[lp-3  ] DEBUG 2023/07/29 18:10:31 charge power: 19W
[lp-4  ] DEBUG 2023/07/29 18:10:31 charge power: 0W
[lp-5  ] DEBUG 2023/07/29 18:10:31 charge power: 2796W
[site  ] DEBUG 2023/07/29 18:10:31 pv power: 3583W
[site  ] DEBUG 2023/07/29 18:10:31 battery soc: 100%
[site  ] DEBUG 2023/07/29 18:10:31 battery power: 0W
[site  ] DEBUG 2023/07/29 18:10:31 grid power: -307W
[site  ] DEBUG 2023/07/29 18:10:31 site power: -207W
[lp-1  ] DEBUG 2023/07/29 18:10:31 charge total import: 24.088kWh
[lp-1  ] DEBUG 2023/07/29 18:10:31 charger status: A
[site  ] TRACE 2023/07/29 18:10:31 telemetry: charge: Δ26/26Wh @ 2819W
[site  ] TRACE 2023/07/29 18:10:31 POST https://api.evcc.io/v1/charge
[site  ] TRACE 2023/07/29 18:10:31 {"instanceId":"***","chargePower":2819,"greenPower":2819,"chargeEnergy":0.025558082153865277,"greenEnergy":0.025543251094807092}
[site  ] ERROR 2023/07/29 18:10:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] DEBUG 2023/07/29 18:10:36 ----
```

thierolm, 2023-07-29:
Den Fehler habe ich auch seit heute Mittag. Ging im laufenden Betrieb los. Leider ist mein evcc.log abgeschnitten, so dass ich nicht sagen kann, wann es genau losging.

thierolm, 2023-07-29:
Ha, hab's doch gefunden. :-) Um 13:41:09h gings los ...

```
[main  ] INFO 2023/07/29 08:07:25 evcc 0.118.11
[main  ] INFO 2023/07/29 08:07:25 using config file: /etc/evcc.yaml
[lp-3  ] ERROR 2023/07/29 12:10:26 charge meter: Get "http://192.168.178.95/cm?cmnd=Status+8&password=&user=": dial tcp 192.168.178.95:80: i/o timeout
[lp-2  ] ERROR 2023/07/29 13:09:29 charger: Post "http://xxx:***@192.168.xxx.xxx/rpc/Shelly.GetStatus": dial tcp 192.168.178.35:80: i/o timeout
[lp-2  ] ERROR 2023/07/29 13:33:25 charge meter: Post "http://xxx:***@192.168.xxx.xxx/rpc/Shelly.GetStatus": dial tcp 192.168.178.35:80: i/o timeout
[lp-3  ] ERROR 2023/07/29 13:36:54 charger: Get "http://192.168.178.95/cm?cmnd=Status+0&password=&user=": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site  ] ERROR 2023/07/29 13:41:09 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/29 13:41:24 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/29 13:41:39 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/29 13:41:54 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/29 13:42:09 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] ERROR 2023/07/29 13:42:24 telemetry: upload failed: unexpected status: 403 (Forbidden)
...
```

VolkerK62, 2023-07-29:
@thierolm  hast du vor Kurzem einen Restart gemacht? Schau mal im UI, ob das Sponsoring angezeigt wird (evtl mit F5 aktualisieren)

thierolm, 2023-07-29:
Hi @VolkerK62 , ja irgendwas stimmt mit dem Sponsoring nicht: => Sponsorship unavailable :-(
[image]

andig, 2023-07-29:
Damit haben wir die Ursache bestätigt, danke!

thierolm, 2023-07-29:
Hab jetzt eine neues Sponsortoken und sponsore jetzt evcc.io und nicht mehr andig. Jetzt erkennt er mich wieder:
[image]

VolkerK62, 2023-07-30:
token habe ich auch schon erneuert. Gelegentlich (nicht immer) nach einem Neustart ist das Sponsoring aber wieder weg. Ein weiterer Neustart behebt das Problem dann meistens.

premultiply, 2023-07-30:
System- oder Dienstneustart?

VolkerK62, 2023-07-30:
Dienst. Also mit "restart evcc"

andig, 2023-08-03:
As of yesterday I have the hope our token management process might work better than before. Let's see if this pops up again during the next week and then tentatively close.

PeterMichaelMueller, 2023-08-11:
Hallo zusammen, 
ich bin auf der Version 0.118.11. Bei mir tritt der Fehler auch auf.

VolkerK62, 2023-09-02:
Leider keine Verbesserung.
Nach einem Update und trotz mehrmaliger Restarts bleibt es bei "sponsorship unavailable" und damit bei den Fehlermeldungen, wenn geladen wird.

Schlafog0r, 2023-09-02:
> Leider keine Verbesserung. Nach einem Update und trotz mehrmaliger Restarts bleibt es bei "sponsorship unavailable" und damit bei den Fehlermeldungen, wenn geladen wird.

Also bei mir ist das Problem schon länger nnicht mehr aufgetreten. Hast Du Dir einen neuen Token geholt? Das und eines der letzten SW Updates hat zum Erfolg geführt.

VolkerK62, 2023-09-02:
Bei mir war es auch eine ganze Zeit ruhig. In der Vergangenheit hat ein Neustart das Problem immer behoben.
Obwohl der Token "ewig" gültig ist, habe ich ihn erneuert. Trotzdem kein Erfolg. Version ist das aktuelle Nightly.

thierolm, 2023-09-02:
Hatte gestern auch das Problem mit der 0.119.3. Hab mit dem zweiten neu generierten Tokens hat es dann funktioniert.

MrASN, 2023-09-02:
Habe das Problem jetzt auch neu mit 0.119.3, bereits zwei neue Token generiert und mehrfach neu gestartet. Habe die Telemetrie jetzt erstmal ausgeschaltet, führt sonst zu Irritation der Nutzer.

markusjnagel, 2023-09-02:
@andig - ist  heute bei mir zum ersten Mal aufgetreten (mit 0.119.3)... ich möchte das Laden jetzt nicht unterbrechen, aber wenn noch Analyse-Bedarf ist, stelle ich gern mal auf Trace... wenn Du mir genau sagst, was:

So wie Du's [weiter oben](https://github.com/evcc-io/evcc/issues/8818#issuecomment-1627661475) beschrieben hast?

Oh, und ich sehe gerade, dass ich evtl ein neues Token generieren sollte (wobei meins vielleicht gerade 2 Wochen alt ist).

[image]

Bevor ich jetzt ein neues Token generiere und das hier zum Heisen-Bug wird... gebt mir einfach kurz Bescheid, ob weitere Logs im aktuellen Zustand gewünscht/hilfreich sind (falls sich das nach einem Neustart nicht in Luft auflöst).

Vielen Dank Euch!!

Markus
```
ep 02 16:57:16 evcc evcc[1294]: [lp-1  ] ERROR 2023/09/02 16:57:16 charge meter: data not available
Sep 02 16:57:16 evcc evcc[1294]: [lp-1  ] ERROR 2023/09/02 16:57:16 charge meter: not available
Sep 02 16:57:16 evcc evcc[1294]: [lp-1  ] INFO 2023/09/02 16:57:16 car connected
Sep 02 16:57:23 evcc evcc[1294]: [lp-1  ] INFO 2023/09/02 16:57:23 vehicle updated: unknown -> Kia
Sep 02 16:57:23 evcc evcc[1294]: [lp-1  ] WARN 2023/09/02 16:57:23 charger out of sync: expected disabled, got enabled
Sep 02 16:57:31 evcc evcc[1294]: [lp-1  ] INFO 2023/09/02 16:57:31 start charging ->
Sep 02 16:57:31 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:57:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:57:46 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:57:46 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:58:06 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:58:06 telemetry: upload failed: Post "https://api.evcc.io/v1/charge": context deadline exceeded
Sep 02 16:58:16 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:58:16 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:58:31 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:58:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:58:46 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:58:46 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:59:01 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:59:01 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:59:16 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:59:16 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:59:31 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:59:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 16:59:46 evcc evcc[1294]: [site  ] ERROR 2023/09/02 16:59:46 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:00:01 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:00:01 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:00:16 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:00:16 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:00:31 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:00:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:00:51 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:00:51 telemetry: upload failed: Post "https://api.evcc.io/v1/charge": context deadline exceeded
Sep 02 17:01:01 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:01:01 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:01:16 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:01:16 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:01:31 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:01:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:01:46 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:01:46 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:02:01 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:02:01 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:02:16 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:02:16 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:02:31 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:02:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:02:46 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:02:46 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:03:01 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:03:01 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:03:16 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:03:16 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:03:31 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:03:31 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:03:46 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:03:46 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:04:01 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:04:01 telemetry: upload failed: unexpected status: 403 (Forbidden)
Sep 02 17:04:16 evcc evcc[1294]: [site  ] ERROR 2023/09/02 17:04:16 telemetry: upload failed: unexpected status: 403 (Forbidden)

```

HooLeeChit, 2023-09-02:
Bin dann jetzt auch betroffen!

Ebenfalls erst seit heute morgen mit Update auf 119.3!

Mehrere Restart des Dienstes ohne Erfolg...

Gruß
Markus

sigma415, 2023-09-02:
Here also since update form 0.119.1 to 0.119.3 yesterday

snapscanii, 2023-09-03:
Ich bin leider auch betroffen. Ewigen Token von Andig bekommen. Was soll ich tun (119.3)

VolkerK62, 2023-09-03:
>Was soll ich tun (119.3)

nichts. Ist ja nicht so tragisch. Du kannst die Übermittlung ausschalten, dann kommen keine Fehlermeldungen mehr.

mdkeil, 2023-09-03:
Bei mir nun auch seit Update auf 119.3 / Lebenslang-Token, habe daher die Telemetrie-Datenübertragung nun deaktiviert, bis es dauerhaft gefixt ist.

MKausHH, 2023-09-03:
bei mir steht (seit 0.119.3 - und auch beim Downgrade auf 0.119.2):
"Du bist Unterstützer" und "Vielen Dank, sponsorship unavailable!"
Token ist aktuell (extra nochmal aktualisiert)

HooLeeChit, 2023-09-03:
Mit 119.4 wurde es bei mir anscheinend gelöst!
Zumindest funktioniert es aktuell wieder...

StefanSchoof, 2023-09-03:
Ich würde ehr ein Zusammenhang mit https://github.com/evcc-io/evcc/issues/9696 als der Version vermuten.

homrunner, 2023-09-04:
Bin seit heute auch betroffen!

Ebenfalls erst mit heutigem Update auf 119.4 über latest von zuvor 118.4 von vor dem Urlaub.

Zahllose Restarts des Dienstes bleiben ohne Erfolg. Sponsorship (mein endloser Token, vor 5 Wochen gesponsert) ist auch zeitgleich betroffen: "Du bist Unterstützer
Vielen Dank, sponsorship unavailable! Dein Beitrag hilft, evcc weiterzuentwickeln."

@markusjnagel, du schreibst von „Token erneuern“. Wie geht denn das?

Grüße 
Roland

sigma415, 2023-09-04:
Bei mir war erst wieder Ruhe an beiden Fronten, nachdem ich ein neues Sponsortoken geholt hatte **und** telemetry wieder auf `false `gesetzt hatte, Seitdem werde ich auch als Sponsor wieder namentlich erkannt:

@homrunner:
Hier Sponsortoken erhalten: https://sponsor.evcc.io/

markusjnagel, 2023-09-04:
@homrunner - was @sigma415 sagt :wink:  (Danke, warst schneller... :grin: ) 

Nur zur Info - habe heute neues Token generiert, eingetragen, neugestartet.... und immer noch denselben Fehler. :shrug: 
Hatte per UI Telemetry ausgeschaltet, werde morgen mal per YAML ausschalten.... (und ggf wieder einschalten, wenn's denn geht).

homrunner, 2023-09-05:
Danke @sigma415, @markusjnagel, 

Ich haben den Token erneuert und die Telemetry im UI vorübergehend deaktiviert. So klappt es aktuell wieder fehlerfrei. 

Gerne würde ich meine Daten wieder über die Telemetry mit evcc teilen und hoffe, dass die Entwickler die Ursache schnell finden.

Digital-DW, 2023-09-05:
Moin, bei mir ist das Problem auch seit ein paar Tagen (nach einem Neustart) vorhanden. 

Ich hatte bis eben die 0.118.11 drauf und jetzt auf die aktuellste Version aktualisiert.

sigma415, 2023-09-05:
Irgendwie habe ich den Eindruck, dass die beiden (konkurrierenden ?) Telemetry-Einstellungen in der yaml und im UI eine unterschiedliche Auswirklung auf den Fehler haben.
Bei mir ist es jetzt so: 
a.) `telemetry: false` **und** im UI "Ich möchte meine Ladedaten teilen" aktiviert => Keine 403-Fehlermeldung
b.) `telemetry: true` **und** im UI "Ich möchte meine Ladedaten teilen" aktiviert => 403-Fehlermeldung und "Vielen Dank, sponsorship unavailable!"
c.) (... die dritte Möglichkeit hab ich jetzt nicht mehr ausprobiert)

Und was mich noch etwas verwundert: 
Hier https://docs.evcc.io/docs/guides/setup/#telemetry--community-daten  steht, dass `telemetry: true` **veraltet ("deprecated")** wäre ....
Nur veraltet ? Oder auch problematisch ?

markusjnagel, 2023-09-05:
Scheint wohl so ne Art Heisenbug zu sein... 

Musste eh grad an die `evcc.yaml` ran, da habe ich gesehen dass bei mir überhaupt kein `telemetry:<true|false>` drin war, also hab ich es mal mit reingenommen (auf `false` wie bei Dir Fall a.) ).
Bei der Gelegenheit nochmal fix nen neues Token geholt (sicher is sicher)... aber: Keine Änderung, immer noch 403 und immer noch "_sponsorship unavailable_"
:shrug:

GrimmiMeloni, 2023-09-05:
@andig ich sehe das auch immer wieder. Ich hab jetzt hier site und main auf trace, zusammen mit log-headers. Ich denke dann kommen wir jetzt schnell(er) zu einer Lösung.

Frage dazu noch: Mein bisheriger Eindruck (kann aber falsch sein) ist, daß ich das vor allem sehe wenn ich zum Testen meine eigenen Builds laufen habe. Gibt es da vielleicht irgendeinen Filter (vielleicht basierend auf Git Tags oder commit IDs) ?

andy6d, 2023-09-05:
Gleiches Problem hier auch. 
v0,119.4
neues Token geholt
Dienst mehrfach neu gestartet und immer noch "Vielen Dank, sponsorship unavailable! Dein Beitrag hilft, evcc weiterzuentwickeln."

ThomasGeissler, 2023-09-05:
Dito

alese82, 2023-09-05:
Bei mir das selbe Problem, nur zusätzlich besteht noch das Problem, dass die Easee Home die eingestellte Ladegrenze nicht erkennt und einfach weiter lädt. Womöglich durch den nicht korrekt erkannten Token!?
v0.119.4
Mein Sponsoring richtet sich noch an andig direkt und nicht an evcc-io! Könnte dies ein Teil des Problems sein, zudem scheint meine evcc.yaml shon recht alt zu sein.

Hier die Ausgabe ohne aktiven Ladevorgang des debug-logs:

[main  ] INFO 2023/09/05 21:32:57 evcc 0.119.4
[main  ] INFO 2023/09/05 21:32:57 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/05 21:32:57 starting ui and api at :7070
[db    ] INFO 2023/09/05 21:33:07 using sqlite database: /home/alese82/.evcc/evcc.db
[lp-1  ] WARN 2023/09/05 21:33:10 vehicles option is deprecated
[lp-1  ] WARN 2023/09/05 21:33:10 locking phase config to 3p for switchable charger
[site  ] WARN 2023/09/05 21:33:10 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/09/05 21:33:10 deprecated: use 'battery' instead of 'batteries'
[site  ] WARN 2023/09/05 21:33:10 battery configured but residualPower is missing (add residualPower: 100 to site)
[site  ] INFO 2023/09/05 21:33:10 site config:
[site  ] INFO 2023/09/05 21:33:10   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/05 21:33:10     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/05 21:33:10     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/05 21:33:10     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/09/05 21:33:10   vehicles:
[site  ] INFO 2023/09/05 21:33:10     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/09/05 21:33:10 loadpoint 1:
[lp-1  ] INFO 2023/09/05 21:33:10   mode:        pv
[lp-1  ] INFO 2023/09/05 21:33:10   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/09/05 21:33:10   meters:      charge ✓
[lp-1  ] INFO 2023/09/05 21:33:10     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/09/05 21:33:10 phase timer inactive
[lp-1  ] DEBUG 2023/09/05 21:33:10 pv timer inactive
[lp-1  ] DEBUG 2023/09/05 21:33:10 guard timer inactive
[site  ] WARN 2023/09/05 21:33:10 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[server] FATAL 2023/09/05 21:33:10 unlinkat /tmp/evcc: operation not permitted

[6 more replies]

closed 2023-09-06 by andig