evcc-io/evcc #8819: VW I.D. verliert verbindung zur ELLI Wallbox (timeout)?

dc-redeclipse, 2023-07-05:
### Describe the bug

Der ID 3 wird eingesteckt am vor Abend, beginnt normal mit dem Laden, falls noch etwas Sonne da ist. Am nächsten Morgen beginnt schon um 6 Uhr bevor die min PV Leistung erreicht ist, das Laden mit (6A) 4.1kW, wenn gerade mal von der PV nur 1kW kommt. Aber evcc zeigt es nicht mehr an, dass der ID 3 noch immer verbunden ist und geladen wird. Lösung ist über die Volkswagen API den ID 3 einmal zu trennen und wieder zu verbinden, dann geht alles wieder.
Ich vermute da einen Bug.
Man könnte so eine reconnect Funktion bestimmt in evcc dafür bestimmt integrieren.

### Steps to reproduce

Der ID 3 wird eingesteckt am vor Abend, beginnt normal mit dem Laden, falls noch etwas Sonne da ist. Am nächsten Morgen beginnt schon um 6 Uhr bevor die min PV Leistung erreicht ist, das Laden mit (6A) 4.1kW, wenn gerade mal von der PV nur 1kW kommt. Aber evcc zeigt es nicht mehr an, dass der ID 3 noch immer verbunden ist und geladen wird. 

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1
  usage: grid
  host: xxx
  name: grid1
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: xxx
  name: pv2

chargers:
- type: template
  template: ellipro
  ski: xxx
  ip: xxx
  name: wallbox5

vehicles:
- type: template
  template: id
  title: ID3
  user: xxx
  password: xxx
  vin: xxx
  capacity: 58
  icon: car
  cache: 15m
  mode: pv
  targetSoC: 80
  name: ev4
- type: template
  template: fiat
  title: 500e
  user: xxx
  password: xxx
  vin: xxx
  capacity: 21
  pin: xxx
  icon: car
  cache: 15m
  mode: pv
  targetSoC: 80
  name: ev5
  
loadpoints:
- title: VW Wallbox
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Halle
  meters:
    grid: grid1
    pvs:
    - pv2

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
xxx
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
xxx
          -----END CERTIFICATE-----

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.3151 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.0580 # [currency]/kWh
```

### Log details

```text
- (im log steht nichts relevantes darüber)
```

### What type of operating system are you running?

Linux

### Version

0.118.5

StefanSchoof, 2023-07-05:
Wenn die Verbindung getrennt wird, lädt die Elli Wallbox von selbst. Ein Wiederaufbau der Verbindung von evcc hilft da leider nicht. Das Problem liegt in der Firmware der Elli Box und lässt sich nicht von evcc lösen. Du kannst nur schauen dein Netzwerk stabiler zu bekommen, am besten mit einem Netzwerk Kabel zu Box. Ich selber habe die Box recht nah am Haus in der Nähe meines Routers und es klappt mit WLAN, das ein Verbindungsabbruch sehr selten passiert.

dc-redeclipse, 2023-07-05:
> Wenn die Verbindung getrennt wird, lädt die Elli Wallbox von selbst. Ein Wiederaufbau der Verbindung von evcc hilft da leider nicht. Das Problem liegt in der Firmware der Elli Box und lässt sich [...]

Die Verbindung ist direkt mit via LAN und nicht das Problem.
Es gibt ja eine Lösung und das ist die über die API über die VW-App wie beschrieben, die gleiche, die evcc schon nutzt, für alle ID Modelle.
Es geht jetzt darum die Situation für einen Reconnect über die API festzustellen, wann dieser erforderlich ist:
ID nicht getrennt -> evcc lädt mit gast -> dann reconnect über die API von VW
Es geht darum das nun zu implementieren da es nervt das händisch zutun

DerAndereAndi, 2023-07-05:
Vielleicht wäre es doch mal sinnvoll das Log bereitzustellen um zu sehen was da wirklich vor sich geht. Ansonsten gibt es generell Fehler in der Firmware der Wallbox die zu dem Szenario führen, wie es @StefanSchoof beschrieben hat.