evcc-io/evcc #8821: Easee permanente Timeouts

josefzahner, 2023-07-05:
### Describe the bug

Heute auf v0.118.5 gewechselt An wechselhaften Tagen, so wie heute oder letzten Sonntag, sehe ich extrem oft Fehler in der Art "max charge current 12A: timeout". Am Sonntag kam noch "charger: timeout" und "charger logic error: disabled but charging" dazu (da war ich aber noch auf einer früherern 0.118.x Version). Das Ganze ging am Sonntag sogar so weit, das ewig die Phasenumschaltung nicht ging, d.h. ich hab deswegen immer wieder grössere Mengen an Strom eingekauft. Ich war davon so genervt das ich schlussendlich auf 1P limitiert habe. Gehe ich recht in der Annahme dass das ein reines Easee Cloud API Problem ist?
Ehrlich gesagt ist das extremst mühsam, auch jetzt gerade wieder innerhalb der letzten 10min mind 10 Meldung das er versucht hat die Ampere umzustellen, es aber nicht klappt. D.h. die Regelung funktioniert überhaupt nicht wie gewünscht und ich kaufe immer mal wieder ein oder die Load ist zu tief.
Sind das wirklich so viele Easee Cloud Probleme? Oder gibts hier noch weitere Issues?

### Steps to reproduce

An wechselhaften Tagen die Umschaltung von Power anschauen -> immer wieder Timeouts

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.118.5

josefzahner, 2023-07-05:
Gerade jetzt auf v0.118.6 gewechselt, Fehler sind direkt wieder aufgetreten.

jheinitz, 2023-07-05:
Das mit der Easee Cloud API ist vorsichtig gesagt nur *suboptimal*. Was für Fehler hattest Du denn? 401 oder429? Ich vermute, dass Du erstmal von Easee geblockt wurdest. Da hilft Dir aktuell auch die neue Version nichts. Diese sorgt nur zukünftig dafür, dass Du erst gar nicht geblockt wirst. Wenn das denn alles so funktioniert, wie durch langwieriges Testen von mehreren Leuten herausgefunden wurde. Aber ich gebe Dir recht, dass es echt nervt. Aber man kauft sich aj auch nicht eben mal eine andere Wallbox wie eine andere Sorte Kaffee .....

jheinitz, 2023-07-05:
Du müsstest in Deiner `evcc.yml` mal den Loglevel für `easee` mindestens aus `debug` stellen. Sonst sieht man zwar die Fehler, aber für eine Analyse reicht das nicht.

josefzahner, 2023-07-05:
Ok, all right. Ich hab das log hochgesetzt. Aktuell fehlt mir aber die Sonne um zu laden - sobald ich was sehe melde ich mich wieder. Was mir nicht klar ist, wenn ich geblockt bin, warum kann ich dann trotzdem noch calls absetzen? Weil zwischendurch scheint die Kommunikation ja zu gehen, sonst würde Power ja nicht angepasst werden.

josefzahner, 2023-07-05:
So also, nach ein paar Minuten schon das hier. Soll ich noch mehr Debug enablen? 

```Jul 05 17:30:45 evcc systemd[1]: Started evcc.
Jul 05 17:30:45 evcc evcc[17821]: [main  ] INFO 2023/07/05 17:30:45 evcc 0.118.6
Jul 05 17:30:45 evcc evcc[17821]: [main  ] INFO 2023/07/05 17:30:45 using config file: /etc/evcc.yaml
Jul 05 17:30:45 evcc evcc[17821]: [main  ] INFO 2023/07/05 17:30:45 starting ui and api at :7070
Jul 05 17:30:55 evcc evcc[17821]: [db    ] INFO 2023/07/05 17:30:55 using sqlite database: /var/lib/evcc/evcc.db
Jul 05 17:31:00 evcc evcc[17821]: [site  ] WARN 2023/07/05 17:31:00 deprecated: use 'pv' instead of 'pvs'
Jul 05 17:31:00 evcc evcc[17821]: [semp  ] WARN 2023/07/05 17:31:00 SEMP_BASE_URL unspecified, using http://192.168.20.185:7070 instead
Jul 05 17:31:00 evcc evcc[17821]: [site  ] INFO 2023/07/05 17:31:00 site config:
Jul 05 17:31:00 evcc evcc[17821]: [site  ] INFO 2023/07/05 17:31:00   meters:      grid ✓ pv ✓ battery ✗
Jul 05 17:31:00 evcc evcc[17821]: [site  ] INFO 2023/07/05 17:31:00     grid:      power ✓ energy ✓ currents ✓
Jul 05 17:31:00 evcc evcc[17821]: [site  ] INFO 2023/07/05 17:31:00     pv 1:      power ✓ energy ✓ currents ✓
Jul 05 17:31:00 evcc evcc[17821]: [site  ] INFO 2023/07/05 17:31:00     pv 2:      power ✓ energy ✓ currents ✓
Jul 05 17:31:00 evcc evcc[17821]: [site  ] INFO 2023/07/05 17:31:00   vehicles:
Jul 05 17:31:00 evcc evcc[17821]: [site  ] INFO 2023/07/05 17:31:00     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
Jul 05 17:31:00 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:31:00 loadpoint 1:
Jul 05 17:31:00 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:31:00   mode:        pv
Jul 05 17:31:00 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:31:00   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Jul 05 17:31:00 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:31:00   meters:      charge ✓
Jul 05 17:31:00 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:31:00     charge:    power ✓ energy ✓ currents ✓
Jul 05 17:31:00 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:31:00 vehicle updated: unknown -> Tesla Model Y Performance
Jul 05 17:31:00 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:31:00 car connected
Jul 05 17:38:42 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:38:42 wait for DCC update: 6.000
Jul 05 17:38:43 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:38:43 send command: resume_charging
Jul 05 17:38:47 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:38:47 wait for DCC update: 32.000
Jul 05 17:38:47 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:38:47 DCC update received, current: 6.000, dynamicChargerCurrent: 32.000
Jul 05 17:38:47 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:38:47 send enable, reset current: 6.000
Jul 05 17:38:47 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:38:47 wait for DCC update: 6.000
Jul 05 17:38:57 evcc evcc[17821]: [lp-1  ] ERROR 2023/07/05 17:38:57 charger enable: timeout
Jul 05 17:38:58 evcc evcc[17821]: [lp-1  ] INFO 2023/07/05 17:38:58 start charging ->
Jul 05 17:38:58 evcc evcc[17821]: [lp-1  ] WARN 2023/07/05 17:38:58 charger out of sync: expected disabled, got enabled
Jul 05 17:38:58 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:38:58 send command: pause_charging
Jul 05 17:39:08 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:08 wait for DCC update: 0.000
Jul 05 17:39:08 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:08 DCC update received, current: 6.000, dynamicChargerCurrent: 0.000
Jul 05 17:39:08 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:08 send command: resume_charging
Jul 05 17:39:11 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:11 wait for DCC update: 32.000
Jul 05 17:39:11 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:11 DCC update received, current: 6.000, dynamicChargerCurrent: 32.000
Jul 05 17:39:11 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:11 send enable, reset current: 6.000
Jul 05 17:39:16 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:16 wait for DCC update: 6.000
Jul 05 17:39:21 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:21 wait for DCC update: 16.000
Jul 05 17:39:40 evcc evcc[17821]: [lp-1  ] ERROR 2023/07/05 17:39:40 max charge current 10A: timeout
Jul 05 17:39:45 evcc evcc[17821]: [easee ] DEBUG 2023/07/05 17:39:45 wait for DCC update: 6.000
Jul 05 17:40:40 evcc evcc[17821]: [lp-1  ] ERROR 2023/07/05 17:40:40 max charge current 8A: timeout
Jul 05 17:40:50 evcc evcc[17821]: [lp-1  ] ERROR 2023/07/05 17:40:50 max charge current 11A: timeout
Jul 05 17:41:00 evcc evcc[17821]: [lp-1  ] ERROR 2023/07/05 17:41:00 max charge current 8A: timeout
Jul 05 17:41:11 evcc evcc[17821]: [lp-1  ] ERROR 2023/07/05 17:41:11 max charge current 8A: timeout```

Das WebGUI zeigt gleichzeit das hier
![Screenshot 2023-07-05 at 17 42 25](https://github.com/evcc-io/evcc/assets/33521474/5fc0440d-a1ff-49a6-81f2-37f47d931009)

josefzahner, 2023-07-05:
Ok, habe ich mir fast gedacht. Ich befürchte ich kann hier heute ev. nicht mehr weiter testen, mal schauen wie das Wetter sich verhält. 
Morgen bin ich den ganzen Tag mit dem Auto weg, d.h. ich würde ev. erst am Freitag wieder ein Update posten. Item, log ist jetzt auf trace und danke für den Tipp bezüglich copy/paste. Werde ich beachten. Sobald ich was habe melde ich mich wieder.

andig, 2023-07-05:
Hier gibt es nichts zu tun. Für die Timeouts sind die Easee Server verantworlich- bitte an den Support wenden.

mesu0815, 2024-05-10:
@josefzahner 
ich wollte mal fragen wie du das Probleme gelöst hast? Ich habe die gleiche Meldungen im LOG. 
Grüße
Markus

josefzahner, 2024-05-13:
Das Problem bestand noch bevor EVCC automatisiert neue Tokens angefordert hat. Es gab 2-3 Monate später einen Fix für EVCC der mein Problem definitiv behoben hat.
Wenn du Timeouts hast, kann es sein das du geblockt wirst von Easee oder deine Credential/Token nicht passt.