evcc-io/evcc #8835: Tesla WakeUp nur bei "Schnell"-Laden

eternalvr, 2023-07-06:
**Is your feature request related to a problem? Please describe.**
Wenn der Modus PV/PV+Min aktiv ist, wird das Fahrzeug zum laden nicht aufgeweckt.

**Describe the solution you'd like**
Fahrzeug sollte auch bei Nicht-"Schnell"-Laden aufgeweckt werden.

**Describe alternatives you've considered**
Das Fahrzeug manuell über die App aufwecken ist keine schöne Alternative.

**Additional context**

Debug Log in Modus PV+MIN
`Jul 06 09:36:50 home.intern.coderain.de evcc[3485419]: [lp-1  ] ERROR 2023/07/06 09:36:50 charger enable: asleep
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 ----
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:00 charge power: 0W
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 static grid tariff
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 pv power: 1662W
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 battery soc: 22%
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 battery power: -844W
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 grid power: -7W
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 grid powers: [475 -501 20]W
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 grid currents: [2.18 -2.22 2.2]A
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 giving priority to battery charging at soc: 22%
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:00 site power: 93W
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:00 charge voltages: [3.6 0 0]V
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:00 charge currents: [0 0 0.4]A
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:00 charger status: B
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:00 pv charge current: 0A = 0A + -0.135A (93W @ 3p)
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:00 POST https://owner-api.teslamotors.com/api/1/vehicles/92985046xxxx/command/charge_start
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:00 {"response":null,"error":"vehicle unavailable: {:error=>\"vehicle unavailable:\"}","error_description":""}
Jul 06 09:37:00 home.intern.coderain.de evcc[3485419]: [lp-1  ] ERROR 2023/07/06 09:37:00 charger enable: asleep
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 ----
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:10 charge power: 0W
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 static grid tariff
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 pv power: 1656W
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 battery soc: 22%
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 battery power: -839W
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 grid power: 8W
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 grid powers: [469 -501 40]W
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 grid currents: [2.16 -2.21 2.23]A
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 giving priority to battery charging at soc: 22%
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:10 site power: 108W
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:10 charge voltages: [3.6 0 0]V
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:10 charge currents: [0 0 0.4]A
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:10 charger status: B
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:10 pv charge current: 0A = 0A + -0.156A (108W @ 3p)
Jul 06 09:37:10 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:10 POST https://owner-api.teslamotors.com/api/1/vehicles/92985046xxxx/command/charge_start
Jul 06 09:37:11 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:11 {"response":null,"error":"vehicle unavailable: {:error=>\"vehicle unavailable:\"}","error_description":""}
`

Kein Wakeup

Umschalten auf "Schnell":

`Jul 06 09:37:29 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:29 charge power: 0W
Jul 06 09:37:29 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:29 static grid tariff
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 pv power: 1649W
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 battery soc: 22%
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 battery power: -786W
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 grid power: 1W
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 grid powers: [416 -497 82]W
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 grid currents: [1.94 -2.19 2.46]A
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 giving priority to battery charging at soc: 22%
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 site power: 101W
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:30 charge voltages: [3.6 0 0]V
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:30 charge currents: [0 0 0.4]A
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:30 charger status: B
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:30 POST https://owner-api.teslamotors.com/api/1/vehicles/92985046xxxx/command/set_charging_amps
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:30 {"charging_amps": 16}
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: --
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: {"response":null,"error":"vehicle unavailable: {:error=>\"vehicle unavailable:\"}","error_description":""}
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [lp-1  ] DEBUG 2023/07/06 09:37:30 max charge current: waking up vehicle
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:30 POST https://owner-api.teslamotors.com/api/1/vehicles/92985046xxxx/wake_up
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [tesla ] TRACE 2023/07/06 09:37:30 {"response":{"id":929850460136086,"user_id":16892530xxxx3,"vehicle_id":168926987xxxx,"vin":"LRWYGCFS3PCxxxx","color":null,"access_type":"OWNER","tokens":["e7dae142fc33081b","d745077c64xxxxe"],"state":"asleep","in_service":false,"id_s":"92985046013xxxx","calendar_enabled":true,"api_version":58,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}}
Jul 06 09:37:30 home.intern.coderain.de evcc[3485419]: [site  ] DEBUG 2023/07/06 09:37:30 ----
`

Es scheint also als hätte der set_charging_amp call ein fallback, falls das Fahrzeug schläft. start_charging aber nicht.

Relevante Config

`chargers:
- type: template
  template: twc3
  host: 192.168.5.21
  name: wallbox5

vehicles:
- type: template
  template: tesla
  title: x
  accessToken: x
  refreshToken: x
  vin: LRWYGCFS3PCx
  capacity: 58
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv
  guardduration: 10m
  mincurrent: 1
  maxcurrent: 16
  resetOnDisconnect: true
`

Feature Request: Zusätzlichen Call zu awake, falls start_charging fehlschlägt

closed 2023-07-06 by andig

eternalvr, 2023-07-07:
0.118.1 was used. Bug has probably been fixed with the .2 

https://github.com/evcc-io/evcc/commit/e88c994fece5b15c839623f09ecb61f7c1903b1a

using latest version to test now

andig, 2023-07-07:
Indeed 👍🏻