evcc-io/evcc #8852: Set charging speed in instant / fast charge mode

Skyleiger, 2023-07-07:
Hello everyone,

first of all, I speak German - but to maintain the unofficial rules among us developers, I make my statement here in English. I hope this is understandable for everyone. However, inquiries with me can also happen in German.

**Is your feature request related to a problem? Please describe.**
Yes, here's the problem: I have to leave at 2pm and I know the sun will be shining all day. So I would like to manually discharge my storage by 2pm (say to 10%) to get the battery fuller. I myself know that the storage will be full again during the day when I am no longer at home with the car.

So I would like to manually define the charging current (amps) with which the vehicle is charged. I can not use the mode instant charge / fast charge, because here is always charged with the maximum 11 kW and my memory can only output a maximum of 5 kW.

I know that I can also do this manually via the app of my wallbox (go eCharger), but evcc always interferes and changes this according to its own wishes. e.g. every few seconds the wallbox is switched off again or the charging speed is changed.

**Describe the solution you'd like**
For me, the best option would be that you can still manually readjust the charging current (amps) via a slider in fast charge / instant charge mode.
Means I select instant charge and then a slider appears, which is by default on the 16/32A. I then set this down to 6A, for example, and this is passed on to the wallbox.

**Describe alternatives you've considered**
An alternative would be, of course, that you can set that evcc does not control and switch the wallbox for the current charging process. Personally, however, I would find the other option better, because then no two apps would have to be used.

**Additional context**
Even though this bothers me, I would like to praise the software. I initially wanted to go with openWB, however openWB only caused problems and the setup (in a Docker container) did not want to work reasonably. Evcc, on the other hand, worked right away and has a very nice user interface. I also found the integration as a home assistant addon really great. By setting the feed-in tariff to 0 cents, I can now always see the real purchase price that I pay. Really great.

I would be very happy if this function would now also be delivered here. 

Thanks a lot,
Dominic

VolkerK62, 2023-07-07:
Wenn du eine Nightly-Version installierst (keine Ahnung ob das in HA geht) dann kannst du oben rechts in den Einstellungen die "experimentellen Funktionen" einschalten.
Sind diese eingeschaltet, kannst du im UI am Loadpoint ganz einfach `maxcurrent` ändern.

Alternative:
Du konfigurierst für diese Fall ein zweites Fahrzeug als Offline-Vehicle und legst dort bei `onIdentify` das entsprechende `maxcurrent` fest.

naltatis, 2023-07-11:
Ja, über die experimental UI im Nightly geht das heute. Allerdings wird das Setting beim Neustart auf den yaml Wert zurückgesetzt. Daher würd ich das auch weiterhin erstmal hinter dem Flag lassen. Die Persistenz für dieses Loadpoint-Setting würde ich im Rahmen der Config-UI (#6029) sehen.

closed 2023-07-11 by naltatis

Skyleiger, 2023-07-11:
> Ja, über die experimental UI im Nightly geht das heute. Allerdings wird das Setting beim Neustart auf den yaml Wert zurückgesetzt. Daher würd ich das auch weiterhin erstmal hinter dem Flag lassen. Die Persistenz für dieses Loadpoint-Setting würde ich im Rahmen der Config-UI (#6029) sehen.

Würde ich auch so sehen. Die Einstellungen direkt im Webpanel zu haben (und bei Bedarf bspw. schnell übers Handy umzustellen), wäre schon echt cool. Besteht denn die Möglichkeit mit dem Home Assistant Addon die Nightly Version zu nutzen?