evcc-io/evcc #8858: Running EVCC withtou loadpoint

negesti, 2023-07-07:
**Is your feature request related to a problem? Please describe.**
No, this is a "real" feture request

A friend of mine saw my grafana dashboard that relies on the data evcc writes to influxdb and wants the same setup but has no loadpoint. As my friend has a SolarEdge inverter, the setup would be 99% the same if I can re-use my evcc -> influxdb -> grafana setup.
Plus: I'm to lazy to re-write my influx/grafana setup to support data from ModBus TCP directly

**Describe the solution you'd like**
EVCC should start an get data from the inverter although there is no loadpoint defined.
***Plus*** This would make people aware of EVCC even if they don't have a EV / wallbox yet 

**Describe alternatives you've considered**
Create a dummy loadpoint. However i can not think of a setup that will not throw a lot of errors

**Additional context**
The Influx proxy feature is great ! Especially if you have a Inverter like that is limited in the number of connections

Gerne auch auf Deutsch, falls mein Englisch zu holprig ist

premultiply, 2023-07-07:
Wontfix. Ladesteuerung ohne Loadpoint macht keinen Sinn.
Du kannst dir aber mit den vorhandenen Mitteln (Custom charger) einen Dummy-Loadpoint selbst bauen.

closed 2023-07-07 by premultiply

orchyAdmin, 2025-07-25:
@premultiply 

Can you share what the dummy loadpoint configuration is?

negesti, 2025-07-25:
Have a look at https://github.com/evcc-io/evcc/discussions/2454 and https://github.com/evcc-io/evcc/discussions/2426#discussion-3840559

StefanSchoof, 2025-07-25:
You can also use the demo charger: https://docs.evcc.io/en/docs/devices/chargers#demo-charger

orchyAdmin, 2025-07-26:
> You can also use the demo charger: https://docs.evcc.io/en/docs/devices/chargers#demo-charger

@StefanSchoof - will this work for a real car configuration?