evcc-io/evcc #8870: invalid phase wiring between charge meter and charger

user8376, 2023-07-08:
### Discussed in https://github.com/evcc-io/evcc/discussions/8857

<div type='discussions-op-text'>

<sup>Originally posted by **user8376** July  7, 2023</sup>
Hi, habe häufiger folgende Meldung "invalid phase wiring between charge meter and charger"
Meine Wallbox ist ein Go-E Charger. Drei Phasig angeschlossen.
Mein Auto, ein ID3 mit zwei Phasiger Elektronik
In der yaml ist das entsprechend eingestellt.
```
vehicles:
  phases: 2
loadpoints:
  phases: 3
```
Einen Fehler in der Verkabelung kann ich nicht erkennen und auch das Laden scheint ganz normal zu funktionieren.</div>

Die Wallbox Go-E Charger V2 misst tendenziell immer eine zu niedrige Spannung und liefert teilweise Werte unter 208V. Die ungenaue Messfunktion ist Modelbedingt bekannt.
Messung mit Multimeter kurz vor der Wallbox liefert 10-15 V höhere Spannungswerte.

Ein (einstellbarer) Korrekturwert wäre wünschenswert

user8376, 2023-07-08:
betrifft viele Besitzer von go-e, aber nicht alle.
siehe auch z.B. hier
https://www.goingelectric.de/forum/viewtopic.php?t=53811

andig, 2023-07-09:
In #8867 wurde die Minimalspannung auf 207V erhöht- das reicht hoffentlich.

closed 2023-07-09 by andig