evcc-io/evcc #8877: Easee too many request

AlexBu14, 2023-07-09:
### Describe the bug

Bekomme öfters folgende Meldung von der Easee api: 
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
..
```

### Log details

```text
..
```

### What type of operating system are you running?

Docker container

StefanSchoof, 2023-07-09:
Siehe https://github.com/evcc-io/evcc/discussions/8757#discussioncomment-6384316

Deinem Bug Report fehlen leider alle notwendigen Infos, damit jemand sinnvoll damit was anfangen kann.

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

AlexBu14, 2023-07-09:
Umstellen auf Aus und dann wieder PV hat das Problem ausgelöst

AlexBu14, 2023-07-09:
Hier ein Auszug von log

ERROR 2023/07/09 08:38:04 charger enable: Post "https://api.easee.com/api/chargers/EH2PBN5C/commands/resume_charging": unexpected status: 429 (Too Many Requests)
--
2023-07-09 06:38:04 | stdout | [easee ] DEBUG 2023/07/09 08:38:04 send command: resume_charging
2023-07-09 06:38:04 | stdout | [easee ] WARN 2023/07/09 08:38:04 smart charging: Post "https://api.easee.com/api/chargers/EH2PBN5C/settings": unexpected status: 429 (Too Many Requests)

worx777, 2023-07-09:
Had the same issue, rebooted everything multiple times and now it's finally working

andig, 2023-07-09:
Ohne trace log nicht zu bewerten, siehe https://github.com/evcc-io/evcc/discussions/8757#discussioncomment-6395311 und https://github.com/evcc-io/evcc/discussions/8757#discussioncomment-6395314.

closed 2023-07-09 by andig