evcc-io/evcc #8896: Easee Token stacht anscheinend weiterhin probleme

Snited, 2023-07-10:
### Describe the bug

Easee Home kann häufig nicht durch evcc gesteuert werden.
Zusätzlich seit dem Update folgender Fehler häufig im log:
[easee ] ERROR 2023/07/10 16:27:16 token refresh: unexpected status: 401 (Unauthorized)

### Steps to reproduce

Nothing

### Configuration details

```yaml
Siehe #8818
```

### Log details

```text
[lp-1  ] ERROR 2023/07/10 16:32:02 max charge current 21A: Post "https://api.easee.com/api/chargers/EHCG646Y/settings": unexpected status: 401 (Unauthorized)
[site  ] ERROR 2023/07/10 16:32:02 telemetry: upload failed: unexpected status: 403 (Forbidden)
[easee ] ERROR 2023/07/10 16:32:16 token refresh: unexpected status: 401 (Unauthorized)
[lp-1  ] ERROR 2023/07/10 16:32:22 max charge current 21A: Post "https://api.easee.com/api/chargers/EHCG646Y/settings": unexpected status: 401 (Unauthorized)
[site  ] ERROR 2023/07/10 16:32:22 telemetry: upload failed: unexpected status: 403 (Forbidden)
[lp-1  ] ERROR 2023/07/10 16:32:42 max charge current 20A: Post "https://api.easee.com/api/chargers/EHCG646Y/settings": unexpected status: 401 (Unauthorized)
[site  ] ERROR 2023/07/10 16:32:42 telemetry: upload failed: unexpected status: 403 (Forbidden)
[lp-1  ] ERROR 2023/07/10 16:33:02 max charge current 20A: Post "https://api.easee.com/api/chargers/EHCG646Y/settings": unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.7

sphinx02, 2023-07-10:
Exakt gleiches Problem hier. Nach EVCC Neustart kann der Ladevorgang ordnungsgemäß gestartet werden. Wenn nach 3-4h dann aber wegen des Ladeziels abgeschalten werden soll, dann kommt die Fehlermeldung und der Ladevorgang wird nicht ordnungsgemäß beendet.

PV Überschuss laden funktioniert dann auch nicht wenn der Einbruch des Überstroms erst nach einer längeren Zeit kommt.

Kann man etwas tun um dieses Verhalten zu verbessern?

josefzahner, 2023-07-10:
Gibt hier schon ein Issue:
https://github.com/evcc-io/evcc/discussions/8757

andig, 2023-07-10:
Siehe https://github.com/evcc-io/evcc/discussions/8757#discussioncomment-6407223

closed 2023-07-10 by andig