evcc-io/evcc #8899: additional Info in OCPP Template

VolkerK62, 2023-07-10:
Hilfreich wäre für mich in der Doku bei https://docs.evcc.io/docs/devices/chargers#wallbox-pulsar-plus-commander-2-copper-sb diese Info gewesen: 

```
- OCPP-Server-URL in der Wallboxkonfiguration: ws://<evcc-IP-Adresse>:8887/ (Verbindung über das lokale Netzwerk)
- Ladepunktidentität bzw. Ladestation Identität: beliebiger Wert, der in der evcc.yaml als stationid eingetragen wird
- Passwort: leer lassen

chargers:
  - name: my_charger
    type: template
    template: ocpp 
    stationid: <Ladepunktidentität aus der Wallboxkonfiguration>

Für weitere Infos siehe: https://docs.evcc.io/docs/devices/chargers#ocpp-16j-kompatible-wallbox-mit-smart-charging-profil
```

_Originally posted by @urylux in https://github.com/evcc-io/evcc/discussions/8876#discussioncomment-6404835_

premultiply, 2023-07-10:
Das muss ich mir selbst erstmal ansehen.
Prinzipiell sollte das aber möglich sein.

Ich meine die genannten Parameter sind aber alle optional. Daher sollte es auch ohne diese funktionieren.

Den Begleittext kann man problemlos hinzufügen. Sowas muss halt definitiv immer von Box-Besitzern zugeliefert werden.

premultiply, 2023-07-10:
@urylux könntest du mal einen Screenshot von der OCPP-Konfiguration der Box hier einfügen damit ich mal sehen kann wie die Felder dort genau bezeichnet sind. Idealerweise auch noch auf Deutsch und Englisch - falls es da Unterscheidungen geben sollte. Danke!

urylux, 2023-07-11:
Hallo @premultiply 
auf jeden Fall. Hier die App:
[image]

Und nochmal auf deren Website:
[image]

Die Sprache kann ich leider nirgendwo umstellen ...

Hier noch ein Screenshot von der Website, wo man den Zugang einschränken kann (relevant für idtag?). Die Nutzer müssen an anderer Stelle angelegt werden. 
[image]

StefanSchoof, 2023-07-11:
Für mindestens eine OCPP Wallbox muss die URL "ws://ip-von-evcc/EVVCXXXXXXXX" sein, siehe https://github.com/evcc-io/evcc/discussions/8681#discussioncomment-6417461

urylux, 2023-07-12:
@VolkerK62 Der Titel ist eigentlich falsch, es geht nicht um OCCP-Wallboxen (bei denen in der Doku schon alle Infos stehen), sondern um die Pulsar Plus, bei der nichts zur Einrichtung der OCCP-Schnittstelle steht. 
An sich würde es aus meiner Sicht reichen, wenn man einfach den Text aus dem ersten Post dort einfügt, dann hat es der nächste Pulsar Plus-Besitzer auf jeden Fall einfacher.

closed 2023-07-13 by premultiply

urylux, 2023-07-13:
Hm, in der Doku steht bei Pulsar Plus (https://docs.evcc.io/docs/devices/chargers#wallbox-pulsar-plus-commander-2-copper-sb) immer noch das gleiche. Oder ging es hier gar nicht darum, die Doku zu verbessern?

PR #8900 (2023-07-13): Device-specific OCPP template improvements
Fixes #8899
Ref #8827