evcc-io/evcc #8904: Elli Connect: Incorrect charging power display using a wallbox without counter

hermann-59, 2023-07-10:
### Describe the bug

I have a SMA Energy Meter, a Cupra Leon Plugin Hybrid to a Cupra Connect Wallbox (without meter). When charging, it always displays a (correct) generated power and grid power=0W and a wrong charging power of (single phase) constant 3.7kW, even if the generation is less than 3.7kW.
Could the actual charging power in the display, which is not available due to the lack of a wallbox meter, be replaced by the charging power demanded by evcc? Then the value would at least be correct if the wallbox actually sets the required charging power.

### Steps to reproduce

1. Configure a charger using the template "elliconnect"
2. Configure a vehicle using template "seat"
3. Start charging the vehicle when generation power is less the 3.7kW
4. Examine the charging power in the Web UI. It shows the wrong value 3.7kW

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 91adb67dc08248e97de72dc93dc16928ab8be1f9ca947f28adbc8d86518684d6

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-energy-meter 
  usage: grid
  host: 192.168.178.56  
  name: EnergyMeter
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.29  
  password: ***********************  
  name: PvGarage
- type: template
  template: shelly-1pm 
  usage: pv  
  host: 192.168.178.70  
  name: PvBalkon
- type: template
  template: shelly-1pm 
  usage: pv  
  host: 192.168.178.76  
  name: PvPavillon

chargers:
- type: template
  template: elliconnect 
  ski: ************************************
  ip: 192.168.178.60  
  name: Wallbox

vehicles:
  - name: Cupra
    type: template
    template: seat
    title: Cupra Leon
    user: *********************
    password: *******************
    vin: *****************
    capacity: 13
    phases: 1

loadpoints:
- title: Garage
  charger: Wallbox
  vehicle: Cupra
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuhause
  meters:
    grid: EnergyMeter
    pv: 
      - PvGarage
      - PvBalkon
      - PvPavillon

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          ****************************
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          ****************************
          -----END CERTIFICATE-----
```

### Log details

```text
not required
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.7

premultiply, 2023-07-10:
Already discussed and explained in #8892

Not possible using buggy Elli box. Elli firmware reports this metering data without a meter.
Not possible without an internal or external charge meter.
Adding an internal or external charge meter solves the issue not related to evcc itself.

closed 2023-07-10 by premultiply

hermann-59, 2023-07-11:
Has anybody an idea why the elli boxs works using a different configuration (StefanShoof wrote "Ich habe auch die Elli Box ohne Zähler, aber im VW Branding und lade 1p. Bei mir zeigt evcc die Werte an die an die Wallbox gesendet werden und nicht immer 3,7 kW.") ?

premultiply, 2023-07-11:
NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.