evcc-io/evcc #8909: Ldevorgang stoppt direkt nach Start, setzt dann wieder fort und stoppt ggf. nochmals (Easee)

Robbe64, 2023-07-11:
### Describe the bug

Wenn das Auto (eGolf) angeschlossen ist und der Ladevorgang mit EVCC gestartet wird, so wird kutz nach dem Start der Ladevorgang wieder gestoppt und anschließend wieder gestartet. Das ist zwar unschön aber noch kein großes Problem.
Wenn ich jedoch das Auto anstecke, der Ladevorgang von der Easee selbst gestartet wird und dann von EVCC zunächst gestoppt wird (wie in #8298 beschrieben), wird dann wieder von EVCC regulär gestartet und dann nochmals wie oben beschrieben sofort wieder gestoppt. Anschließend streikt aber scheinbar das Auto (eGolf) und will nicht nochmal starten, so dass dann nicht geladen wird und ich auch nicht mehr manuell, weder in EVCC noch in der App des eGolf, den Ladevorgang starten kann. Erst wenn ich das Auto kurz abstecke und wieder anstecke lädt es. Diesmal dann mit "nur" einer kurzen Unterbrechung.

### Steps to reproduce

Da hier zwei Probleme zusammen wirken (#8298 und dass stoppen ohne an und ab stecken) beschränke ich mich auf den Fall, dass das Auto bereits angeschlossen ist und noch nicht lädt da z.B. das Ladelimit erreicht ist:

1. Auto ist angesteckt und lädt nicht, da Ladelimit erreicht.
2. Ladelimit erhöhen, so dass geladen werden kann. (ggf. noch von PV auf MIN+PV stellen, damit tatsächlich geladen wird)
3. Beobachten, dass das Laden startet
4. Beobachten, dass das Laden kurz stoppt
5. Beobachten, dass das Laden wieder startet
...

### Configuration details

```yaml
#uri: 0.0.0.0:7070 # uri for ui
plant: *****
interval: 20s # control cycle interval

log: debug
levels:
  id: trace
  #easee: trace
#  lp-1: trace
#  tibber: trace

sponsortoken: *****

meters:
  - type: template
    template: kostal-plenticore
    id: 71
    host: 192.168.178.101
    port: 1502
    usage: pv
    modbus: tcpip
    name: pv1
  - type: template
    template: kostal-plenticore
    id: 71
    host: 192.168.178.101
    port: 1502
    usage: battery
    modbus: tcpip
    name: battery2
    capacity: 5.2
  - type: template
    template: kostal-ksem-inverter
    id: 71
    host: 192.168.178.101
    port: 1502
    usage: grid
    modbus: tcpip
    name: grid3

chargers:
  - type: template
    template: fritzdect
    uri: https://192.168.178.1
    user: *****
    password: *****
    ain: *****
    standbypower: -1
    name: fritzdect
    integrateddevice: true
    icon: generic
[... 245 lines cut ...]
[lp-1  ] DEBUG 2023/07/11 07:36:56 charger status: C
[lp-1  ] DEBUG 2023/07/11 07:36:56 vehicle soc: 86%
[lp-1  ] DEBUG 2023/07/11 07:36:56 vehicle range: 127km
[lp-1  ] DEBUG 2023/07/11 07:36:56 pv charge current: 0.453A = 7A + -6.55A (1506W @ 1p)
[site  ] DEBUG 2023/07/11 07:36:56 telemetry: charge: Δ15/16Wh @ 1502W
[site  ] DEBUG 2023/07/11 07:37:13 ----
[lp-1  ] DEBUG 2023/07/11 07:37:13 charge power: 1502W
[lp-2  ] ERROR 2023/07/11 07:37:16 charge meter: not available
[site  ] DEBUG 2023/07/11 07:37:16 dynamic grid tariff
[site  ] DEBUG 2023/07/11 07:37:16 pv power: 510W
[site  ] DEBUG 2023/07/11 07:37:17 battery soc: 62%
[site  ] DEBUG 2023/07/11 07:37:17 battery power: 1452W
[site  ] DEBUG 2023/07/11 07:37:17 grid power: 12W
[site  ] DEBUG 2023/07/11 07:37:17 site power: 1564W
[lp-2  ] ERROR 2023/07/11 07:37:17 charge total import: not available
[lp-2  ] ERROR 2023/07/11 07:37:18 charger: not available
```

### What type of operating system are you running?

Docker container

### Version

0.118.7

SuperDa75, 2023-07-11:
Kann ich bestätigen, ist bei meiner easee genauso. Ladevorgang im PV Modus startet und stoppt nach ein paar Sekunden, gefolgt von den Meldungen "charger enable:timeout" und "charger out of sync". Danach lädt easee ohne weitere Unterbrechungen. Noch so ein Cloud-Problem bei easee nach 401/429?

jheinitz, 2023-07-11:
Moin!
Wenn ich mich so richtig zurück erinnere, dann hatte ich das glaube ich auch vor Kurzem. Mein Skoda Enyaq fing an zu laden, aber dann ging nix mehr. Nachdem ich ihn dann nochmal abgesteckt und wieder angeschlossen habe, ging es. Und von Zeit zu Zeit bekomme ich auch mal ein Timeout. Aber manchmal auch beim Setzen der Stromstärke. Da hatte ich mir bisher nichts bei gedacht. Werde das jetzt mal beobachten und mich dann melden.

Sorry, dass ich noch nicht früher geantwortet habe, war viel unterwegs .... :-)

GrimmiMeloni, 2023-07-12:
Es tritt ein Timeout beim ersten Enable bzw Resume auf. Daher versucht evcc es erneut im nächsten Zyklus.

GrimmiMeloni, 2023-07-12:
> Wenn das Auto (eGolf) angeschlossen ist und der Ladevorgang mit EVCC gestartet wird, so wird kutz nach dem Start der Ladevorgang wieder gestoppt und anschließend wieder gestartet. Das ist zwar unschön aber noch kein großes Problem.

Die hier dargestellte Annahme, dass es evcc immer den Ladevorgang einmal startet und stoppt, ist nicht korrekt.
@Robbe64 betreibst du die Easee ohne RFID Authentifizierung?

Robbe64, 2023-07-12:
Ja, aktuell ohne RFID Authentifizierung.
Aber: Ich wollte in diesem Bug explizit nicht das Problem aus #8298 betrachten, da es dafür ja bereits einen Bug gibt. Habe es nur erwähnt, da die Kombination dieser beiden Probleme bei mir dazu führt, dass anschließend garnicht geladen werden kann.

Mir geht es hier darum, dass das Laden bei bereits angestecktem Fahrzeug beim Starten des Ladevorgangs erst kurz wieder gestoppt wird und dann wieder gestartet wird. Das ist auch der Fall der im Log zu sehen ist.

Womöglich ist es aber letztendlich doch das gleiche Problem, da nach dem Timeout der Fehler "charger out of sync" kommt und das dann im nächsten Durchlauf zu "send command: pause_charging" führt? Dann wäre dieses Problem tatsächlich nur eine etwas andere Ausprägung von #8298 und hätte die gleiche Ursache. Wenn ihr das auch so seht können wir diesen Buck gerne als Duplikat schließen und ich würde estmal nur den Timeout hoch setzen.

SuperDa75, 2023-07-12:
Ich nutze evcc und easee jetzt seit über einem Jahr, ich bin mir sehr sicher, dass es diese Unterbrechungen in den ersten Monaten nicht gab. 

Aktuell fallen mir die Ladepausen auch beim Phasenwechsel auf.

Heute morgen startete PV Überschuss zumindest ohne weitere "Mitteilungen" und viel wichtiger ohne 401/429.

GrimmiMeloni, 2023-07-12:
> Womöglich ist es aber letztendlich doch das gleiche Problem, da nach dem Timeout der Fehler "charger out of sync" kommt und das dann im nächsten Durchlauf zu "send command: pause_charging" führt? Dann wäre dieses Problem tatsächlich nur eine etwas andere Ausprägung von #8298 und hätte die gleiche Ursache. Wenn ihr das auch so seht können wir diesen Bug gerne als Duplikat schließen und ich würde estmal nur den Timeout hoch setzen.

Für mich sieht es danach aus. Aktuell haben wir das Timeout allerdings hart auf 10s verdrahtet. Wenn Du dieses Verhalten dauerhaft und reproduzierbar siehst, wäre ich an einen Trace log interessiert.

GrimmiMeloni, 2023-07-16:
@Robbe64 wie ist denn jetzt der Stand? Sollen wir das als Duplicate schließen, oder wartest Du noch auf eine Repro mit Trace Log?

andig, 2023-07-16:
> Aktuell fallen mir die Ladepausen auch beim Phasenwechsel auf.

@SuperDa75 der Phasenwechsel hatte schon immer eine Ladepause und muss die auch haben- sonst gibts Kurzschluss am Auto.

Mangels besserer Erkenntnisse Duplikat zu #8298

closed 2023-07-16 by andig

GrimmiMeloni, 2023-07-18:
@Robbe64 habe mir das [log.txt](https://github.com/evcc-io/evcc/files/12076650/log.txt) angeschaut.
Das sieht erstmal alles korrekt aus. Easee quitttert auch den Ladestart als akzeptiert. Warum es dann trotzdem nicht los geht weiß vermutlich nur Easee.... 🤷‍♂️

SuperDa75, 2023-07-18:
Ich erhalte seit Tagen keine 401/429 mehr angezeigt. Allerdings startet der Ladevorgang im PV Modus wieder mit 11kW, dann Ladepause, um danach mit Überschuss weiter zu laden.