evcc-io/evcc #8917: Add charges and tax to Tibber tariff

kalaws, 2023-07-11:
Would be great to be able to add `charges` and `tax` also to `tibber` dynamic tariff.

naltatis, 2023-07-11:
The Tibber API already returns prices including charges and taxes. At lease in Germany there is nothing else that needs to be added.

kalaws, 2023-07-11:
In Sweden most charges are left out, unfortunately.

naltatis, 2023-07-11:
Ok, thanks. Then we'll have to add this. But using the `total` price as a basis for adding `tax` and `charges` works for you, right?

The Tibber API doc states that this field contains `The total price (energy + taxes)`.

kalaws, 2023-07-11:
Yes, I guess a fixed addition should do the trick. However, I also use tibber for `feedin`, and in this case it would actually be great to do a negative multiplication, since we don’t get VAT for feedin.

andig, 2023-07-11:
> in this case it would actually be great to do a negative multiplication

Final price is calculated as

    return (price + t.Charges) * (1 + t.Tax)

closed 2023-07-11 by andig

paschdan, 2024-01-01:
hi, i would like to add my 2ct here, because yesterday i had some problems with tibber api and price retrieval (they were late). 

so i changed to awattar and added my charges, unfortunately this does not work to get the real price, because of the following reason:

when you calculate the final price, then it is:  `return (price + t.Charges) * (1 + t.Tax)`

but tibber does calculate it differently:

the charges of tibber does already include the tax, so the calculation would be 

` return (( price * ( 1+t.Tax)) + t.Charges)`

@andig do we can somehow achieve the above calculation?

paschdan, 2024-01-02:
Hello, i got it, i just can divide my real Charges by 1.19 and use this value for charges in the config. math is a beast :D