evcc-io/evcc #8933: unexpected status: 401 (Unauthorized)

Gumma74, 2023-07-12:
### Describe the bug

Seit Version 0.118.7 bekomme ich folgende Fehlermeldungen:

[image]

und SOC wird nicht korrekt ausgelesen.

Gehe ich zurück auf die vormals installierte Version 0.118.1 ist der Fehler wieder weg.

### Steps to reproduce

1. Version 0.118.7 oder .8 installieren, Volvo XC40 Pure Electric anstecken.

### Configuration details

```yaml
interval: 5s # control cycle interval

sponsortoken: ***
plant: ***
log: info
telemetry: true

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3759 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.07 # [currency]/kWh

meters:
- type: template
  template: solaredge-hybrid
  usage: grid 
  modbus: tcpip
  host: 192.168.178.40  # IP-Adresse oder Hostname
  port: 1502  # Port 502 (SetApp) oder 1502 (LCD)
  timeout: 3s 
  name: grid1
- type: template
  template: solaredge-hybrid
  usage: pv
  modbus: tcpip  
  host: 192.168.178.40  # IP-Adresse oder Hostname
  port: 1502  # Port 502 (SetApp) oder 1502 (LCD)
  timeout: 3s 
  name: pv1 
  id: 1 

vehicles:
- name: XC40
  type: template
  template: volvo-connected
  title: XC40  # Wird in der Benutzeroberfläche angezeigt
  user: ***  # Benutzername um auf das Gerät zuzugreifen
  password: ***  # Passwort des Benutzerkontos
  vccapikey: ***
  capacity: 75  # Akku-Kapazität in kWh
  targetSoC: 90
  phases: 3
  priority: 1
  
- name: e-Golf
  type: template
  template: vw
  title: e-Golf  # Wird in der Benutzeroberfläche angezeigt
  user: ***  # Benutzername um auf das Gerät zuzugreifen
  password: ***  # Passwort des Benutzerkontos
  capacity: 36  # Akku-Kapazität in kWh
  targetSoC: 90
  phases: 2
  priority: 0
  
chargers:
- type: template
  template: abl
  id: 1 
  host: 192.168.178.201  # IP-Adresse oder Hostname
  port: 502  # Port
  rtu: true 
  modbus: rs485tcpip 
  name: wallbox1
  
- type: template
  template: abl
  id: 2 
  host: 192.168.178.201  # IP-Adresse oder Hostname
  port: 502  # Port
  rtu: true 
  modbus: rs485tcpip 
  name: wallbox2
  
loadpoints:
- title: Carport Golf
  charger: wallbox1
  vehicle: e-Golf
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable:
    threshold: -1480
    delay: 1m
  disable:
    threshold: 1480
    delay: 1m
  #guardduration: 10m 
  soc:
   estimate: true
   poll:
     mode: connected
     interval: 30m
       
- title: Carport Volvo
  charger: wallbox2
  vehicle: XC40
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable:
    threshold: -2170
    delay: 1m
  disable:
    threshold: 2170
    delay: 1m  
  #guardduration: 10m
  soc:
   estimate: true
   poll:
     mode: connected
     interval: 30m
      
site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv: pv1    
  residualPower: 0
```

### Log details

```text
n.a.
```

### What type of operating system are you running?

Docker container

### Version

0.118.7

naltatis, 2023-07-12:
See https://github.com/evcc-io/evcc/discussions/8757 for details.

Please update to 0.118.8

naltatis, 2023-07-12:
Ok, zu schnell gelesen. Hier gehts um Volvo und mal nicht um Easee :)

andig, 2023-07-12:
Missing logfile.

closed 2023-07-12 by andig

verhaeghep, 2023-07-12:
2023-07-12T11:02:14.130551+02:00 zenna systemd[1]: Started evcc.service - evcc.
2023-07-12T11:02:14.209954+02:00 zenna evcc[926305]: [main  ] INFO 2023/07/12 11:02:14 evcc 0.118.8
2023-07-12T11:02:14.210112+02:00 zenna evcc[926305]: [main  ] INFO 2023/07/12 11:02:14 using config file: /etc/evcc.yaml
2023-07-12T11:02:14.210619+02:00 zenna evcc[926305]: [main  ] INFO 2023/07/12 11:02:14 starting ui and api at :7070
2023-07-12T11:02:14.355645+02:00 zenna evcc[926305]: [db    ] INFO 2023/07/12 11:02:14 using sqlite database: /var/lib/evcc/evcc.db
2023-07-12T11:02:16.519462+02:00 zenna evcc[926305]: [semp  ] WARN 2023/07/12 11:02:16 SEMP_BASE_URL unspecified, using http://192.168.2.6:7070 instead
2023-07-12T11:02:16.520558+02:00 zenna evcc[926305]: [site  ] INFO 2023/07/12 11:02:16 site config:
2023-07-12T11:02:16.520635+02:00 zenna evcc[926305]: [site  ] INFO 2023/07/12 11:02:16   meters:      grid ✓ pv ✗ battery ✗
2023-07-12T11:02:16.520677+02:00 zenna evcc[926305]: [site  ] INFO 2023/07/12 11:02:16     grid:      power ✓ energy ✓ currents ✓
2023-07-12T11:02:16.520721+02:00 zenna evcc[926305]: [site  ] INFO 2023/07/12 11:02:16   vehicles:
2023-07-12T11:02:16.520759+02:00 zenna evcc[926305]: [site  ] INFO 2023/07/12 11:02:16     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
2023-07-12T11:02:16.520795+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:16 loadpoint 1:
2023-07-12T11:02:16.521084+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:16   mode:        pv
2023-07-12T11:02:16.521513+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:16   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
2023-07-12T11:02:16.521565+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:16   meters:      charge ✓
2023-07-12T11:02:16.521612+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:16     charge:    power ✓ energy ✓ currents ✓
2023-07-12T11:02:16.521661+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:16 vehicle updated: unknown -> XC40
2023-07-12T11:02:17.277930+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:17 car connected
2023-07-12T11:02:27.510240+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:02:27 start charging ->
2023-07-12T11:24:27.397470+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:24:27 stop charging <-
2023-07-12T11:24:37.251233+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:24:37 start charging ->
2023-07-12T11:30:27.396361+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:30:27 stop charging <-
2023-07-12T11:30:37.250099+02:00 zenna evcc[926305]: [lp-1  ] INFO 2023/07/12 11:30:37 start charging ->
2023-07-12T11:45:37.485139+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:45:37 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:45:47.309927+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:45:47 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:45:57.271707+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:45:57 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:46:07.363195+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:46:07 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:46:17.255573+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:46:17 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:46:27.257441+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:46:27 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:46:37.362538+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:46:37 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:46:47.248556+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:46:47 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:46:57.251186+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:46:57 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:47:07.269737+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:47:07 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:47:17.272584+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:47:17 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:47:28.131984+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:47:28 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:47:37.259713+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:47:37 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:47:47.251194+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:47:47 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:47:57.254750+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:47:57 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:48:07.257414+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:48:07 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:48:17.271029+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:48:17 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:48:27.248699+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:48:27 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:48:37.251335+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:48:37 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:48:47.250961+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:48:47 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:48:57.293314+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:48:57 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:49:07.257142+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:49:07 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T11:49:17.260819+02:00 zenna evcc[926305]: [lp-1  ] ERROR 2023/07/12 11:49:17 vehicle soc: unexpected status: 401 (Unauthorized)

verhaeghep, 2023-07-12:
2023-07-12T12:41:07.788309+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:07 ----
2023-07-12T12:41:07.788436+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:07 modbus: send 04 d9 00 00 00 06 01 03 01 58 00 02
2023-07-12T12:41:07.909141+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:07 modbus: recv 04 d9 00 00 00 07 01 03 04 46 2b f1 33
2023-07-12T12:41:07.909251+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:07 charge power: 11004W
2023-07-12T12:41:07.909323+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:07 grid power: -15952W
2023-07-12T12:41:07.909363+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:07 grid powers: [-6279 -3437 -6236]W
2023-07-12T12:41:07.909405+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:07 grid currents: [-25.7 -14.5 -25.3]A
2023-07-12T12:41:07.909445+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:07 pv power: 27957W
2023-07-12T12:41:07.909486+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:07 site power: -14952W
2023-07-12T12:41:07.909538+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:07 modbus: send 04 da 00 00 00 06 01 03 01 40 00 06
2023-07-12T12:41:08.034119+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: recv 04 da 00 00 00 0f 01 03 0c 41 71 fb e8 41 6c d0 e6 41 73 fb e8
2023-07-12T12:41:08.034275+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:08 charge currents: [15.1 14.8 15.2]A
2023-07-12T12:41:08.034321+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:08 detected active phases: 3p
2023-07-12T12:41:08.034481+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: send 04 db 00 00 00 06 01 03 01 76 00 04
2023-07-12T12:41:08.159817+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: recv 04 db 00 00 00 0b 01 03 08 41 4b 6d 42 00 00 00 00
2023-07-12T12:41:08.159970+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: send 04 dc 00 00 00 06 01 03 01 76 00 04
2023-07-12T12:41:08.286744+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: recv 04 dc 00 00 00 0b 01 03 08 41 4b 6d 43 80 00 00 00
2023-07-12T12:41:08.286846+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:08 charge total import: 3594.887kWh
2023-07-12T12:41:08.286939+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: send 04 dd 00 00 00 06 01 03 04 b1 00 05
2023-07-12T12:41:08.412851+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: recv 04 dd 00 00 00 0d 01 03 0a 43 32 00 00 00 00 00 00 00 00
2023-07-12T12:41:08.413017+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:08 charger status: C
2023-07-12T12:41:08.414020+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:08 vehicle soc: 81%
2023-07-12T12:41:08.414100+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:08 vehicle range: 290km
2023-07-12T12:41:08.414145+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: send 04 de 00 00 00 06 01 03 04 ba 00 02
2023-07-12T12:41:08.537753+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:08 modbus: recv 04 de 00 00 00 07 01 03 04 42 00 00 00
2023-07-12T12:41:08.537916+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:08 pv charge current: 38.9A = 17.2A + 21.7A (-14952W @ 3p)
2023-07-12T12:41:17.176979+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:17 modbus: send 04 df 00 00 00 0b 01 10 04 ba 00 02 04 42 00 00 00
2023-07-12T12:41:17.287382+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:17 modbus: recv 04 df 00 00 00 06 01 10 04 ba 00 02
2023-07-12T12:41:17.712260+02:00 zenna evcc[933385]: [semp  ] TRACE 2023/07/12 12:41:17 send: {XMLName:{Space: Local:} Xmlns:urn:schemas-upnp-org:device-1-0 SpecVersion:{Major:1 Minor:0} Device:{DeviceType:urn:schemas-simple-energy-management-protocol:device:Gateway:1 FriendlyName:evcc Manufacturer:github.com/evcc-io/evcc ModelName:EVCC SEMP Server 0.118.8 UDN:uuid:70ee4183-209c-11ee-8d4a-000c29444233 PresentationURL:http://192.168.2.6:7070 ServiceDefinition:{Xmlns:urn:schemas-simple-energy-management-protocol:service-1-0 Server:http://192.168.2.6:7070 BasePath:/semp Transport:HTTP/Pull ExchangeFormat:XML WsVersion:1.1.0} ServiceList:[]}}
2023-07-12T12:41:17.788529+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:17 ----
2023-07-12T12:41:17.788637+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:17 modbus: send 04 e0 00 00 00 06 01 03 01 58 00 02
2023-07-12T12:41:17.912429+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:17 modbus: recv 04 e0 00 00 00 07 01 03 04 46 2c 52 67
2023-07-12T12:41:17.912564+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:17 charge power: 11029W
2023-07-12T12:41:17.912639+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:17 grid power: -15785W
2023-07-12T12:41:17.912695+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:17 grid powers: [-6217 -3312 -6256]W
2023-07-12T12:41:17.912739+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:17 grid currents: [-25.3 -13.9 -25.3]A
2023-07-12T12:41:17.912775+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:17 pv power: 27814W
2023-07-12T12:41:17.912817+02:00 zenna evcc[933385]: [site  ] DEBUG 2023/07/12 12:41:17 site power: -14785W
2023-07-12T12:41:17.912859+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:17 modbus: send 04 e1 00 00 00 06 01 03 01 40 00 06
[... cut ...]
2023-07-12T12:41:58.411036+02:00 zenna evcc[933385]: [lp-1  ] ERROR 2023/07/12 12:41:58 vehicle soc: unexpected status: 401 (Unauthorized)
2023-07-12T12:41:58.411082+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:58 modbus: send 04 fe 00 00 00 06 01 03 04 ba 00 02
2023-07-12T12:41:58.535799+02:00 zenna evcc[933385]: [alfen ] TRACE 2023/07/12 12:41:58 modbus: recv 04 fe 00 00 00 07 01 03 04 42 00 00 00
2023-07-12T12:41:58.535904+02:00 zenna evcc[933385]: [lp-1  ] DEBUG 2023/07/12 12:41:58 pv charge current: 38.4A = 17.2A + 21.2A (-14626W @ 3p)

andig, 2023-07-12:
Feel free to email testing credentials to info@evcc.io

Gumma74, 2023-07-12:
@andig 

Das Schließen meines Bug-Reports mit dem schlichten Kommentar „Missing logfile“ macht auf mich keinen besonders interessierten Eindruck. Da stellt sich mir die Frage, warum alle User, welche nicht über so viel geschätztes IT-Knowhow verfügen wie du, überhaupt Bug-Reports schrieben dürfen, wenn formale Fehler dazu führen, dass der Report achtlos ignoriert wird. 

Ich habe versucht, den Fehler so gut es für mich möglich war zu beschrieben. Zur Erstellung eines Logfiles war und bin in Ermangelung von IT-Kenntnissen schlicht nicht in der Lage. Mea culpa, mea culpa, mea maxima culpa! 

Aber ehrlicherweise vergeht mir die Lust, mich einzubringen, wenn ein Bugreport auf diese Art und Weise einfach weggebügelt wird. Mag sein, dass es nachvollziehbare Gründe gibt, die Problematik nicht weiter zu verfolgen. Etwas freundlichere Umgangsformen würden dir und deinem Open-Source Projekt trotzdem gut zu Gesichte stehen. Aber vielleicht romantisiere ich zu sehr und der raue Ton einer Schiffskombüse ist der Standard.

Nicht falsch verstehen: Ich finde das Gesamtprojekt genial, weshalb ich auch ein wenig mehr bezahle, als für einen Sponsor-Token erforderlich wäre. Bug-Reports sind von mir nicht mehr zu erwarten. Nach deiner „Rückmeldung“ bleibt mir nur festzustellen, dass mir meine Zeit zu schade ist, unbeachtete Bug-Reports und danach vermutlich ebenso unbeachtete Beschwerde-Posts zu schreiben und dass Version 0.118.1 für mich okay ist.

andig, 2023-07-13:
> Das Schließen meines Bug-Reports mit dem schlichten Kommentar „Missing logfile“ macht auf mich keinen besonders interessierten Eindruck. 

> Da stellt sich mir die Frage... Mea culpa... ehrlicherweise vergeht mir die Lust... dass mir meine Zeit zu schade ist...

Ich spreche jetzt mal ganz offen: jedesmal die gleiche Leier. Diese Diskussion verschwendet einfach nur unsere Zeit und knappen Ressourcen. Die Regeln für ein Issue sind beim Anlegen ganz klar beschrieben:

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

Sollte das mißverständlich sein sag mir gerne inwiefern.

> Zur Erstellung eines Logfiles war und bin in Ermangelung von IT-Kenntnissen schlicht nicht in der Lage.

Das ist schade, aber auch nicht relevant. Denn ohne Logfile und mangels Glaskugel kann ich dann nicht helfen. Schließen des Issues ist absolut foglerichtig da es in diesem Fall nichts zu tun gibt.

Wie Du Dir selbst helfen kannst habe ich oben geschrieben:

> Feel free to email testing credentials to [info@evcc.io](mailto:info@evcc.io)

Das ist bisher nicht geschehen, stattdessen hier die Grundsatzdiskussion. Mein Angebot steht weiter. Da es hier nichts für mich zu tun gibt close.