evcc-io/evcc #8939: Konfigurator klarer definieren

VolkerK62, 2023-07-12:
Du versuchst ein "PV-System" hinzuzufügen obwohl du offenbar nur einen einzelnen SE-Wechselrichter hast.
Das ist im Konfigurator leider etwas missverständlich. Ein komplettes "PV-System" beinhaltet u.a. auch ein zugehöriges Smartmeter am Netzübergang. Dies hast du offensichtlich nicht, daher kann dies nicht angesprochen werden.

Lösung: Du musst im ersten Schritt "Mein Gerät ist nicht in der Liste" auswählen und erst dann dort den SolarEdge Wechselrichter.

Alternativ kannst du auch einfach die Rückfrage mit N beantworten. Der Wechselrichter selbst sollte dann korrekt hinzugefügt werden und das gridmeter weggelassen werden.

```
 In welchem Modus soll die Konfiguration durchgeführt werden? Standard Modus (So einfach und schnell wie möglich)

- Hausinstallation einrichten

Wähle eines der folgenden PV Komplettsysteme aus, oder 'Mein Gerät ist nicht in der Liste' falls keines dieser Geräte vorhanden ist

 Wähle ein PV System: Mein Gerät ist nicht in der Liste

 Möchtest du einen Netz-Stromzähler hinzufügen? No

 Möchtest du einen PV Wechselrichter (oder entsprechender Stromzähler) hinzufügen? Yes

- Konfiguration PV Wechselrichter (oder entsprechender Stromzähler)

 Wähle einen PV Wechselrichter (oder entsprechender Stromzähler): SolarEdge Wechselrichter

-------------------------------------------------
Das Gerät hat die folgenden Voraussetzungen:
Nur ein System kann und darf auf den Wechselrichter zugreifen!
-------------------------------------------------

Führe folgende Einstellungen durch:

 Wähle die ModBus Schnittstelle aus TCP/IP
 Modbus ID 2
 IP-Adresse oder Hostname 192.168.178.20
 Port 502

Teste die pv Konfiguration von SolarEdge Wechselrichter ...
```

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/8932#discussioncomment-6425525_

VolkerK62, 2023-07-12:
>Das ist im Konfigurator leider etwas missverständlich. Ein komplettes "PV-System" beinhaltet u.a. auch ein zugehöriges Smartmeter am Netzübergang

Da bin ich auch schon drauf reingefallen.
Könnte man das evtl eindeutiger beschreiben?

"Wähle eines der folgenden PV Komplettsysteme (**Wechselsrichter UND Grid-Meter**) aus, oder 'Mein Gerät ist nicht in der Liste' falls keines dieser Geräte vorhanden ist **oder Einzelkomponenten konfiguriert werden sollen**"

andig, 2023-07-12:
Wir brauchen also einen gute Begriff für "PV Komplettsystem". Was ist ein PV Komplettsystem per Definition? WR&Grid Meter trifft es m.E. auch nicht. Etwas "integriertes"?

/cc @DerAndereAndi @premultiply

andig, 2023-07-22:
Gibts hier noch Ideen? Sonst würde ich zu machen.

VolkerK62, 2023-07-22:
"Wähle eines der folgenden integrierten PV-Systeme aus, oder 'Mein Gerät ist nicht in der Liste' falls keines dieser Geräte vorhanden ist oder Einzelkomponenten konfiguriert werden sollen"

andig, 2023-07-22:
Also: 

    Flow_NewConfiguration_Select = "Wähle eines der folgenden integrierten PV Systeme aus, oder '{{ .ItemNotPresent }}' falls es sich nicht um ein integriertes PV System handelt"

Was ich nicht finde ist die Auswahl:

     Wähle ein PV System

@DerAndereAndi hast Du eine Idee, wo dieser String her kommt? Hier müsste dann auch "integriertes" mit rein.

DerAndereAndi, 2023-07-22:
```toml
Choose = "Wähle"

Category_SystemTitle = "PV System"
Category_SystemArticle = "ein"
```

andig, 2023-07-23:
Siehe PR- so ok?

closed 2023-07-23 by andig