evcc-io/evcc #8956: panic with ocpp

musbach, 2023-07-14:
### Describe the bug

Wenn ich im yaml den `idtag` mit auskommentiere, erhalte ich ein panic und die Box verbindet sich nicht mit evcc. 

```
chargers:
- type: template
  template: ocpp
  name: evbox
  metervalues: Current.Import.L1,Current.Import.L2,Current.Import.L3,Energy.Active.Import.Register,Power.Active.Import,Voltage.L1,Voltage.L2,Voltage.L3
  meterInterval: 10s
  # stationId: EVB-Pnnnnnnnn
  # connector: 1
  # idtag: NL-EVB-Cnnnnnnnnnn-E
  # connecttimeout: 5m
  # timeout: 5m
```

Wenn ich den Tag wieder aktiviere, tritt der Fehler nicht auf.
Ich bin auf `Ubuntu 22.04.2 LTS`.

[trace10.zip](https://github.com/evcc-io/evcc/files/12047578/trace10.zip)

### Steps to reproduce

1. ohne ittag: panic
2. mit idtag läuft es
3. wöchentliches update von Ubuntu: 
<details><summary>Details des Upgrades</summary>
<p>
`Upgrade: libcanberra-gtk3-0:amd64 (0.30-10ubuntu1, 0.30-10ubuntu1.22.04.1), libxtables12:amd64 (1.8.7-1ubuntu5, 1.8.7-1ubuntu5.1), libidn11-dev:amd64 (1.3    8-4build1, 1.38-4ubuntu1), libmm-glib0:amd64 (1.20.0-1~ubuntu22.04.1, 1.20.0-1~ubuntu22.04.2), iptables:amd64 (1.8.7-1ubuntu5, 1.8.7-1ubuntu5.1), libglx-m    esa0:amd64 (22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0ubuntu0.1~22.04.3), libidn12:amd64 (1.38-4build1, 1.38-4ubuntu1), modemmanager:amd64 (1.20.0-1~ubuntu22.04.    1, 1.20.0-1~ubuntu22.04.2), libcanberra-gtk3-module:amd64 (0.30-10ubuntu1, 0.30-10ubuntu1.22.04.1), tzdata:amd64 (2023c-0ubuntu0.22.04.1, 2023c-0ubuntu0.2    2.04.2), idn:amd64 (1.38-4build1, 1.38-4ubuntu1), osinfo-db:amd64 (0.20220214-1ubuntu2.1, 0.20230518-1ubuntu0.22.04.1), libgbm1:amd64 (22.2.5-0ubuntu0.1~2    2.04.2, 22.2.5-0ubuntu0.1~22.04.3), grub-pc-bin:amd64 (2.06-2ubuntu7.1, 2.06-2ubuntu7.2), libfwupd2:amd64 (1.7.9-1~22.04.1, 1.7.9-1~22.04.3), ubuntu-drive    rs-common:amd64 (1:0.9.6.1, 1:0.9.6.2~0.22.04.4), libinput10:amd64 (1.20.0-1ubuntu0.2, 1.20.0-1ubuntu0.3), libip4tc2:amd64 (1.8.7-1ubuntu5, 1.8.7-1ubuntu5    .1), appstream-util:amd64 (0.7.18-2, 0.7.18-2ubuntu1), libxatracker2:amd64 (22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0ubuntu0.1~22.04.3), mesa-va-drivers:amd64 (    22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0ubuntu0.1~22.04.3), libgl1-mesa-dri:amd64 (22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0ubuntu0.1~22.04.3), libappstream-glib8:am    d64 (0.7.18-2, 0.7.18-2ubuntu1), libidn-dev:amd64 (1.38-4build1, 1.38-4ubuntu1), fwupd-signed:amd64 (1.51~22.04.1+1.2-3ubuntu0.2, 1.51.1~22.04.1+1.4-0ubun    tu0.1), autopkgtest:amd64 (5.20ubuntu1, 5.28ubuntu0.22.04.0), mesa-vulkan-drivers:amd64 (22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0ubuntu0.1~22.04.3), libip6tc2:    amd64 (1.8.7-1ubuntu5, 1.8.7-1ubuntu5.1), grub2-common:amd64 (2.06-2ubuntu7.1, 2.06-2ubuntu7.2), libglapi-mesa:amd64 (22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0u    buntu0.1~22.04.3), grub-common:amd64 (2.06-2ubuntu7.1, 2.06-2ubuntu7.2), ubuntu-standard:amd64 (1.481, 1.481.1), libcanberra-pulse:amd64 (0.30-10ubuntu1,     0.30-10ubuntu1.22.04.1), libegl-mesa0:amd64 (22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0ubuntu0.1~22.04.3), libfwupdplugin5:amd64 (1.7.9-1~22.04.1, 1.7.9-1~22.04.    3), mesa-vdpau-drivers:amd64 (22.2.5-0ubuntu0.1~22.04.2, 22.2.5-0ubuntu0.1~22.04.3), grub-pc:amd64 (2.06-2ubuntu7.1, 2.06-2ubuntu7.2), libcanberra0:amd64     (0.30-10ubuntu1, 0.30-10ubuntu1.22.04.1), fwupd:amd64 (1.7.9-1~22.04.1, 1.7.9-1~22.04.3), ubuntu-minimal:amd64 (1.481, 1.481.1), libinput-bin:amd64 (1.20.    0-1ubuntu0.2, 1.20.0-1ubuntu0.3)
`
</p>
</details> 

5. Ohne `idtag` tritt nun zwar kein panic mehr auf aber es braucht 160 Restarts, bis sich evcc mit dem Charger komplett verbindet. Der Charger antwortet zwar mit heartbeats und liefert auch die ersten Daten, aber es führt immer wieder zu timeouts, egal wie hoch ich sie stelle. Ich wüßte gern, welche Lib dafür verantworlich ist, um den Fehler nachzustellen.

### Configuration details

```yaml
# encoding: utf-8
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: info
  opcc: trace
  charger: trace

# unique installation id
plant: xxxx
sponsortoken: xxxxx
interval: 10s # control cycle interval

meters:
- type: template
  template: kostal-piko
  usage: grid
  host: 192.168.2.90
  name: grid1
- type: template
  template: kostal-piko
  usage: pv
  host: 192.168.2.90
  name: pv2
- type: template
[... 101 lines cut ...]
[main ] INFO 2023/07/12 08:27:45 using config file: /etc/evcc.yaml
[main ] INFO 2023/07/12 08:27:45 starting ui and api at :7070
[db ] INFO 2023/07/12 08:27:45 using sqlite database: /var/lib/evcc/evcc.db
[db ] TRACE 2023/07/12 08:27:45 SELECT count() FROM sqlite_master WHERE type='table' AND name="settings" -1
[db ] TRACE 2023/07/12 08:27:45 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1
[db ] TRACE 2023/07/12 08:27:45 SELECT * FROM settings LIMIT 1 -1
[db ] TRACE 2023/07/12 08:27:45 SELECT * FROM settings 6
[ocpp ] DEBUG 2023/07/12 08:27:46 waiting for chargepoint: 5m0s
[ocpp ] INFO 2023/07/12 08:27:50 chargepoint connected, registering: EVB-PXXX
[ocpp ] TRACE 2023/07/12 08:27:50 sent JSON message to EVB-PXXX: [2,"3688170230","GetConfiguration",{}]
[ocpp ] TRACE 2023/07/12 08:28:35 received JSON message from EVB-PXXX: [2,"2c410493-df11-4675-b7f9-40efb4fccd60","Heartbeat",{}]
[ocpp ] TRACE 2023/07/12 08:28:35 sent JSON message to EVB-PXXX: [3,"2c410493-df11-4675-b7f9-40efb4fccd60",{"currentTime":"2023-07-12T06:28:35Z"}]
[main ] FATAL 2023/07/12 08:28:50 cannot create charger 'evbox': cannot create charger 'template': cannot create charger 'ocpp': ocpp message (3688170230): GenericError - Request timed out
[main ] FATAL 2023/07/12 08:28:50 will attempt restart in: 5m0s
[ocpp ] TRACE 2023/07/12 08:29:35 received JSON message from EVB-PXXX: [2,"2c4104cf-df11-4675-b7f9-40ef90b76ec6","Heartbeat",{}]
[ocpp ] TRACE 2023/07/12 08:29:35 sent JSON message to EVB-PXXX: [3,"2c4104cf-df11-4675-b7f9-40ef90b76ec6",{"currentTime":"2023-07-12T06:29:35Z"}]
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.8

andig, 2023-07-16:
Ich hab den Überblick verloren. Geht es hier jetzt um ein panic in der aktuellen Version oder um was Anderes?

musbach, 2023-07-17:
Nach dem Update ist der Panic-Fehler verschwunden. Was ich nicht verstehe ist, warum der Fehler überhaupt auftrat, denn in den vorherigen evcc-Versionen trat kein derartiger Fehler auf.

andig, 2023-07-17:
Da unklar ist, was vor/nach dem Update ist und anscheinend kein Fehler besteht machen wir dann mal zu.