evcc-io/evcc #8963: Set feedin price via MQTT

kalaws, 2023-07-15:
Would be great to be able to set `feedin` price via mqtt. I’m currently using `tibber` tariff for `feedin`, but the price returned includes VAT and additional costs. While VAT is always the same and can easily be redacted, the additional costs which Tibber tacks onto the price tend to vary. Therefore it is difficult to set the correct feedin price. The correct feedin price would, in Sweden, be the current spot rate (eg, right now 0,3 SEK)+subsidy (0,6 SEK from state, some 0,3 SEK from net owner).

AFAIK, feedin is not factored in into the planner, so we don’t really need a list of prices, only the current price. Therefore mqtt should suffice. Perhaps in the future the planner feature considers whether charge or feedin is optimal at a given time, based partly on pv production forecast, but this seems rather far off?

andig, 2023-07-16:
> AFAIK, feedin is not factored in into the planner, so we don’t really need a list of prices, only the current price. 

That doesn't help, all tariffs have the same signature. Alternative: we have this tibber structure:

type Price struct {
	Currency    string
	StartsAt    time.Time
	Total       float64
	Energy, Tax float64
	// Level    string
}

We could use a different field (which one?) and add configurable charges+taxes.

kalaws, 2023-07-16:
Hmm, unfortunately I’m not versed enough in code to interpret the implications of such a `tibber` structure. Or was that a typo, and you suggested a new type, aside from tibber, awattar, octopus etc, called `Price`? If so, that sounds reasonable to me.

andig, 2023-07-16:
My question is only if the tibber api gives you the required base cost. If not taxes and charges won't help.

kalaws, 2023-07-17:
Oh. Their `Price.total` is incorrect, because it adds taxes and charges that do not apply at feedin. I think their api gives access to spot rate (the cost of energy). That would be `Price.energy`? So if it’s possible to chose between `total` and `energy` in the above structure, and then add charges and tax, all is well.

kalaws, 2023-07-17:
Hmm, where would I set that? Console? evcc.yaml?

kalaws, 2023-07-17:
Got it, log level. 

So perhaps as follows:

```
type Price struct {
	Currency    string
	StartsAt    time.Time
	Energy float64
	// Level    string
}
```

kalaws, 2023-07-17:
@andig please reopen if the above makes sense.

andig, 2023-07-17:
It doesn't. Please look at the logfile for tibber and decide which field has the correct cost.

kalaws, 2023-07-17:
Right, so here’s the log:

```
[tibber] TRACE 2023/07/17 21:26:42 {"query":"{viewer{homes{id,timeZone,address{address1,postalCode,city,country},meteringPointData{gridCompany},currentSubscription{id,status,priceInfo{current{currency,startsAt,total,energy,tax},today{currency,startsAt,total,energy,tax},tomorrow{currency,startsAt,total,energy,tax}}}}}}"}
--
{"data":{"viewer":{"homes":[{"id":"***","timeZone":"Europe/Stockholm","address":{"address1":"***","postalCode":"***","city":"***","country":"SE"},"meteringPointData":{"gridCompany":"***"},"currentSubscription":{"id":"***","status":"running","priceInfo":{"current":{"currency":"SEK","startsAt":"2023-07-17T21:00:00.000+02:00","total":0.3403,"energy":0.1873,"tax":0.153},"today":[{"currency":"SEK","startsAt":"2023-07-17T00:00:00.000+02:00","total":0.1061,"energy":0,"tax":0.1061},{"currency":"SEK","startsAt":"2023-07-17T01:00:00.000+02:00","total":0.1061,"energy":0,"tax":0.1061},{"currency":"SEK","startsAt":"2023-07-17T02:00:00.000+02:00","total":0.1061,"energy":0,"tax":0.1061}, {"currency":"SEK","startsAt":"2023-07-17T03:00:00.000+02:00","total":0.1063,"energy":0.0001,"tax":0.1062},{"currency":"SEK","startsAt":"2023-07-17T04:00:00.000+02:00","total":0.1204,"energy":0.0114,"tax":0.109},{"currency":"SEK","startsAt":"2023-07-17T05:00:00.000+02:00","total":0.152,"energy":0.0367,"tax":0.1153},{"currency":"SEK","startsAt":"2023-07-17T06:00:00.000+02:00","total":0.172,"energy":0.0527,"tax":0.1193},{"currency":"SEK","startsAt":"2023-07-17T07:00:00.000+02:00","total":0.1924,"energy":0.069,"tax":0.1234},
…
```

In this case, `energy` returns the correct spot rate. See attached image, which verifies this, at 04 AM. 

[image]

andig, 2023-07-17:
Then lets do this: add charges+taxes and if either is != 0 use `energy` instead of `price`.

andig, 2023-07-17:
@kalaws it would have helped if you did mention https://github.com/evcc-io/evcc/issues/8917. We're gonna break that behaviour now.

closed 2023-07-17 by andig