evcc-io/evcc #8975: Geänderter Fahrzeug SoC wird nur nach Wechsel des Fahrzeugs in der UI aktualisiert

binderth, 2023-07-16:
### Describe the bug

Es wird der zuletzt bekannte SoC eines Fahrzeugs so lange verwendet, bis in der UI manuell das Fahrzeug gewechselt und zurückgewechselt wird.
Dies gilt sowohl für http als auch mqtt als Quelle, direktem API-Zugriff hab ich nicht getestet.

ist natürlich doof, wenn das Fahrzeug mit 80% unterwegs war und man es einfach wieder ansteckt.

### Steps to reproduce

1. Auto abstecken bei bekanntem SoC
2. man kann in evcc währenddessen sogar Fahrzeuge wechseln
3. Auto wieder anstecken
4. der unter 1. bekannte SoC wird weiter verwendet, auch wenn er sich inzwischen geändert hat

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http

  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. Host must be set to the docker host's name
  #host: evcc.local
  host: 0.0.0.0

  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

  # per default, evcc should be available at http://evcc.local:7070

interval: 10s # control cycle interval
log: trace

meters:
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
- type: custom
  name: PVErzeugung
  power:
    source: mqtt
    topic: devices/Kostal/totalPVPower
    timeout: 2m
#  energy:
#    source: mqtt
#    topic: devices/Kostal/totalHomeConsumptionPV
#    timeout: 2m
- type: custom
  name: Netzbezug
  power:
    source: mqtt
    topic: devices/Kostal/totalActivePower
    timeout: 2m
#  energy:
#    source: mqtt
#    topic: devices/Kostal/totalHomeConsumptionGrid
#    timeout: 2m
- type: custom
  name: BYD10
  power:
    source: mqtt
    topic: devices/Kostal/actualBatteryChargeDischargePower
    timeout: 2m
  soc:
    source: mqtt
    topic: devices/Kostal/batteryActualSoc
    timeout: 2m

chargers:
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
- type: template
  template: tinkerforge-warp
  #fw2: true
  host: 192.168.78.18
  port: 1883
  topic: devices/Warp2
  timeout: 30s
  name: WarpWallbox

vehicles:
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/evcc-io/config#vehicles
- name: 500e
  type: template
  template: offline
  title: Fiat 500e
  capacity: 22
- name: EV6
  type: custom
  title: Kia EV6 # display name for UI
  capacity: 77 # byttery capacity (kWh)
  soc: # battery soc (%)
    source: mqtt
    topic: devices/KiaEV6/batteryStatus
    timeout: 20m
  range: # optional electric range (km)
    source: mqtt
    topic: devices/KiaEV6/availableRange
    timeout: 20m
  status: # FahrzeugStatus (%)
    source: mqtt
    topic: devices/KiaEV6/vehicleStatusIEC
    timeout: 20m
  onIdentify: # set defaults when vehicle is identified
    mode: pv
    minSoC: 25
    targetSoC: 80

tariffs:
# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
  currency: EUR # (default EUR)
  grid:
    #type: fixed
    #price: 0.319 # [currency]/kWh
    type: tibber
    #cheap: 0.25
    token: "yyy"
    homeid: "yyy"
  feedin:
    type: fixed
    price: 0.065 # [currency]/kWh

loadpoints:
# loadpoint describes the charger, charge meter and connected vehicle
- title: Warp
  charger: WarpWallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  vehicle: 500e
  guardduration: 3m

site:
# site describes the EVU connection, PV and home battery
  title: Zuhause
  meters:
    grid: Netzbezug
    pvs:
    - PVErzeugung
    batteries:
    - BYD10
  prioritySoC: 66
  bufferSoC: 75
  residualPower: 600
  smartCostLimit: 0.15 # 15 ct

mqtt:
  broker: 192.168.78.18:1883
  clientid: evccClient
  user: m2m
  password: openHAB2m
  topic: servers/evcc
```

### Log details

```text
* https://1drv.ms/u/s!AhNoDZl5POyV3LII__hGoxWNM-J4mw?e=uNiNLg
```

### What type of operating system are you running?

Docker container

### Version

0.188.8

andig, 2023-07-16:
Beschreibung ist nicht klar. Was wäre hier das erwartete Verhalten in Abweichung zu welchem beobachteten?

binderth, 2023-07-16:
oh, sorry. das erwartete Verhalten wäre 7. ohne 4-6.

Also konkret: evcc kann jederzeit per HTTP den aktuellen SoC abfragen, bzw. erhält diesen in meinem Fall den Soc sowieso ständig per mqtt gepusht.
Die Erwartung ist: sobald der SoC des Fahrzeugs sich geändert hat, wird dieser auch zur Anzeige und vor allem zur Logik verwendet. Scheinbar verlässt sich evcc auf einen internen Cache.

Zumindest, wenn ein Fahrzeug angesteckt wird (auch das kriegt evcc ja mit, egal welcher Fahrzeug-Typ), sollte evcc den Cache invalidieren oder einfach nachsehen, ob sich was geändert hat.

mdkeil, 2023-07-16:
Standardmäßig aktualisiert evcc den soc nur während eines Ladevorgangs, was für die Steuerung ausreichend ist. Dieses Verhalten kann man via evcc.yaml anpassen.

VolkerK62, 2023-07-16:
Ich vermute, das Problem ist der Fiat als Standardfahrzeug. Deshalb erfolgt ein Fahrzeug Refresh erst beim manuellen Wechsel.

binderth, 2023-07-16:
Selbst wenn der Fiat das Standard-Fahrzeug ist. Ich ändere per API auf den EV6, wenn mein Smarthome erkennt, dass er lädt. Aber wie beschrieben war der EV6 ja bereits vorher angesteckt und evcc hat ihn noch immer "aktiv".

Und gerade bei mqtt erwarte ich eigtl schon eine Anpassung, wenn der SoC gesendet wird?

PS: @naltatis der Bug

binderth, 2023-07-16:
Und @mdkeil meinst du die Einstellungen zu `estimate`? Dachte, die gelten während des Ladens?

mdkeil, 2023-07-16:
> Und @mdkeil meinst du die Einstellungen zu `estimate`? Dachte, die gelten während des Ladens?

Ich hätte eher an poll.mode gedacht.

binderth, 2023-07-16:
> Ich hätte eher an poll.mode gedacht

Ah! Danke! Das probier ich dann auch mal aus!

andig, 2023-07-16:
> Die Erwartung ist: sobald der SoC des Fahrzeugs sich geändert hat, wird dieser auch zur Anzeige und vor allem zur Logik verwendet

siehe Doku poll.mode

closed 2023-07-16 by andig