evcc-io/evcc #8982: Goodwe EV Charger GW 11K-HCA Wallbox

DieterRudolph, 2023-07-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

DieterRudolph, 2023-07-16:
I installed the new Wallbox "Goodwe EV Charger GW 11K-HCA" to my Goodwe Hybrid Inverter GW10KNET. I red that the Converter was supported but I could not see the Wallbox. The Inverter and the Wallbox are controlled by Sems-Portal. Is it possible to include the Goodwe Wallbox Goodwe EV Charger GW 11K-HCA ?

andig, 2023-07-16:
@DieterRudolph happy to reopen if you can find any documentation

closed 2023-07-16 by andig

DieterRudolph, 2023-07-18:
Hi,
as documentation I found this:  [https://forum.iobroker.net/assets/uploads/files/1666333836822-goodwe_modbus_protocol_hybrid_et_eh_bh_bt__arm205-hv__v1.7-_-read-only_20200226-1.pdf](url)

DieterRudolph, 2023-07-18:
and this 
[https://community.openems.io/t/kommunikation-mit-goodwe-et-im-lokalen-netz/541](url)