evcc-io/evcc #8992: cannot create vehicle 'template': cannot create vehicle 'seat': - invalid character 'd'

leika65, 2023-07-17:
### Describe the bug

Seit heute kann EVCC meine Fahrzeug nicht mehr einbinden.
Gestern und die Tage davor ging es noch. Keine Änderung der EVCC version oder der Easee WB von mir gemacht.

[main  ] ERROR 2023/07/17 16:11:42 creating vehicle ev4 failed: 
cannot create vehicle 'template': cannot create vehicle 'seat':
invalid character 'd' looking for beginning of object key string

EVCC :
Software version : 0.118.8

Easee WB hat die firmware:
Software (v313)

### Steps to reproduce

1. new EVCC version
2. restart service
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: seat
  title: Cubra Leon ST
  user: xxx
  password: xxx
  vin: xxx
  capacity: 10.4
  name: ev4
```

### Log details

```text
[main  ] ERROR 2023/07/17 16:11:42 creating vehicle ev4 failed: 
cannot create vehicle 'template': cannot create vehicle 'seat':
invalid character 'd' looking for beginning of object key string
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.8

c-fy, 2023-07-17:
Hier exakt das selbe mit meinem E-Golf.

Bei Seat heißt das Pendant zu WE Connect doch Seat Connect, richtig? Funktioniert das bei dir, kannst du dich anmelden? WE Connect hat seit Tagen Probleme. Laut allestörungen.de werden auch bei Skoda Connect aktuell Störungen gemeldet.

Ich habe also die "Hoffung", dass es an dem VW-Konzern liegt.

Als Gastfahrzeug kann ich ihn aktuell laden.

`[main  ] ERROR 2023/07/17 16:35:09 creating vehicle egolf failed:
cannot create vehicle 'template': cannot create vehicle 'vw':
invalid character 'd' looking for beginning of object key string
.Soc:     vehicle not available: cannot create vehicle 'template':
cannot create vehicle 'vw':
invalid character 'd' looking for beginning of object key string`

leika65, 2023-07-17:
Hallo,
hast recht die Cubra Connect app geht auch nicht. Wobei man sich auf die Webseiten von Seat oder Cubra einloggen kann aber kein Fahrzeug findet.
Scheint eine Störung dort zu geben.

horstvogel, 2023-07-17:
Moin, kann ich bei meinem ID Buzz auch bestätigen. Bei mir hingegen funktioniert die VW App.

horstvogel, 2023-07-17:
Einmal alles durchprobiert, bei mir geht die VW App und auch meine Home Assistant Integration für den ID Buzz funktioniert, nur in EVCC funktioniert das nicht.

andig, 2023-07-17:
Vmtl. neue Terms&Conditions die zu akzeptieren sind.

closed 2023-07-17 by andig

fab7282, 2023-07-17:
Hallo zusammen. Ist unter Einstellungen-> Dienstepakete -> einloggen. zu finden. Viele Grüße fab

horstvogel, 2023-07-17:
> Hallo zusammen. Ist unter Einstellungen-> Dienstepakete -> einloggen. zu finden. Es hat bei mir funktioniert. Viele Grüße fab

Bei VW finde ich den Punkt leider nicht. Kann mir da eventuell jemand helfen? 
Danke

Finn3690, 2023-07-17:
Wo im Audi finde ich diesen Punkt?

c-fy, 2023-07-17:
Ich habe keine offenen Terms&Coonditions finden können. Unter Einstellungen gibt es keinen Punkt Dienstepakete. Weder in den Einstellungen unter Volkswagen ID noch in denen unter myVolkswagen.

horstvogel, 2023-07-18:
> Hallo zusammen. Ist unter Einstellungen-> Dienstepakete -> einloggen. zu finden. Es hat bei mir funktioniert. Viele Grüße fab

Moin, ich habe das nach mehrfachen suchen auch bei VW nicht gefunden. App funktioniert, eine Anbindung via Home Assistant auf die API liefert auch zuverlässig weiterhin seine Werte.

StefanRu1, 2023-07-18:
Auch hier kein Connect zum Audi seit ein paar Tagen. In App und my Audi nach neuen neue Terms&Conditions gesucht ist aber nichts zu finden. Fehler ist:
[main  ] ERROR 2023/07/18 09:33:06 creating vehicle ev1 failed: cannot create vehicle 'template': cannot create vehicle 'etron': invalid character 'd' looking for beginning of object key string
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'etron': invalid character 'd' looking for beginning of object key string

pzeinlinger, 2023-07-18:
Selbes Problem hier. Weder bei Audi noch bei VW neue Terms&Conditions.

leika65, 2023-07-18:
Inzwischen läuft die Cubra App und auch die Cubra Webseite zum login-
Alle Daten sind nun wider aktiv aber leider läuft  EVCC nicht.
Komischerweise hatte der Iobroker Adapter „vw-connect“ überhaupt keine Probleme auf die Daten zuzugreifen.

EVCC bring immer noch den Fehler:
cannot create vehicle 'template': cannot create vehicle 'seat':
invalid character 'd' looking for beginning of object key string

Thema "Terms&Coonditions" ist wohl was für VW aber in den SEAT Einstellungen bzw. Cubra nicht zu finden.

tomgeyer, 2023-07-18:
Bei mir das gleiche Problem mit VW ID.3. Hat schon jemand eine Lösung?

horstvogel, 2023-07-18:
Moin, sorry für die Verständnisfrage: 
es ist jetzt ein bug da es auf der Seite von EVCC behoben werden muss/wird? 
Danke!

horstvogel, 2023-07-18:
Sieht gut aus, bei mir ist die Fehlermeldung weg. 
Vielen Dank!

leika65, 2023-07-18:
Mit der neuen Version 0.118.9 ist die Fehlermeldung behoben und das Fahrzeug wird wieder eingebunden.

Es scheint der Bug seitens EVCC behoben.

Finn3690, 2023-07-18:
Besten Dank.😊