evcc-io/evcc #8995: Citroën eC4 not longer recognized - displayed as offline

mailkluge, 2023-07-17:
### Describe the bug

With Version 0.118.8 current SoC of Citroen eC4 is not longer recognized. EC4 is visualized as (Offline) in evcc.

After I downgraded to version 0.117.4 it works again. 

### Steps to reproduce

1. -
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.118.8

closed 2023-07-17 by andig

mailkluge, 2023-07-22:
@andig  Still not working with version 0.118.9. 
bug is still existing.

andig, 2023-07-22:
Kann gut sein. Ohne Logfile wie angefragt wird sich daran nichts ädern. Daher ist das Issue geschlossen.

mailkluge, 2023-07-22:
Sorry @andig das steht hier nirgends. Im Kommentar stand „already solved“. Gerne schicke ich das logfile. Gehe davon aus, dass ihr automatische TestCases habt gegen die Schnittstellen der Automobilisten.

andig, 2023-07-22:
> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

Das steht bei jedem neuen Issue. Hier brauchts ein trace log.