evcc-io/evcc #9032: KEBA,RFID,AWATTAR,OCPP (beschrieben in #8356)

KaiseAl1, 2023-07-20:
Ich darf nochmal auf #8356 zurück kommen

Grundsätzlich kann das Problem, dass der Ladevorgang welcher auf einen günstigeren Tarif (zB Awattar) wartet, aber zum tatsächlichen Startzeitpunkt an der Keba nicht mehr authentifiziert ist, damit umgangen werden, dass im yaml ein fixer RFID TAG beim charger hinterlegt wird (rfid: xxxxx). Damit startet der Ladevorgang immer mit diesem hinterlegten Tag.

Im Umkehrschluss ist die RFID Authentifizierung an der Keba aber vollständig unwirksam (und für die Abrechnung via OCPP auch falsch), da jeder Ladevorgang auch ohne Vorhalten einer Karte startet. Mehrfach getestet. Man könnte daher die Authentifizierung auch gleich abschalten

Für die Nutzung auch in etwas größeren Umgebungen (Firmenladenetzen) wäre eine korrekte Funktion mit RFID Authentifizierung aber Voraussetzung:

Ich möchte daher nochmals einen Workarround ins Spiel bringen der ggfs am einfachsten umzusetzen ist:

- Das Grundproblem besteht darin, dass beim Anstecken des Fahrzeugs und anschliessender Anmeldung mit RFID, der Ladevorgang nur in den Status "preparing" geht (lt OCPP) aber wegen des Tarifs von EVCC nicht freigegeben wird.
- Von diesem "preparing" Status kommt der Ladevorgang auch nicht mehr weg wenn der Tarif später stimmt. Man müsste dann erneut die RFID Karte vorhalten.
- Würde man den Ladevorgang nach dem Anstecken/Anmelden kurz starten und danach wieder unterbrechen hätten man den Status "suspended" und damit eine aufrechte Authentifizierung für später. Ein folgender Restart des Ladevorgangs wenn der Tarif passt würde dann problemlos erfolgen.

Ich würde daher vorschlagen im yaml zB eine Einstellung "prestart: 1" beim charger zu hinterlegen wobei "1" die Anzahl der Minuten ist, die der Prestart andauern soll. Gleich nach dem Anstecken/Anmelden wird der Ladevorgang in den Modis PV und Min+PV für diese Zeit gestartet bzw für diese Zeit in den Modus "schnell" umgeschaltet.

Manuell durchgeführte Tests führen genau zum gewünschten Ergebnis.

lg

Alex

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

premultiply, 2023-07-20:
Duplicate of https://github.com/evcc-io/evcc/issues/8356

Leonidos1973, 2023-07-20:
Ich habe exakt dieselbe Situation und stimme mit Dir in allen Punkten überein. Auch den Lösungsansatz und die Beobachtungen von Dir kann ich exakt nachvollziehen. 
Ich nutze Steve und OCPP ausschließlich für die Authentifizierung meiner Outdoor-Wallbox. Es sind hierbei zwei Keba Wallboxen x-Serie + c-Serie mit RFID Leser die als Keba-Ladenetzwerk funktionieren. Die RFID Tags sind hierbei in Steve angelernt und stehen so beiden Wallboxen für die Authentifizierung zur Verfügung. Den RFID Tag in evcc direkt der Wallbox zuzuordnen macht tatsächlich keinen Sinn - mir geht's dabei noch nicht mal um die Unterscheidungsmöglichkeit sondern um den Sicherheitsaspekt. Dann könnte ich die Authentifizierung auch lassen.
Ich werde vermutlich in meinem NodeRed nen Workflow erstellen welcher auf das Anstecken und Authentifizieren reagiert und dann evcc auf Schnellladen umstellt und nach 5 Minuten auf den vorherigen Wert zurücksetzt. Dieser Weg, wie auch von Alex beschrieben funktioniert bei mir ebenso.

premultiply, 2023-07-20:
Die zugelassenen Identifier werden in evcc den jeweiligen Fahrzeugen zugewiesen.
Alle anderen Identifier erhalten keine Ladeerlaubnis.

Am Loadpoint:
```yam
resetOnDisconnect: true
mode: off
```

KaiseAl1, 2023-07-21:
Evtl liegt hier das Problem begraben:
Damit die Erkennung des Fahrzeuges über die vorgehaltene Karte funktioniert, müssen die in den identifiers hinterlegten tags mit Nullen auf 16 Stellen aufgefüllt werden. Damit scheint aber später die Authentifizierung beim Start des Ladevorgangs nicht mehr zu funktionieren. 

identifiers:
      - abcdef0000000000

premultiply, 2023-07-21:
Welche spätere Authentifizierung beim Start ???

Karte dranhalten, evcc erkennt das Fahrzeug und schaltet seinen Lademodus von Default off auf den Default des Fahrzeugs, fertig.

premultiply, 2023-07-21:
Den genauen übermittelten Identifier fragt man mittels `evcc charger`. Siehe Doku.

Der Identifier sollte ein String sein.

Diese idtag-Option ist IMHO überflüssig.
Mir konnte noch niemand dazu einen wirklich sinnvollen Einsatzzweck zeigen.
Basiert vermutlich ebenfalls auf einem Missverständnis.

KaiseAl1, 2023-07-21:
Nochmals kurz zusammengefasst:

- Die Fahrzeuge sind bei mir mit den identifiers angelegt und zwar so wie der RFID tag im evcc log angezeigt wurden mit 16 Stellen hinten mit Nullen aufgefüllt. Wird der Tag so wie auf der Keba erkannt (oder mit Smartphone ausgelesen) ohne die abschliessenden Nullen im yaml hinterlegt klappt die Fahrzeugerkennung im EVCC nicht.
- Ursprünglich war bei mir in den chargers KEIN "rfid: xxxx" angelegt. Damit hat ein Ladevorgang der verzögert durch einen Tarif gestartet hätte werden sollen eben nicht gestartet.
- Erst als ich im chargers "rfid: xxx" hinzugefügt habe (und zwar OHNE der abschliessenden Nullen) hat der Ladevorgang bei späterer Aktivierung durch den Tarif geklappt. Aber dann ist natürlich immer dieser statisch am charger hinterlegte tag ans OCPP backend gesendet worden was der Abrechnung nicht dienlich ist.

Ich werde es demnächst ohne OCPP Backend testen. Mag sein, dass es dann klappt. Es wäre aber trotzdem keine Lösung weil die Abrechnung ja stimmen muss. 
Spannend wäre herauszufinden ob es ein spezielles Verhalten der Keba ist oder auch mit anderen Wallboxen mit OCPP Backend auftritt. Wobei Keba in diesem Markt kein Nischenplayer ist.

premultiply, 2023-07-21:
Es ist müßig hier immer weiter zu philosophieren solange es mehrere Köche im System gibt.
Die Wallbox darf und kann nur von einem einzigen System und über einen Weg angesteuert werden.

Die Daten landen im evcc-Ladelog und können dort exportiert werden.

Die Erkennung der Fahrzeuge würde recht wahrscheinlich auch ganz ohne RFID funktionieren.

`rfid: xxx` weglassen. Das ist hier und auch sonst in der Regel unnötig und stört mehr als es nutzt.

Ob die gemeldeten Identifier so oder so formatiert sind spielt erstmal keine wirkliche Rolle. evcc muss sie nur (wieder)erkennen.
Vielleicht ist da noch intern irgendwas im Byte-Decoding vertauscht.
Das ist aber ein ganz eigenes Thema. -> Bitte **neues Thema** mit entsprechenden vergleichenden Traces von Keba via OCPP, UDP und Modbus und der Anzeige in der Keba Web-UI.

KaiseAl1, 2023-07-21:
Wie ich bereits gesagt habe, akzeptiere ich es, wenn evcc der Meinung ist, dass der Schuh nicht passt. Aber die Anforderung von Keba, OCPP-Backend, EVCC, Awattar/PV gibt es ganz klar, weil eine Abrechnung zu internationalen Kartensystemen möglich sein soll/muss. Ich habe eine ganze internationale Hotelkette die daran interessiert ist. Ein Export aus dem Ladelog aus EVCC heraus bringt in diesem Fall nichts.

ich werde gerne nochmals div Tests durchführen nur so lange EVCC der Meinung ist, dass sie die alleinige Steuerung der Ladung haben müssen (ohne OCPP Auth), ist es auch nur Aufwand der nichts bringt.

lg

Alex

premultiply, 2023-07-21:
Mit Meinung hat das nichts zu tun.
Die Protokolle und Boxen funktionieren nun mal so wie sie funktionieren.
evcc deckt bestimmte Use Cases ab.
Es gibt bisher nicht den Anspruch und auch nicht die Möglichkeit "alles" zu können.

Um aber erstmal zu analysieren ob es überhaupt eine Chance gibt dass das was du dir da vorstellst funktionieren könnte bedarf es eben eines strukturierten Vorgehens die nach Möglichkeit zunächst alle potenziellen externen Störfaktoren und Fehlerquellen eliminiert.
Erst wenn die Basis sicher funktioniert kann man weitersehen.
Jedenfalls handhabe ich das so.
Ansonsten müsstest du hier vielleicht alleine weitermachen.

KaiseAl1, 2023-07-24:
Protokolle zum Test in folgender Konfiguration:

- OCPP Backend Steve aus der Keba-Config entfernt und durchgestartet
- RFID Auth in der Keba weiterhin aktiviert mit Setting: "Onlylocal"
- 2 RFID Karten lokal in der Keba angelernt: 89e5xxxxxxxxxxxx und 04dbxxxxxxxxxxxx (04db wird für den Testladenvorgang verwendet, mit 89e5 wurde der vorangegangene Ladevorgang durchgeführt)
- Yaml: Rfid: xxx aus dem Charger entfernt
- Yaml: Loadpoint: resetOnDisconnect: true und mode: off
- Yaml: Vehicle: identifiers bleiben weiterhin hinterlegt (16-stellig), sowie onIdentifiy: mode: pv .....

Was zuerst auffällt:
Keba oder EVCC merkt sich jenen RFID Tag mit welchem der letzte Ladevorgang abgeschlossen wurde und liefert diesen bei der Abfrage evcc charger, auch wenn kein Auto angesteckt ist.

keba2
-----
Power:          0W
Energy:         887.9kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false
Identifier:     89e5xxxxxxxxxxxx

Wird folgend ein Auto angesteckt wird ebenso dieser "alte" RFID tag geliefert, obwohl eine ganz andere Karte an der Keba vorgehalten wurde. Der Tag der vorgehalten wurde, wird im Log ignoriert.

Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 charge currents: [0 0 0]A
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 charge total import: 887.930kWh
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 charger status: B
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 charge start energy: 888kWh
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] INFO 2023/07/24 11:21:08 car connected
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 vehicle api refresh
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 pv timer elapse
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 pv timer inactive
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 charger: guard elapse
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 guard timer inactive
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 charge total import: 887.930kWh
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 charger vehicle id: 89e5xxxxxxxxxxxx
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] INFO 2023/07/24 11:21:08 vehicle updated: unknown -> BMW
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 set charge mode: pv
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 set min current: 6
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 set max current: 13
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 set min soc: 15
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 set target soc: 80
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 set priority: 1
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 vehicle status: B
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 vehicle soc: 34%
Jul 24 11:21:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:21:08 vehicle range: 229km
Jul 24 11:21:08 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:21:08 ----

Warten auf passenden Tarif ab 12:00h....
Keba GUI zeigt zu diesem Zeitpunkt die Ladestation als "Verfügbar" an.
EVCC GUI zeigt "Verbunden" mit dem richtigen Fahrzeug (hinterlegt in identifiers), mode ist korrekt mit PV angezeigt.

Jul 24 11:59:48 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:48 ----
Jul 24 11:59:48 evcc evcc[1042]: [lp-1  ] DEBUG 2023/07/24 11:59:48 charge power: 0W
Jul 24 11:59:48 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:59:48 charge power: 0W
Jul 24 11:59:48 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:48 grid power: 13W
Jul 24 11:59:48 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:48 grid powers: [12 0 1]W
Jul 24 11:59:48 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:48 grid currents: [0.146 0.026 0.029]A
Jul 24 11:59:48 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:48 pv power: 0W
Jul 24 11:59:48 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:48 site power: 13W
Jul 24 11:59:48 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:59:48 charge currents: [0 0 0]A
Jul 24 11:59:48 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:59:48 charge total import: 887.930kWh
Jul 24 11:59:48 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:59:48 charger status: B
Jul 24 11:59:48 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:59:48 pv charge current: 0A = 0A + -0.0183A (13W @ 3p)
Jul 24 11:59:58 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:58 ----
Jul 24 11:59:58 evcc evcc[1042]: [lp-1  ] DEBUG 2023/07/24 11:59:58 charge power: 0W
Jul 24 11:59:58 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 11:59:58 charge power: 0W
Jul 24 11:59:58 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:58 grid power: 11W
Jul 24 11:59:58 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:58 grid powers: [10 0 1]W
Jul 24 11:59:58 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:58 grid currents: [0.133 0.027 0.029]A
Jul 24 11:59:58 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 11:59:58 pv power: 0W
[... 2 lines cut ...]
Jul 24 11:59:58 evcc evcc[1042]: [lp-1  ] DEBUG 2023/07/24 11:59:58 charge total import: 9187.326kWh
Jul 24 11:59:58 evcc evcc[1042]: [lp-1  ] DEBUG 2023/07/24 11:59:58 charger status: A
Jul 24 12:00:08 evcc evcc[1042]: [site  ] DEBUG 2023/07/24 12:00:08 ----
Jul 24 12:00:08 evcc evcc[1042]: [lp-1  ] DEBUG 2023/07/24 12:00:08 charge power: 0W
[... cut ...]
Jul 24 12:00:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 12:00:08 max charge current: 13A
Jul 24 12:00:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 12:00:08 pv timer elapse
Jul 24 12:00:08 evcc evcc[1042]: [lp-2  ] DEBUG 2023/07/24 12:00:08 pv timer inactive
Jul 24 12:00:08 evcc evcc[1042]: [lp-2  ] ERROR 2023/07/24 12:00:08 charger enable: missing credentials for RFID authorization

EVCC GUI zeigt rechts oben ebenso die Warnmeldung "Missing credentials"

Für mich im Verhalten kein Unterschied zum aktivierten OCPP Backend Steve feststellbar. 

lg

Alex

KaiseAl1, 2023-07-24:
Habe mir jetzt noch den Spaß gemacht und ein anderes Auto an der gleichen Keba angesteckt. Es wird wieder der "alte" Tag 89e5xxxxxxxxxxxx im log angezeigt und somit das Auto falsch erkannt

KaiseAl1, 2023-07-24:
Protokolle zum Test mit Keba Anbindung via Modbus
alle anderen Einstellungen gleich wie im Test oben

Folgende Veränderungen fallen auf:
RFID Tags werden wie in der Keba-Modbus Doku beschrieben auf 4 Bytes beschränkt übergeben
Es wurden nur 2 Testläufe durchgeführt, aber es scheint, als wird immer der korrekte tag übergeben welcher tatsächlich an der Keba vorgehalten wurde

Jul 24 13:51:37 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:37 ----
Jul 24 13:51:37 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 13:51:37 charge power: 0W
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 charge power: 0W
Jul 24 13:51:37 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:37 grid power: 11W
Jul 24 13:51:37 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:37 grid powers: [10 0 1]W
Jul 24 13:51:37 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:37 grid currents: [0.134 0.027 0.03]A
Jul 24 13:51:37 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:37 pv power: 0W
Jul 24 13:51:37 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:37 site power: 11W
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 charge currents: [0 0 0]A
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 charge total import: 887.948kWh
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 charger status: B
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 charge start energy: 888kWh
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] INFO 2023/07/24 13:51:37 car connected
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 vehicle api refresh
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 pv timer elapse
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 pv timer inactive
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 charge total import: 887.948kWh
Jul 24 13:51:37 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 charger vehicle id: 04dbxxxx
Jul 24 13:51:37 evcc evcc[1378]: [main  ] DEBUG 2023/07/24 13:51:37 vehicle status: B (BMW)
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] INFO 2023/07/24 13:51:37 vehicle updated: unknown -> BMW
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 set charge mode: pv
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 set min current: 6
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 set max current: 13
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 set min soc: 15
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 set target soc: 80
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:37 set priority: 1
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 vehicle status: B
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 vehicle soc: 34%
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 vehicle range: 229km
Jul 24 13:51:38 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:38 ----
Jul 24 13:51:38 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 13:51:38 charge power: 0W
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 charge power: 0W
Jul 24 13:51:38 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:38 grid power: 11W
Jul 24 13:51:38 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:38 grid powers: [10 0 1]W
Jul 24 13:51:38 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:38 grid currents: [0.135 0.027 0.029]A
Jul 24 13:51:38 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:38 pv power: 0W
Jul 24 13:51:38 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:38 site power: 11W
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 vehicle odometer: 5613km
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 charge currents: [0 0 0]A
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 charge total import: 887.948kWh
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 charger status: B
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 pv charge current: 0A = 0A + -0.0162A (11W @ 3p)
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 pv enable timer reset
Jul 24 13:51:38 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:38 pv timer inactive
Jul 24 13:51:47 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:47 ----
Jul 24 13:51:47 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 13:51:47 charge power: 0W
Jul 24 13:51:47 evcc evcc[1378]: [lp-2  ] DEBUG 2023/07/24 13:51:47 charge power: 0W
Jul 24 13:51:47 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:47 grid power: 9W
Jul 24 13:51:47 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:47 grid powers: [8 0 1]W
Jul 24 13:51:47 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:47 grid currents: [0.125 0.027 0.029]A
Jul 24 13:51:47 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:47 pv power: 0W
Jul 24 13:51:47 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 13:51:47 site power: 9W
Jul 24 13:51:47 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 13:51:47 charge currents: [0 0 0]A
Jul 24 13:51:47 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 13:51:47 charge total import: 9187.326kWh
Jul 24 13:51:47 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 13:51:47 charger status: A

Warten auf Tarif ab 14:00h

Jul 24 14:00:17 evcc evcc[1378]: [site  ] DEBUG 2023/07/24 14:00:17 ----
Jul 24 14:00:17 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 14:00:17 charge power: 0W
[... cut ...]
Jul 24 14:11:47 evcc evcc[1378]: [lp-1  ] DEBUG 2023/07/24 14:11:47 charger status: A

EVCC GUI: Günstige Energie verfügbar. Ladevorgang aktiv … 
Keba GUI: Zustand "Laden"

Auch wenn es sich in den Logs anders darstellt, sehe ich die gleiche Problematik, dass Keba zum eigentlichen Start der Ladung (Tarif OK) über keine RFID Authentifizierung mehr verfügt.
Wenn ich es in der Modbus-Doku richtig verstehe, kann EVCC die Ladestation nur aus/ein-schalten, nicht aber einen RFID Tag zur Authentifizierung mitschicken.

KaiseAl1, 2023-07-24:
Zur Vollständigkeit wurde noch ein Test mit abgeschalteter RFID Authentifizierung an der Keba durchgeführt. 
Die Erkennung der Fahrzeuge erfolgt hier mehr oder weniger zufällig bzw mit dem Tag des vorhergehenden Ladevorgangs. Detaillierte Tests machen in dieser Config aus meiner Sicht keinen Sinn.

premultiply, 2023-07-25:
Vergiss erstmal die UDP-Geschichte und wir konzentrieren uns zukünftig hier erstmal nur auf die Modbus-Anbindung.

> Auch wenn es sich in den Logs anders darstellt, sehe ich die gleiche Problematik, dass Keba zum eigentlichen Start der Ladung (Tarif OK) über keine RFID Authentifizierung mehr verfügt.

Dazu sehe ich keinen Grund.
Eine lokale Authentifizierung an der Box startet dort eine lokale Session.
Diese bleibt normalerweise erhalten bis das Fahrzeug abgesteckt wird.

> Wenn ich es in der Modbus-Doku richtig verstehe, kann EVCC die Ladestation nur aus/ein-schalten, nicht aber einen RFID Tag zur Authentifizierung mitschicken.

Das ist auch unnötig und unüblich.
Denn das Ganze funktioniert ja im Prinzip wie eine Tür mit Schloss und Sperriegel welchen von einem Portier von Innen bedient wird.
Der Portier (evcc) kann zusätzlich anhand der Farbe erkennen welcher Schlüssel (Identifier) im Schloss steckt und er weiss dann auch gleich welchem Besitzer (Fahrzeug) dieser jeweils zugeordnet ist.
Entweder direkt über dessen Farbe, indirekt über eine Zeilennummer einer Farbtabelle oder über er kennt den Spitznamen. Hauptsache alle wissen welcher Schlüssel genau gemeint ist und wem er gehört.

Damit die Tür aufgeht muss natürlich sowohl das Schloss offen sein UND auch der Riegel geöffnet sein.
Es ist aber nicht erforderlich immer beide Schließmechanismen zu betätigen, allerdings ist eine Zutrittskontrolle durch den Portier seitens des Chefs geboten.
Nachts bleibt der Riegel immer vorgelegt.
Im Regelfall bleibt das Schloss aber immer auf oder ist sogar mit jedem beliebigen Schlüssel bedienbar. Der Portier regelt den Zugang nur durch die Wiedererkennung der Schlüssel und dann über den Riegel. Im Notfall (Schlüssel vergessen) oder bei fremden Besuchern kann der Chef dem Portier trotzdem mitteilen wer da vor der Tür steht damit der Portier ihn einlassen und ihn begrüßen kann.
Jeder Eintretende wird in einem Anwesenheitszettel vom Portier protokolliert und beim Verlassen des Orts nach den Schließen der Tür wird der Zettel gleich abgeheftet und es wird auch gleich gründlich durchgewischt und alles wieder in den Ursprungszustand gebracht.

premultiply, 2023-07-25:
> Zur Vollständigkeit wurde noch ein Test mit abgeschalteter RFID Authentifizierung an der Keba durchgeführt.
> Die Erkennung der Fahrzeuge erfolgt hier mehr oder weniger zufällig bzw mit dem Tag des vorhergehenden Ladevorgangs. Detaillierte Tests machen in dieser Config aus meiner Sicht keinen Sinn.

Dazu müssten die Identifier aus der evcc Konfig entfernt werden wenn die Box die nicht sauber resettet.
Für die automatische Erkennung werden diese nicht benötigt.

KaiseAl1, 2023-07-25:
OK, modbus - sieht für mich auch runder aus aber ich sehe im Ergebnis tatsächlich keinen Unterschied zu UDP

Das mit Portier und Schlüssel verstehe ich schon, aber offensichtlich akzeptiert das Schloss (Keba) vom Portier (evcc) die Aufforderung zur Öffnung nach einer bestimmten Zeit nicht mehr.

Aus den Tests geht hervor, dass die Keba dem EVCC nur für kurze Zeit vertraut:
Bekommt die Keba den Schlüssel und öffnet der Portier sofort, dann passt alles
Bekommt die Keba den Schlüssel, der Portier wartet aber noch ab (Tarif), vergisst die Keba den Schlüssel wieder und der Portier kann danach nicht mehr öffnen.

Um bei der Analogie zu bleiben, schaltet die Keba die Falle nur für einen bestimmte Zeit frei.
Daher war ja mein Workaround "Türe gleich für kurze Zeit öffen, dann wieder anlehnen"

Soll ich noch etwas testen?

Zur Variante mit abgeschalteter RFID:
zB ein Auto ist ein Mercedes EQV, der lt logs keine automatische Erkennung unterstützt. Das war auch bei dem Test so

premultiply, 2023-07-27:
Zumindest kann ich schon mal sagen dass dies ein spezifischer Effekt der Keba sein muss.
Habe es gerade nochmal mit einem Testboard nachgestellt, welches uns Phoenix Contact freundlicher Weise zur Verfügung gestellt hat, mit dessen CHARX-Controller nachgestellt. Dort gibt es so ein Problem nicht.
Die Authentifizierung bleibt wie erwartet bestehen bis das Fahrzeug abgezogen wird.

Vielleicht müsste man da bei der Keba erstmal mit einem vergleichenden Modbus-Tracelog dran gehen um zu sehen wie sich die Status-Register 1000 und 1004 in den beiden Fällen ggf. voneinander unterscheiden.

Vielleicht kann man den Zustand zumindest in der Implementierung erkennen.
Die Doku ist dazu zu wage.

Ein anderer Test ist zu prüfen ob die Keba auch im Free-Charge-Mode (also ohne Authentifizierung) den Kartenleser auswertet und dessen Identifier übermittelt oder ob dann dort nichts mehr kommt.
Kann man mit evcc meter schnell sehen.

KaiseAl1, 2023-07-28:
Den letzten Test im "Free-Modus" habe ich bereits durchgeführt. Mein EQV wurde nicht erkannt und ich konnte im log auch keinen Tag erkennen.

Bez Log, werde ich noch einen Test versuchen und dann das Wallbox log auswerten. Ich denke, dass man damit am weitesten kommt:

2023-07-24T07:54:48,497 DEBUG - LogEvt: Event time: [2023-07-24 07:54:47,734] [20943373] Check 4 HostAuthReq without RFID 0x5 0x7e 0xc 0
2023-07-24T07:54:48,687 DEBUG - LogEvt: Event time: [2023-07-24 07:54:47,773] [20943373] [CPLD] (0:04,01) EvPlugged;HW_Lock
2023-07-24T07:54:54,256 DEBUG - LogEvt: Event time: [2023-07-24 07:54:54,220] [20943373] AUTARK state change 0 -> 1
2023-07-24T07:54:54,372 DEBUG - LogEvt: Event time: [2023-07-24 07:54:54,321] [20943373] AUTARK state change 1 -> 7
2023-07-24T07:54:54,966 DEBUG - LogEvt: Event time: [2023-07-24 07:54:54,922] [20943373] AUTARK state change 7 -> 6
2023-07-24T07:54:55,065 DEBUG - LogEvt: Event time: [2023-07-24 07:54:55,023] [20943373] AUTARK state change 6 -> 0
2023-07-24T07:54:56,481 DEBUG - LogEvt: Event time: [2023-07-24 07:54:56,444] [20943373] AUTARK state change 0 -> 1
2023-07-24T07:54:56,588 DEBUG - LogEvt: Event time: [2023-07-24 07:54:56,545] [20943373] AUTARK state change 1 -> 7
2023-07-24T08:02:49,116 DEBUG - LogEvt: Event time: [2023-07-24 08:02:49,092] [20943373] Pilot mapped : 0
2023-07-24T08:02:49,140 DEBUG - LogEvt: Event time: [2023-07-24 08:02:49,093] [20943373] Pilot Pulse: 100
2023-07-24T08:02:49,487 DEBUG - LogEvt: Event time: [2023-07-24 08:02:49,141] [20943373] [CPLD] (0:00,01) ;HW_Lock
2023-07-24T08:02:49,563 DEBUG - LogEvt: Event time: [2023-07-24 08:02:49,162] [20943373] logState 1 : 0x0
2023-07-24T08:02:49,646 DEBUG - LogEvt: Event time: [2023-07-24 08:02:49,263] [20943373] AUTARK state change 7 -> 6
2023-07-24T08:14:06,232 DEBUG - LogEvt: Event time: [2023-07-24 08:14:06,200] [20943373] [GPB] DSO is offline