evcc-io/evcc #9041: Planner: Incorporation of feedin/charge tradeoff

kalaws, 2023-07-20:
If I have scheduled planner to complete charging at X o’clock, say Monday 6 AM and time is now Sunday 10 AM, sometimes it is better to export pv surplus rather than charge. For example, feedin price may presently be 0,15, and around midnight grid cost may be 0,05. Hence it is better to postpone charging to midnight hours, and export the existing surplus now. As far as I can see, planner does not currently consider this, but only considers differences in grid cost. 

I don’t know whether it would actually be possible to implement this behavior into planner, but perhaps there are others who would like to speculate on how it might be done?

andig, 2023-07-20:
> I don’t know whether it would actually be possible to implement this behavior

What exactly would that be? A cost function that is a combination (sum or difference) of tariffs? 

/cc @premultiply

kalaws, 2023-07-20:
I’m not sure how it is properly expressed. I suppose if surplus>charge minimum, then difference between hr 1 feedin price and hr 2 grid cost. If difference is positive (above some defined threshold, to not get silly) then schedule charging for hr 2, not hr 1. Something along those lines?

andig, 2023-07-20:
I don't understand :/

kalaws, 2023-07-20:
Ok, so I'll try to illustrate my thinking.

Let's say 5 hrs of charging is needed to reach a specified SOC in 24 hrs.
Two tariff lists, `Grid` and `Feedin`, exist. Whether there is PV `Surplus` or not determines which one reflects the `True cost` of energy. Out comes a single tariff list, `True cost`. This allows us to rank the hours by price. The existing planner logic applies from there, and decides whether to `Charge`. The difference from the current functioning of the planner is that some charging will be done when there is surplus, some charging will be scheduled for when there is not surplus.

```
	Grid	Feedin	Surplus	True	Charge
Hr	tariff	tariff	Y/N	cost	Y/N
1	1	0,8	N	1	N
2	1	0,8	N	1	N
3	1	0,8	N	1	N
4	1	0,8	N	1	N
5	1	0,8	N	1	N
6	0,8	0,64	N	0,8	Y
7	0,8	0,64	N	0,8	Y
8	1	0,8	N	1	N
9	1,3	1,04	Y	1,04	N
10	1,2	0,96	Y	0,96	Y
11	1,5	1,2	Y	1,2	N
12	1,5	1,2	Y	1,2	N
13	1,6	1,28	Y	1,28	N
14	1,6	1,28	Y	1,28	N
15	1,5	1,2	Y	1,2	N
16	1,3	1,04	Y	1,04	N
17	1,4	1,12	Y	1,12	N
18	1,2	0,96	Y	0,96	Y
19	0,9	0,72	Y	0,72	Y
20	1,1	0,88	N	1,1	N
21	1,3	1,04	N	1,3	N
22	1,4	1,12	N	1,4	N
23	1,3	1,04	N	1,3	N
24	1,1	0,88	N	1,1	N
```

In reality, this may be complicated by the fact that it is not necessarily easy to know the non-present state of `Surplus`, without forecasting. But perhaps this logic can be used on the present hour, when surplus is known?

kalaws, 2023-07-20:
Below table illustrates the same solution but without assuming that we can forecast surplus. In this case, 4 hrs are needed to reach the wanted SOC. (Of course, hrs 1-12 before `NOW` are not really relevant at all.)

```
	Grid	Feedin	Surplus	True	Charge
Hr	tariff	tariff	Y/N	cost	Y/N
1	1	0,8	N	1	N
2	1	0,8	N	1	N
3	1	0,8	N	1	N
4	1	0,8	N	1	N
5	1	0,8	N	1	N
6	0,8	0,64	N	0,8	N
7	0,8	0,64	N	0,8	N
8	1	0,8	N	1	N
9	1,3	1,04	N	1,3	N
10	1,2	0,96	N	1,2	N
11	1,5	1,2	N	1,5	N
12	1,5	1,2	N	1,5	N
----------------------------------------------
13 NOW	1,6	1,28	Y	1,28	N
----------------------------------------------
14	1,6	1,28	N	1,6	N
15	1,5	1,2	N	1,5	N
16	1,3	1,04	N	1,3	N
17	1,4	1,12	N	1,4	N
18	1,2	0,96	N	1,2	Y
19	0,9	0,72	N	0,9	Y
20	1,1	0,88	N	1,1	Y
21	1,3	1,04	N	1,3	N
22	1,4	1,12	N	1,4	N
23	1,3	1,04	N	1,3	N
24	1,1	0,88	N	1,1	Y
```

kalaws, 2023-07-21:
Perhaps, with the second suggestion, we have stepped outside of what is the point of the planner? I.e., the planner is not concerned with continually evaluating the present hour, but only ranks _one time_ (when a departure hour is set) a list of hours by grid tariff, which is the only tariff that can safely be assumed as long as we cannot forecast pv surplus?

andig, 2023-07-21:
Afaikt the planner would have made the exact same plan with grid information only. Still not clear what you‘re asking.